dresden-elektronik/deconz-rest-plugin #585: Cannot read from device (Raspbee not working)

Jasperste, 2018-05-15:
I'm trying to setup my new Raspbee to connect to my (IKEA) lights, but after searching no lights can be found. I've tried a lot over the past couple of days and weeks, but it seems like it's having troubles with the firmware/communicating with the device.
I tried updating the firmware (`sudo GCFFlasher -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF`) and get the following result:
```
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
no device specified use RaspBee (/dev/ttyS0)
using firmware file: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF
reset target

error: failed to read from device
```

Running `sudo GCFFlasher -r` however does succeed:
```
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
no device specified use RaspBee (/dev/ttyS0)
reset target SUCCESS
```

The Deconz web-app doesn't seem to be able to find the firmware either and states the firmware is `00000000` (also giving an option to update, which also does not succeed).

I'm currently using the preconfigured MicroSD-card version from here https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/raspbee-gw-sd-card-image/?L=1, but I also tried everything described in README with a fresh `jessie` image. In the end, both end up with above errors.
The Raspbee is connected to the correct pins (described here: https://www.dresden-elektronik.de/fileadmin/Downloads/Dokumente/Produkte/ZLL/RaspBee-BHB-en.pdf) and has a steady red light when connected (some green when booting). 
I've factory resetted all lights and turning them on and of when deconz goes searching for them. 

The Deconz desktop app also forcing me to update the firmware, but pressing the 'update Firmware' button and awaiting the loader just ends up with showing that button again (screenshot below):
[image]

I'm starting to think my (newly bought) Raspbee is broken, but maybe it's 'just' a wrong state somewhere.

Thanks

ebaauw, 2018-05-15:
Some ideas you probably checked already:
- Did you exit deCONZ before running GCFFlasher?
- Did you double-check the serial port settings of the RaspBerry? (see https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements)
- Did you unplug any ConBee and/or other USB devices? (see #558)
- Did you check the firmware image checksum?  Did you try another firmware version? (see https://www.dresden-elektronik.de/rpi/deconz-firmware)

Jasperste, 2018-05-15:
@ebaauw thanks for your ideas! I've checked everything again but it unfortunately didn't help:
- Deconz isn't running while updating (killed all processes, `pgrep deCONZ` isn't showing any processes now)
- Double checked the serial port settings multiple time
- No Conbee or other USB devices are plugged in
- Thanks for the link, tried serveral other firmware, but same result (`error: failed to read from device`)

Thanks

manup, 2018-05-16:
With the pre-configured SD-Card it should just work since everything is configured already.

To flash manually use:

`sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF`

Is the RaspBee connected to the correct pins?

[image]

Jasperste, 2018-05-16:
Thanks! Just tried that line again, but still results in `error: failed to read from device`. It's connected as shown in your picture:
 
[image]

Jasperste, 2018-05-23:
@manup any suggestions left to get this Raspbee to function? Is it possible that I received a broken Raspbee?

manup, 2018-05-24:
Sorry for the late response, to me it does indeed look like the RaspBee is broken (which wonders me since its really rare that that happens) &mdash; maybe it's the Pi but I haven't seen this either.

The only thing to check for a broken Raspberry Pi is `$cat /proc/cpuinfo`. The Revision should look like
```
Revision        : a02082
```
If it starts with an `1` there might be something broken (overcurrent, undervoltage).

At least with the dresden elektronik sd-card image it should work out of the box. If you have another spare Raspberry Pi you may try it with it. Otherwise please send in the RaspBee for replacement.

Jasperste, 2018-05-28:
@manup No problem, thanks for your response. I've tried `$cat /proc/cpuinfo` and got
```
Revision        : a02082
```

I've tried the SD-card (Dresden Elektronik SD-card) and the Raspbee on other Raspberry Pi's, but unfortunately I got the same results here.

Jasperste, 2018-05-28:
Will do, thanks

closed 2018-05-28 by Jasperste

buckettea, 2018-07-11:
I'm now facing the same error, and it seems my raspbee is almost certainly bricked. I had it working fine on a RPi 2 model B+ and upgraded to a RPi3. All I did was change out the Rpi and noticed that the Pi didn't light up at first (known good pi). a couple of jacks with the power and it came up but with the continuous 'upgrade firmware' loop.

i had made various other changes to the O/S, but mainly around pi/root passwords, ssh, etc. and i doubt this contributed but thought i'd mention it.

i tried various versions and firmware with both the flasher and deCONZ and keep getting this error below. I even went back to the original Pi 2 and a fresh image that came with the device (i.e. exactly as i unpacked and installed).

i noticed the module felt a bit hot, but i don't know if this is normal? it still has an LED and doesn't give any errors on a reset and is listed with the flasher -l flag as /dev/ttyAMA0
The firmware reported in the GUI is "unknown" while in the web tool is "0000000".

Are there any other diagnostic tools i can use to verify if the Raspbee is damaged?

error: bootloader not responding
device used by other application?

i can see /dev/ttyA

manup, 2018-07-12:
>i had made various other changes to the O/S, but mainly around pi/root passwords, ssh, etc. and i doubt this contributed but thought i'd mention it.

It is very important to configure the serial port correctly. Please also try the official sd-card image, it should work out of the box.

https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/raspbee-gw-sd-card-image/?L=1