dresden-elektronik/deconz-rest-plugin #587: Request an external url

ooii, 2018-05-15:
Hi,

Is it possible to send a request on an external rest server in the actions?
I'd like to have some integration between my `deconz` and my `domoticz`.
Thanks.

ebaauw, 2018-05-15:
No.  You need to use a home automation platform (like HomeKit) which connects to both and supports rules.

manup, 2018-05-15:
The feature request (Webhooks) is already on the todo list, it's a cool feature but doesn't have a ETA yet.

ebaauw, 2018-05-15:
> HomeKit does not help. I'm far beyond the limit of ~120 accessories and HomeKit automations do not fire. I cross my fingers for an enhancement during next WWDC.

Afaik, there's a hard limit of 99 accessories per HomeKit bridge (like homebridge) and one of 1000 accessories overall.  There's also a limit that the whole (?) HomeKit configuration needs to fit in a single CKRecord of 1MB for iCloud synchronisation between devices (see https://forums.developer.apple.com/thread/64682), which is needed to sync automations to the AppleTV.  That doesn't translate directly to a fixed number of accessories, but 120 seems a bit low.  I used to hit that around 150 accessories with homebridge-hue (see https://github.com/KhaosT/HAP-NodeJS/issues/408), but that was before exposing the sensor history to Eve.

ooii, 2018-05-15:
I'm exposing, for sure, more than 100 accessories from domoticz but all are not needed. I need to exclude the ones I don't need. Will do that soon I hope.
My hue setting contains more than 100 accessories (eg. if I consider that a hue motion sensor is 3 homekit accessories). I'm planning to use resourcelinks to reduce this number. Do you confirm that it's a way of doing it @ebaauw?
To avoid the 99 accessories per HomeKit bridge, I'm running multiple instances of homebridge. And I'm still far from the 1000 limit per home. But the limit of 1 CKRecord is probably reached for me. I had issues several weeks ago syncing automations between different devices and, if I remember well, I has around 120 accessories at that time.

ebaauw, 2018-05-16:
> My hue setting contains more than 100 accessories (eg. if I consider that a hue motion sensor is 3 homekit accessories).

No, it’s not.  By default, homebridge-hue exposes the Hue motion sensor as one accessory (or two if you want history for the temperature sensor).  This accessory has multiple services (for motion, light level, temparature, but also _Accessory Information_, Battery_, and _History_).  See also https://github.com/ebaauw/homebridge-hue/wiki/Devices, but I haven’t updated that yet for history.

Apple’s Home app uses the term “accessory” inconsistently - sometimes it refers to accessories, sometimes to services.  Typically, the tiles show services (except for stateless programmable switches).  Check Eve or another HomeKit app to see what’s really happening.  Also, on startup, homebridge-hue issues log messages with the number of accessories.

> I'm planning to use resourcelinks to reduce this number. Do you confirm that it's a way of doing it @ebaauw?

Yes.  I’m running several MultiLight and MultiCLIP accessories to stay below the 99 bridged accessories limit.  This will also help a bit in the CKRecord limit (e.g. there only one _Accessory Information_ service for a MultiLight/CLIP accessory instead of one per light or sensor), but, of course the memory footprint of the MultiLight/CLIP accessory is larger.  Effectively, you’ll be able to run fewer accessories, but more services.

ooii, 2018-05-16:
Thanks Erik.
>>Yes. I’m running several MultiLight and MultiCLIP accessories to stay below the 99 bridged accessories limit.

Are the lights which belong to a multilight accessory automatically excluded?

ebaauw, 2018-05-16:
They are included, in that they are exposed to HomeKit, as a single accessory, which multiple LightBulb srvices.  In Apple’s Home app you hardly notice the difference from exposing each light as its own accessory (it still shows a tile per light), except that all lights move rooms at the same time, and all react to _Identify_.

ooii, 2018-05-17:
Thank you very much, I saved several accessories this way.