dresden-elektronik/deconz-rest-plugin #589: Rasbee not working after reboot

tomsteenbakkers, 2018-05-16:
Yesterday I received the Raspbee. Have installed it and installed the software using the instructions at GitHub. After installing the deConz GUI shows the Raspbee. 

After I reboot the PI neither deConz is working and homebridge.I can start the deConz software but I can not connect to the Raspbee. I see him in the list, but when pressing Connect nothing happens.

 I get the following log from deConz. 

```
-- Logs begin at Wed 2018-05-16 08:09:25 CEST. --
May 16 08:09:27 RaspberryPi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
May 16 08:09:31 RaspberryPi deCONZ[277]: libpng warning: iCCP: known incorrect sRGB profile
May 16 08:09:33 RaspberryPi deCONZ[277]: This plugin does not support propagateSizeHints()
May 16 08:09:33 RaspberryPi deCONZ[277]: This plugin does not support propagateSizeHints()
May 16 08:09:33 RaspberryPi deCONZ[277]: This plugin does not support propagateSizeHints()
```

fincha, 2018-06-08:
same for me, I am connected via remote desktop

tomsteenbakkers, 2018-06-08:
I’m using VNC

manup, 2018-06-09:
You may check if there is already a deCONZ version running.

    $ ps ax | grep deCONZ

tomsteenbakkers, 2018-06-09:
Check, this was the issue. I will connect a monitor 🤗

closed 2018-06-09 by tomsteenbakkers

manup, 2018-06-09:
Depending on the VNC server it should also work remotely. The one installed in our sd-card image does always open the current display session so remote logging into it just works fine.

ebaauw, 2018-06-09:
I'm using the Raspbian built-in VNC server, with authentication set to VNC password, so it works with macOS screen sharing.  Better copy/paste integration than VNC Viewer.  I created a Bonjour announcement on the pi, so the mac shows it in the Finder, see http://4dc5.com/2012/06/12/setting-up-vnc-on-raspberry-pi-for-mac-access/.  Needed Avahi for homebridge anyways.