dresden-elektronik/deconz-rest-plugin #590: Sensor control wouldn't leave first state

peer69, 2018-05-16:
Hi, 

I'm using a hue motion sensor to control lights in the hallway. They are supposed to be turned on when motion is detected and then turned off after two minutes after no motion has been detected, with two more tiggers to dim the lights after 30 sec and 1 min. Simple rule which looks like this in pwa: 

[image]
My problem is, only the initial trigger works for me. If motion is detected, the lights are turned on. But then the first state remains. All other states don't trigger at all even though the motion sensor stops detecting motion as you can see: 

[image]

So, obviously I'm doing something wrong but I can't seem to figure out what.

tomfritz1, 2018-05-20:
I have the same issue with my Xiaomi motion sensors after updating to latest firmware. With the old firmware it worked without problems.

manup, 2018-05-20:
There might be some glitches due the latest changes in the rules and daylight engine, I'll check the Phoscon App generated rules if they need to be adjusted to support latest code.

manup, 2018-05-22:
I can confirm the issue, the problem is that the rule operator `ddx` does only trigger on the exact second which was more relaxt in prior versions. 

I've adapted the code of the rule creation in Phoscon App, so the next version 2.05.28 should fix the problem. However in order to fix existing rules, the sensor control needs to be changed slightly to trigger the recreation of the rules, this can be done by any change &mdash; like changing brightness from 100% to 99%.

peer69, 2018-05-22:
Great! I will create new rules and test them as soon as 2.5.28 is released.

tomfritz1, 2018-05-25:
I still have the same issue with 2.05.28 -> going back to 2.05.25 solves the problem

manup, 2018-05-25:
Note you need to force the recreation of rules as mentioned above.

peer69, 2018-05-31:
With 2.05.28 the problem is solved for me, sensor states are triggered as expected. Thank you for solving this issue!

closed 2018-05-31 by peer69