dresden-elektronik/deconz-rest-plugin #591: Not recognising RaspBee

uli6, 2018-05-16:
Hey Awesome People,

Sorry if I don't something wrong, I'm no Dev. :(

I'm trying to use a Raspbee on my Raspberry (running a Hassio) and it looks like it's not recognizing the Raspbee.

Here's the log. 

`
23:32:22:461 Sqlite sensors: uniqueid = 00:00:00:00:00:00:00:00-01
23:32:22:461 ~ResourceItem() attr/uniqueid -- str 0x7005f0
23:32:22:461 Sqlite sensors: swversion = 1.0
23:32:22:462 Sqlite sensors: state = {"daylight":false,"lastupdated":"2018-05-16T22:15:26","status":230}
23:32:22:462 Sqlite sensors: config = {"configured":true,"lat":"38.722252","long":"-9.139337","on":true,"sunriseoffset":30,"sunsetoffset":-30}
23:32:22:462 Sqlite sensors: deletedState = normal
23:32:22:462 Sqlite sensors: mode = 1
23:32:22:462 DB found sensor Daylight 1
23:32:22:462 ~ResourceItem() attr/uniqueid -- str 0x5c71d0
23:32:22:462 ~ResourceItem() config/configured -- str (nil)
23:32:22:462 ~ResourceItem() config/sunriseoffset -- str (nil)
23:32:22:462 ~ResourceItem() attr/name -- str 0x5cc460
23:32:22:462 ~ResourceItem() attr/modelid -- str 0x703ac8
23:32:22:462 ~ResourceItem() attr/type -- str 0x704318
23:32:22:462 ~ResourceItem() config/on -- str (nil)
23:32:22:462 ~ResourceItem() attr/uniqueid -- str 0x7005f0
23:32:22:462 ~ResourceItem() state/lastupdated -- str 0x700dc0
23:32:22:462 ~ResourceItem() config/configured -- str (nil)
23:32:22:462 ~ResourceItem() config/sunriseoffset -- str (nil)
23:32:22:462 ~ResourceItem() config/sunsetoffset -- str (nil)
23:32:22:462 ~ResourceItem() config/lat -- str 0x700dc0
23:32:22:462 ~ResourceItem() config/long -- str 0x700dc0
23:32:22:462 ~ResourceItem() state/daylight -- str (nil)
23:32:22:462 ~ResourceItem() state/status -- str (nil)
23:32:22:463 ~ResourceItem() attr/name -- str 0x5b0180
23:32:22:463 ~ResourceItem() attr/modelid -- str 0x5c81d8
23:32:22:463 ~ResourceItem() attr/type -- str 0x5c3230
23:32:22:463 ~ResourceItem() config/on -- str (nil)
23:32:22:463 ~ResourceItem() attr/uniqueid -- str 0x5bd7a8
23:32:22:463 ~ResourceItem() state/lastupdated -- str 0x5b13c8
23:32:22:463 ~ResourceItem() config/configured -- str (nil)
23:32:22:463 ~ResourceItem() config/sunriseoffset -- str (nil)
23:32:22:463 ~ResourceItem() config/sunsetoffset -- str (nil)
23:32:22:463 ~ResourceItem() config/lat -- str 0x704318
23:32:22:463 ~ResourceItem() config/long -- str 0x5cc460
23:32:22:463 ~ResourceItem() state/daylight -- str (nil)
23:32:22:463 ~ResourceItem() state/status -- str (nil)
23:32:22:463 sql exec SELECT * FROM gateways
23:32:22:464 don't close database yet, keep open for 900 seconds
23:32:22:464 Daylight now: nightStart, status: 230
23:32:22:464 ~ResourceItem() attr/name -- str 0x5aa258
23:32:22:464 ~ResourceItem() attr/name -- str 0x5a6af0
23:32:22:464 ~ResourceItem() state/all_on -- str (nil)
23:32:22:464 ~ResourceItem() attr/name -- str 0x5a5230
23:32:22:464 ~ResourceItem() state/all_on -- str (nil)
23:32:22:464 ~ResourceItem() state/any_on -- str (nil)
23:32:22:464 ~ResourceItem() action/scene -- str 0x5a5230
23:32:22:464 ~ResourceItem() attr/name -- str 0x5a8f28
23:32:22:464 ~ResourceItem() state/all_on -- str (nil)
23:32:22:464 ~ResourceItem() state/any_on -- str (nil)
23:32:22:464 ~ResourceItem() action/scene -- str 0x5ae940
23:32:22:464 ~ResourceItem() attr/type -- str 0x5a5230
23:32:22:464 ~ResourceItem() attr/class -- str 0x5a5230
23:32:22:464 ~ResourceItem() attr/name -- str 0x6fe4a0
23:32:22:464 ~ResourceItem() state/all_on -- str (nil)
23:32:22:464 ~ResourceItem() state/any_on -- str (nil)
23:32:22:465 ~ResourceItem() action/scene -- str 0x6fe700
23:32:22:465 ~ResourceItem() attr/type -- str 0x6fe340
23:32:22:465 ~ResourceItem() attr/class -- str 0x6fde10
23:32:22:465 ~ResourceItem() attr/name -- str 0x5d1ab8
23:32:22:465 ~ResourceItem() state/all_on -- str (nil)
23:32:22:465 ~ResourceItem() state/any_on -- str (nil)
23:32:22:465 ~ResourceItem() action/scene -- str 0x5c3b10
23:32:22:465 ~ResourceItem() attr/type -- str 0x6fb048
23:32:22:465 ~ResourceItem() attr/class -- str 0x6fcd18
23:32:22:465 <?xml version="1.0"?>
23:32:22:465 <root xmlns="urn:schemas-upnp-org:device-1-0">
23:32:22:465   <specVersion>
23:32:22:465     <major>1</major>
23:32:22:465     <minor>0</minor>
23:32:22:465   </specVersion>
23:32:22:465   <URLBase>http://192.168.1.71:8080/</URLBase>
23:32:22:465   <device>
23:32:22:465     <deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType>
23:32:22:465     <friendlyName>Philips hue (192.168.1.71) compatible Wireless Light Control Gateway</friendlyName>
23:32:22:465     <manufacturer>Royal Philips Electronics</manufacturer>
23:32:22:465     <manufacturerURL>http://www.dresden-elektronik.de</manufacturerURL>
23:32:22:465     <modelDescription>dresden elektronik Wireless Light Control</modelDescription>
23:32:22:465     <modelName>Philips hue bridge 2012</modelName>
23:32:22:465     <modelNumber>929000226503</modelNumber>
23:32:22:465     <modelURL>http://www.dresden-elektronik.de</modelURL>
23:32:22:465     <serialNumber>0000000000000000</serialNumber>
23:32:22:465     <UDN>uuid:35a54a66-430a-4302-9827-5e4198c5ff60</UDN>
23:32:22:465     <gatewayName>zigbee_gw</gatewayName>
23:32:22:465     <presentationURL>index.html</presentationURL>
23:32:22:465     <iconList>
23:32:22:465       <icon>
23:32:22:465         <mimetype>image/png</mimetype>
23:32:22:466         <height>48</height>
23:32:22:466         <width>48</width>
23:32:22:466         <depth>24</depth>
23:32:22:466         <url>hue_logo_0.png</url>
23:32:22:466       </icon>
23:32:22:466       <icon>
23:32:22:466         <mimetype>image/png</mimetype>
23:32:22:466         <height>120</height>
23:32:22:466         <width>120</width>
23:32:22:466         <depth>24</depth>
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()23:`

If you guys need any more info, please ask me. 

Thanks SO MUCH for your help.

manup, 2018-05-22:
If I remember correctly there were some unresolved issues with Hassio. I can only guess and think there might be colliding serial port configuration in Hassio which needs to be fixed.

josephyancey, 2018-06-20:
I had this same problem and was able to solve it by disabling bluetooth on the pi. Here is a really good summary: https://community.home-assistant.io/t/marthoc-deconz-addon-cant-connect-lights/57143/12

closed 2019-03-30 by stale[bot]