dresden-elektronik/deconz-rest-plugin #594: Beta update 27 changed port?

Snailboy2, 2018-05-18:
Heya.

I just updated to beta 27 from 24 and all my lights got removed and now the port is changed from 8080 to 80 so now my pihole dont work. 
How do i change back to 8080? and why did the update remove all lights? Some sensors are still there. 
Is there any changelog for the updates? 
Would be nice to know if an update does major changes. 

Got a Deconz usb stick on Raspberry pi 2 running Raspbian.

Snailboy2, 2018-05-18:
Uhum after a cuuple of restarts its now back to port 8080 but it wont find any of my Ikea or Innr lights for some reason.

ebaauw, 2018-05-19:
deCONZ defaults to port 80 when run as root, or with the right privileges to use ports < 1024; otherwise it defaults to 8080.  To change this, you can specify --http-port=xxxx when staring deCONZ.

Did deCONZ lose all devices, or only some?  If all, check the deCONZ network settings, and reset the ConBee / power-cycle the pi; if some, try power-cycle the missing devices.

Snailboy2, 2018-05-25:
Aa thats why. Thanks.
It lost all lights after the upgrade and after downgrading to stable .23 i could add lights again and sensors start working after re-ading them. So now it works as it should.

closed 2018-05-25 by Snailboy2