dresden-elektronik/deconz-rest-plugin #599: Farbwechsel/Color Change

joecas1, 2018-05-19:
Ich vermisse die Option Farbwechsel in Verbindung mit Phoscon und FLS-PP! Gibts das noch nicht, oder finde ich es nur nicht?

I‘m missing the option to enable the color change in Phoscon together with FLS-PP! Is there nothing like this, or I‘m not able to find it?

manup, 2018-05-22:
The colorloop is currently not integrated in the Phoscon App. It will first reappear in the generic switch editor. Maybe later on also in the light controls.

Heres a internal screenshots how it may look like in the switch editor.

[image]

davidreher, 2019-02-06:
@manup any updates on this?

manup, 2019-02-06:
>@manup any updates on this?

Enabling color loop can be done with help of a switch as mentioned above. This is part of recent versions.

bubffm, 2019-05-31:
I don't have a switch and dont intend ti buy one. Is there any chance to trigger such color change / color loop feature using fake switches aka in homebridge?  Thanks

closed 2019-10-05 by stale[bot]