dresden-elektronik/deconz-rest-plugin #600: Reach old web app

Snailboy2, 2018-05-19:
How do i reach the old web app?
When i go to http://ip:8080 it redirects me to http://ip:8080/pwa for some reason.

I need it because for some reason i couldn't find my Ikea lights in pwa but in old web app it works.

Snailboy2, 2018-05-19:
After a reset i now can add lights. Yey

closed 2018-05-19 by Snailboy2

manup, 2018-05-20:
The old webapp is also reachable via *Phoscon > Help > Open Old WebApp*.

johansvenson, 2018-08-14:
I am afraid that in 2.05.34 you are simply redirected to the PWA application - also tried by specifing index.html as well as index.html?v=2.05.34 to no avail.

manup, 2018-08-14:
Did you try it with the link from the *Phoscon App > help > Open Old WebApp*?
Works here.

You can also open the old WebApp via `http://<ip-address>/login.html`

johansvenson, 2018-08-14:
Yeps i did... got redirected in all cases... but updated from .20 to .34  and now that part is fine (needed to enable OTAU for which i could not find the checkbox in the pwa app)

OrbitingOcelot, 2020-08-03:
I think this should be reopened, the "old web app" link does not work thru ingress.

Mimiix, 2020-08-03:
@OrbitingOcelot this is something that the HA addon should fix, we can't fix it.