dresden-elektronik/deconz-rest-plugin #603: Ikea lights not reachable

sweebee, 2018-05-20:
I tried to reset a ikea light and re-add it. Deconz does find it but 5s later its not reachable anymore. I can try that 100 times and its always the same issue.

sweebee, 2018-05-20:
I fully resetted the gateway and added every light again. Lights that I could not add before I could add now. But 1 light that worked fine before I can't add right now (tradfri).

All these problems only exists on tradfri lights.

sweebee, 2018-05-21:
I gave it another try, without success. I have now 1 bulb and 4 gu10 spots from tradfri. No way I can addd them to deconz anymore. I tried to reset them with touchlink.

sweebee, 2018-05-21:
I went to my parents house, I also installed a raspbee there. I tried to add the lights there, and I could add them on the first try. So I think there is something wrong with deconz <> tradri combination.

ebaauw, 2018-05-21:
> So I think there is something wrong with deconz <> tradri combination (sometimes)

Trådfri seems to have a funny way to build up the ZigBee mesh network, which deCONZ doesn't always appreciate.  You might want to try and power off all lights already connected to the RaspBee, and move to light to be connected close.  This will force it to connect directly to the RaspBee, instead of through other Trådfri lights.

If that still doesn't work, while the other lights are still powered off, try resetting the RaspBee, by power-cycling the Raspberry, or by stopping deCONZ and issuing `sudo GCFFlasher_internal -r`.  This will clear the RaspBee's neighbour table, in case it thinks it has too many direct connections already.

sweebee, 2018-05-21:
No luck with the GU10 spots. I re-added them to the tradfri bridged and removed them again so they reset, but deconz won't find them.

manup, 2018-05-24:
The latest deCONZ firmware version contains a few route optimizations, it may help a bit.
Please also make sure that your IKEA lights do have the latest firmware.

sweebee, 2018-05-25:
Updated to the latest beta version. Resetted all the gu10 spots with powertoggling it 5 times, but still can't add them to the gateway.

manup, 2018-05-25:
It's toggling 6 × times quite fast, have a look to this video showing it:

https://www.youtube.com/watch?v=mJm9YpPrGzk

sweebee, 2018-05-25:
yes I did the following:

toggle it 6 times, after that the light flashed to 100% brightness.

set deconz in search/include mode

power toggle the light once again.

but it doesn't show up.

manup, 2018-05-25:
set deconz in search/include mode, must be the first step

IKEA lights search only once directly after reset (or further power cycle).

sweebee, 2018-05-25:
Also tried that:

enable search/include
powercycle 6 times / reset
nothing shows up

this happens for all my 4 gu10 spots.

While yesterday I bought a tradfri e14 and it worked immediately. 

I think I have to try to add it to my parents' raspbee, that also worked for a tradfri e27 bulb that I could not find anymore on my raspbee. But they dont have a gu10 socket, but thats a different problem.

manup, 2018-05-25:
Hmm weird, I've 5 IKEA GU10 tunable white running, but added them a few months ago.
I'll check later tomorrow to reset and re-add one and maybe I find something to reproduce the issue.

ErnstEeldert, 2018-06-05:
I have the exact same issue with various TRÅDFRI lights. They just refuse to pair, while they were connected previously. Pairing with ikea gateway works just fine though.

In general, I wouldn't consider deconz very usable with ikea lights. It seems like every week there is a new issue of some sort, requiring resets, repairs, rule updates or something. Perhaps it's due to my mesh size? Just over 60 devices, which shouldn't be too crazy for a 4 bedroom home?

mickejo, 2018-07-26:
I have the same issue. Some lights that has been connected to my Raspbee before refuse to show up as lights in deConz after a gateway reset. What I noticed is that they seem to be part of the mesh. I tried pairing a light with an IKEA remote, then doing the touchlink scan seeing the light in the list, after touchlink reset while the gateway is in open network mode, the light changes network and channel to my raspbee network and channel, but the light does not show up in Phoscon nor in the old "Wireless Light Control" nor in the Rest API. I tried changing the channel for my network and the light changes channel, so it is in the network but it just does not show up. I have tried repeating adding the light to the IKEA remote and then trying to readd it to the raspbee but the same result. 

I had issues with one of my 23 lights bahaving like this so I tried to reset the gateway to readd everything. Now I have six lights of the 23 that behave in this way, not exactly what I hoped for..

manup, 2018-07-27:
Can you please provide the deCONZ and firmware version you are using (visible in Phoscon App > Menu > Gateway).

mickejo, 2018-07-27:
Version 2.05.32 / 2018-07-11
Firmware 26210500
I have also tried 2.05.25/261f0500 with the same result.

I realized that all lights are present in the devices table in zll.db, but the missing lights are not in the nodes table.

closed 2018-12-05 by stale[bot]

8engan, 2019-02-03:
Had a similar issue and found this issue and figured Id add what worked for me.
All of a sudden 4 of my GU10 spots stopped reporting, tried resetting (power toggle 6 times) but every time when trying to add them again they appeared as the old name but still no connection, showing as grey in phoscon and with no links in deconz.
After a lot of trouble shooting I figured my next step would be to try to start all over again but as a last step I tried removing ligths from phoscon, then go to lights -> Add new light, and while listening for new devices I reset the ikea lights again (power cycle 6 times) and voila - they reappeared and are again controllable for me :)