dresden-elektronik/deconz-rest-plugin #610: Firmware 26210500 strange behaviour

stefan1957, 2018-05-25:
Hello,
The firmware 26210500 from the beta 2.05.28 is not working as expected with homebridge-hue v0.9.3. Only 40 of the 51 nodes work in HomeKit. In the phoscon app is everything working normally. I ll try to downgrade to 2.05.27 with the same result. So I think that there is something wrong with the firmware. I hope you can fix this
Gr Stefan

ebaauw, 2018-05-25:
I upgraded to v2.05.28 and 0x26210500 this morning.  One of my IKEA lights would only be found by deCONZ after power-cycling it, but otherwise I've seen nothing weird.  Not even sure if the IKEA light issue is related to the upgrade.

If things are working normally in the Phoscon app, I don't think there's an issue with deCONZ.  Not sure what you mean by "work in HomeKit".  Best to open an issue in homebridge-hue, include debug log file and debug dump file.

alexvandervegt, 2018-05-25:
@manup When will the new firmware be uploaded to:

http://dresden-elektronik.de/rpi/deconz-firmware/?C=M;O=A ?

manup, 2018-05-25:
Firmware is part of the deCONZ package, you can find in in `/usr/share/deCONZ/firmware`.

stefan1957, 2018-05-26:
@manup 
i downgrade to deCONZ_Rpi_0x261f0500.bin.GCF and my problems are solved.  So i still think, that in my case, the firmware 0x26210500 isn't oke. Btw the phoscon app is also unstable with the new firmware.

alexvandervegt, 2018-05-27:
I have a lot xiaomi wireless switches which lose the network since the upgrade

sberryman, 2018-05-30:
I just upgraded and wished I hadn't... My system is no longer pushing updates from (hue motion) sensors. They are all there and show up with the correct data for polling but nothing updates over the websocket

ebaauw, 2018-05-30:
I noticed that after recalling a scene, the light states aren’t updated.  Also, my Hue motion sensors blink red when detecting motion, and there seems to be a slight delay, before deCONZ reacts to these.  I won’t be able to test until this weekend.

@sberryman you can downgrade the firmware using GCFFlasher_internal and you can downgrade deCONZ by simply installing an older version.

manup, 2018-05-30:
@sberryman do the sensors blink blue from time to time or is there no communication at all?
I've running the latest firmware version in two setups with hue motion sensors, here they work &mdash; I need to do more tests, also with the IKEA join problems.

ebaauw, 2018-06-01:
I got myself an OSRAM Gardenspot mini RGB for my terrace.  It's on the edge of my ZigBee network.  It already had a regular (dimmable) Trådfri bulb at the Terrace.  I upgraded the Gardenspot's firmware (using the OSRAM gateway), paired it to deCONZ and tested it in my Living Room.  I restarted deCONZ (to change the number of the light resource) and installed the Gardenspot.  Neither the Trådfri nor the Gardenspot would react to unicast commands, but both continued to react to group commands.

I downgraded the RaspBee firmware to 261f and both are working now.  Also the Hue motion sensors no longer blink red, and the delays seem to have gone.  I'm still seeing the issue that recalling a scene won't update the light states, however.

manup, 2018-06-01:
Interesting, the 0x26210500 firmware has maximum route optimisations enabled. That is routes (to lights) are created also from incoming commands without former route requests. I have testet it in a larger network with 80 lights and heavy routing, there it worked fine and very fast (but the network only contained dresden elektronik and hue lights). 

I'm not sure why it breaks with routing involved by OSRAM and IKEA lights, normally if a route doesn't work the in between routing node should respond with a 'no route' NWK message so that the normal routing procedure can start as with 0x261f0500 firmware.

My home network has mixed OSRAM and IKEA lights but no routing since all lights are connected directly to RaspBee, bad for testing :)

I'll need to rearrange some lights and put metal around them to force routing, hopefully the sniffer will show where the issue is.

ebaauw, 2018-06-10:
> I noticed that after recalling a scene, the light states aren’t updated.
> I'm still seeing the issue that recalling a scene won't update the light states, however.

The corresponding code was commented out?!  I undid the commenting and now the light states are updated again.

djwlindenaar, 2018-07-01:
So what's the verdict on using 26210500? I've just updated deconz to the latest beta .31, but I'm hesitant to do a firmware upgrade given the mentioned issues. I've got IKEA lights and Xiaomi plugs and switches in my network. Should I stay with 261f or not? 

Regards
Daniel

djwlindenaar, 2018-08-26:
Well... I went ahead and updated anyway. My verdict is that it does not work. The network initially functions well, but soon endpoints and even routers loose connectivity. Often endpoints don't get disconnected, but the first report gets lost. E.g. a button press is missed, but a second try comes though. 

Anyway, I'm back to the much more stable 261f and all is fine again.

closed 2018-12-31 by stale[bot]