dresden-elektronik/deconz-rest-plugin #613: Xiaomi Aqara (lumi.sensor_86sw2) appearing as Smart Plug/Light in REST API

saurabh984, 2018-05-27:
Every time I search for lights an Unknown Smart Plug is appearing in the list. From MAC Address it looks like it's Aqara Remote Wall Switch Double appearing as Smart Plug.

Started happening since v2.05.28

`    "5": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true,
            "temperature": 2100
        },
        "ep": 1,
        "etag": "1f4f5f8ff9a16c2c39cd15c104bc2c81",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.sensor_86sw2",
        "name": "Hallway Front Switch",
        "state": {
            "buttonevent": 1002,
            "lastupdated": "2018-05-24T22:14:21"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:01:f4:0e:0c-01-0006"
    }`

`"12": {
        "etag": "b4493f0c411ac685da96976cfda4a1fa",
        "hascolor": false,
        "manufacturername": "Unknown",
        "modelid": null,
        "name": "Light 12",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "Smart plug",
        "uniqueid": "00:15:8d:00:01:f4:0e:0c-03"
    }`

ebaauw, 2018-05-27:
I suspect, somehow, deCONZ sees device type 0xffff on endpoint 3 of your wireless switch, instead of 0x5f03.  This causes deCONZ to think it's a wired in-wall switch, and create the corresponding light resource.  
I don't know why deCONZ sometimes sees different device types.  I suspect you'll see the proper device type 0x5f02 on endpoint 2, or deCONZ would create a light resource for that endpoint as well.

Structurally, we need a better way of identifying the Xiaomi in-wall switches and plugs.  In the mean time, you might try and re-pair the switch in the hope that deCONZ reads the correct device types.  This might take several attempts, see #543.  Be sure to reset the switch, delete the node, delete the resources, and delete physically the records from the database, before re-pairing the switch, or deCONZ will simply restore the previous resources.

saurabh984, 2018-05-28:
Thank you @ebaauw 

Will try that.

closed 2018-05-28 by saurabh984

GatoPharaoh, 2018-06-16:
@ebaauw how would I `delete the resources, and delete physically the records ` ?
I'm having trouble with a `lumi.sensor_motion_aq2 `which deconz only exposes with temperature and lux but no motion - which also leads to `homebridge-hue` only exposing lux but no change of motion