dresden-elektronik/deconz-rest-plugin #614: Lost Hue and Tradfri lights

kylegordon, 2018-05-29:
Hi,

I had reason to restart my home automation server (on deconz 2.05.20), and following the restart there's 2 of 3 Hue bulbs not present, and 6 Tradfri bulbs missing. Previously they had all reliably been connected. 

Upon starting deCONZ, I get the following errors repeatedly...

23:39:01:880 neigbor 0x0017880100d3e3b6 is unknown child
23:39:07:131 neigbor 0x90fd9ffffe747bf7 is unknown child

I suspect they are critical bulbs in the mesh leading towards the rest of the missing bulbs.

Using the app is a pain due to it running remotely on my x64 server, needing X, and my laptop running Wayland. Is there a way of trusting the above IDs without having to extract the bulbs from the ceiling again?

kylegordon, 2018-05-29:
I'm so confused. I can use Phoscon to turn the my two groups on and off, and all the bulbs respond. However, the groups don't have any bulbs listed in them, except for my one remaining working bulb.

How can I have working groups with no bulbs listed in them? They're also all showing as unavailable in Home-Assistant, bar the one working bulb.

ebaauw, 2018-05-30:
In ZigBee, a group is more like a multicast address than an object.  Group membership is stored in the bulbs.  The bulbs are still in the ZigBee network, and react to group commands.

The problem lies with the RaspBee not discovering the bulbs, and consequently the REST API plugin not creating the light resources.  You might try and power-cycle the bulbs.

If the mountain doesn’t move to the prophet...  For resetting/re-pairing, I power my Raspberry from a USB powerbank and hold it next to the bulb, rather than moving the bulb.  If you have a ConBee, you might use a USB extention cable to hold the ConBee next to the bulb.

kylegordon, 2018-05-30:
@ebaauw thanks for the info. Certainly makes more sense now :-)
I've tried power cycling (sometimes leaving off overnight) to no avail. 

The layout of the house is that there's 2, maybe 3, Hue bulbs in range of the Conbee, and the 6 Tradfri are at the end of the 'Hue corridor'. I can't remember the graph precisely, but I think the Tradfri relied on the Hue bulbs as mesh routers.

My Conbee runs from my HP Microserver, and no USB cable in the world will be long enough :-) Is it possible to copy the deCONZ database/config/etc to a laptop, take the Conbee and a laptop to the lights, and then port the database and config back?

closed 2018-06-05 by kylegordon