dresden-elektronik/deconz-rest-plugin #616: DeCONZ running, but no gateway

QempZor, 2018-05-30:
I just installed DeCONZ on my new Raspberry Pi 3b+ 
However, I only see a "Daylight" node showing up.
PWA/WebAPp are working, but im not able to connect one of my ZIgbee devices to the Raspbee.
It looks like there is something missing

manup, 2018-05-31:
Does the blue coordinator node show up in deCONZ? 

Also note that on a vanilla installation you need to configure the serial port as described here:
https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

QempZor, 2018-05-31:
Just found out that Deconz wasn't reading the RaspBee macadress and stuff.
Just added the mac address, TC address and TC Link adress in DeCONZ interface, now i'm able to pair my devices again :)

However, it's stranged. Never had it before when I installed a new RaspBee..

closed 2018-06-05 by QempZor