dresden-elektronik/deconz-rest-plugin #617: After Version 2.05.25 no Working with IKEA an Lumi Human Sensors

mark82436, 2018-05-31:
All Version after 2.05.25 (x.26/x.27/x.28) not Working right
with Ikea and Lumi Human Sensors.
Lights are switch on after aktiv ir but light go not off the
Sensor are on 100% fix and go not on timeline 

rollback to version 2.05.25 are working fine

manup, 2018-05-31:
This was fixed in version 2.05.28 but requires manually "touching" the sensor control one time.

See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/590

mark82436, 2018-05-31:
sorry it's not fixed i have reset all sensors and restart deconz
one sensor have it delete and new bind but it is not working 
right and i must rollback to version 2.05.25

manup, 2018-05-31:
You don't need to reset anything, in the Phoscon App motion sensor control you need to modify a  parameter like dimm value (for example from 100% to 99%), then the underlying rules should be recreated.

I see to automate this in the next versions.

mark82436, 2018-05-31:
ok i test it again
but is not so nice i have 15 sensors with 20 profiles

mark82436, 2018-06-01:
i have test again but is it not working
some sensors working and some sensors are not working
i have delete the sensor control set and make a new one it's the same error

i rollback to 2.05.25

mark82436, 2018-06-13:
Same problem with Version 2.05.29

ROLLBACK TO 2.05.25

manup, 2018-06-14:
Found the root problem, rule condition operator *ddx* was corrected a while ago in the REST-API. It is currently used to provide the delayed dimm behavior which doesn't work at all times. I figure the *stable* operator is a better fir to provide the needed functionality.

Not implemented yet, on it ...

ebaauw, 2018-06-14:
> It is currently used to provide the delayed dimm behavior which doesn't work at all times. I figure the stable operator is a better fir to provide the needed functionality.

Out of interest: what is the issue, exactly?  Is it a problem with the `ddx` implementation (the changes to `ddx` expect `/config/localtime` to be considered for potential rule firing every second), or does the desired logic simply not match the `ddx` functionality?

manup, 2018-06-16:
The logic as it was implemented in Phoscon App expected `ddx` to trigger for times *>=* the specified operator value. With the latest changes `ddx` will only fire on the exact second and only once, which is correct I guess. However I'm not sure if the fire once per second will always hit the `ddx` value, for example if database is saved or any other heavy task consumes more time, the exact second might be missed.

manup, 2018-06-16:
Depends how the rule is designed and supposed to work. in the app we use them to control a state machine and various virtual sensors.

The `stable` operator can be used as rule condition to check if a value hasn't changed in a specified time. For example if *presence* was false for the last 30 seconds &mdash; *"value": "PT00:00:30"*.

saurabh984, 2018-06-16:
Thank you @manup that’ll do nicely!

saurabh984, 2018-06-16:
```
{
        "error": {
            "address": "/sensors/15/state/status",
            "description": "Condition error",
            "type": 607
        }
    }
```

```
{
                "address": "/sensors/15/state/status",
                "operator": "stable",
                "value": "PT00:00:30"
            }

```

What am I doing wrong here?

saurabh984, 2018-06-16:
Sorted! Had to restart deCONZ after installing 2.05.30

closed 2019-03-30 by stale[bot]