dresden-elektronik/deconz-rest-plugin #619: latest release breaks Sensor Control

ErnstEeldert, 2018-06-01:
I was using Sensor Control with Xiaomi motion sensors. It somewhat worked before, but with the last release .27 the lights go on, but never turn off. I've deleted and recreated the sensor control setup to no avail.

manup, 2018-06-01:
There was a breaking change in 2.05.26 rules API triggering logic, the sensor control was fixed in 2.05.28, but requires manually recreation of rules like:

>However in order to fix existing rules, the sensor control needs to be changed slightly to trigger the recreation of the rules, this can be done by any change — like changing brightness from 100% to 99%.

I'll see to automate the fix in the next versions.

Related issues: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/618 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/617 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/590

ErnstEeldert, 2018-07-01:
I am still experiencing this issue on Version 2.05.28 / 5/3/2018 Firmware 261F0500 it seems.

So the lights turn on, but intermittently never turn off based on the sensor control.

I noticed the sensor control count being two, where I can only see one if I drill down into the sensor control screen. If I delete that single sensor control, the counter goes to 1, and the screen shows up empty. I checked the /rules endpoint on the API, and nothing seems to be referring to that group, so not sure where the 1 is coming from. Perhaps it's related to the rules not behaving correctly?
[image]
[image]

closed 2019-03-30 by stale[bot]