dresden-elektronik/deconz-rest-plugin #621: Get manufacturer specific data over REST-API

MiroBudja, 2018-06-05:
How to get attributes from manufacturer specific data over the REST-API?
Should I use `bool DeRestPluginPrivate::readAttributes(RestNodeBase *restNode, quint8 endpoint, uint16_t clusterId, const std::vector<uint16_t> &attributes, uint16_t manufacturerCode)` method to get the data? Or should I use any other way? Maybe later also the `updateLightNode` method to update the LightNode? Are there maybe some examples?

Thanks!

ebaauw, 2018-06-05:
That's not the REST API, but the interface that deCONZ provides to plugins (like the REST API plugin).

But, yes, that's the function I used to read the manufacturer-specific attributes of the Hue motion sensor:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/60491c02216fdcf620c8c418153aea6a788de19c/de_web_plugin.cpp#L10972

MiroBudja, 2018-06-06:
My goal is to achieve to GET data which are manufacturer specific over JSON
`http://IP/api/apikey/lights/lightid/manufacturerspecificaddressextension`

When I "GET" this requested URL, I like to read (probably with `readAttributes`) and also show this manufacturer specific data as JSON response (QVariantMap).

How to get this data from `readAttributes` to `someMap["showdata"] = isearchthisdata`

Can I get standard CASE how to get data from manufacturer specific cluster. Thanks and sorry for my poor knowledge about this code

ebaauw, 2018-06-06:
> When I "GET" this requested URL, I like to read (probably with readAttributes) and also show this manufacturer specific data as JSON response (QVariantMap).

That’s not how the REST API plugin works (or, for that matter, any other ZigBee gateway I’ve seen).  When doing a GET, the data is returned from the plugin’s cache.  The communication over the ZigBee network happens asynchronously, either through polling the ZigBee devices, or through attribute reporting, where the device pushes its state to the gateway.

You would have to:
- Create resource attributes for the manufacturer-specific attributes;
- Extend the lights API GET to return these resource attributes;
- Update the resource attribute values by extending the light polling to include the manufacturer-specific attributes, or by extending the light discovery to setup attribute reporting for the manufacturer specific attributes and extending the handling of these reports.

MiroBudja, 2018-06-07:
Yes, I have already implement manufacturer specific attributes, extend the lights API GET only what I don't understand or I don't find the correct way to implement is how to update manufacturer specific resource values.

I should use readAttributes with manufacturer specific vendor ID and cluster ID and then what? updateLightNode method maybe? Please some examples if possible, thanks!

Sorry for misunderstandings. Thanks!

ebaauw, 2018-06-08:
Indeed, `readAttributes()` only queues a ZigBee command and the response is handled in `updateLightNode()` and/or `updateSensorNode()`.  Basically they loop over the light's or sensor's endpoints (`i`), clusters (`ic`), and attributes (`ia`), check whether they're present in the response (`event`), find the corresponding resource attribute (`item`), update the value, and queue an event (for web socket notification and rule triggering).  Make sure to define the cluster and attribute in the ZCLDB (default `/usr/share/deCONZ/zcl/general.xml`) or they won't be present in the light or sensor node.

closed 2019-03-30 by stale[bot]