dresden-elektronik/deconz-rest-plugin #622: Xiaomi HoneyWell smoke sensors low battery

QempZor, 2018-06-05:
Xiaomi Smoke detectors are reporting Battery 0% (Low battery in Homekit)
However, in the REST API config:battery is null;
[image]
and DeCONZ reports Power Level 100%
[image]

ebaauw, 2018-06-05:
The battery percentage in `config.battery` is read from the _Power Configuration_ cluster, not from the _Power Descriptor_ (which, I think, isn't even read at all - the sensor sure isn't mains powered).

QempZor, 2018-06-05:
That is kinda weird, it's battery powered, indeaad. So the Power Descriptor would not be read I guess.
My other Xiaomi sensors, like the door/windows, floodsensors, and buttons are reporting 100%
Wich I also can't bbelieve, because the running for over 6 months.. and didn't lose any %

wvuyk, 2018-06-05:
Could be possible though. Xiaomi states two years for most battery devices. Here I have had Philips motion sensors reporting 100% for over an year. Only now the Philips motion sensors start reporting 53% after more then 1 1/2 year of usage. Would not be surpirsed if Xiaomi will report 100% for some time more?

BigWumpus, 2018-06-08:
Wait some days for a readout

closed 2019-03-30 by stale[bot]