dresden-elektronik/deconz-rest-plugin #623: Centralite

jonathanweinberg, 2018-06-05:
I've noticed in issues #404 / #286 / #229 mention of Centralite / SmartThings motion sensors working.

I have a number of the Lowe's Iris variants, slight built difference, and different model number. I'm attaching some screenshots that might be useful. Please let me know if I need to gather more info!

[image]
[image]
[image]
[image]
[image]
[image]

dbmet, 2018-09-11:
Did you ever get these working?

AVanVlack, 2018-11-10:
I have gotten this working, along with the contact sensor (3320-L), although my motion sensor reports as 3326-L. Not sure what the difference is on the models but they seem to report all the same attributes. I added them both to the device list and opened the network through phoscon app and it started working. 

Both the temperature and motion is working but for web-sockets I did have to set config.duration to 0 or the rest-plugins timer starts doing odd things. Also there is no way to change the no-motion delay, the hardware is set to end presence 30 second after no motion . This is my first time working with an IAS device. I am more use to ZLL/Ikea motion sensors and how they work.

Also the buttons on both devices are only setup for config/factory reset but are set as the IAS tamper alarm. Not sure if this is an issuse or common with the centralite devices.

I still have all those unknown clusters that I am hoping to find some kind of delay/threshold settings as there seems to not be a 0406 Occupancy Sensing cluster. I will make a pull request when I am sure everything is how it should be.

dbmet, 2018-11-10:
Can you tell me more how you did this ?   I added them both to the device list

dbmet, 2018-11-10:
Can you tell me more how you did the following

Added them to the device list and set config.duration to 0

Thanks 

> I have gotten this working, along with the contact sensor (3320-L), although my motion sensor reports as 3326-L. Not sure what the difference is on the models but they seem to report all the same [...]

AVanVlack, 2018-12-02:
Can someone point me to the function that writes the IAS_CIE_Address attribute to IAS devices during setup. I have everything working when setting these device up except for that. I am seeing zone enrollment in sendIasZoneEnrollResponse but the the CIE never gets written so the device never requests zone enrollment.

manup, 2018-12-02:
Currently this isn't be done automatically since all other IAS sensors did work without that, so we need to add some code to write it fir specific sensors like the Centralite. I'll see to add it in the next versions.

dbmet, 2018-12-19:
Just curious if these are fully supported yet ?

blueleader07, 2018-12-23:
FYI, I got the Centralite Temperature sensor to appear in the REST API by following the steps posted above:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/623#issuecomment-437649093

Two things that still don't seem to work:
1. The sensor doesn't appear in the Phoscon App
2. The battery level is being reported as null

Values I added to de_web_plugin.cpp around line 98 with the other centralite devices.

```{ VENDOR_CENTRALITE, "3310-G", emberMacPrefix }, // Centralite Temperature```

... and added these lines to bindings.cpp in the whitelist after line 1381:

```sensor->modelId().startsWith(QLatin1String("3310")) ||```

As it is though just having the temperature returned by the REST API should do the trick for me.  I've been trying the Mozilla Things gateway (UI) along with the Zigbee Conbee Adapter (polls the REST API) and it works very well.

closed 2019-05-23 by stale[bot]