dresden-elektronik/deconz-rest-plugin #624: Power usage on Osram Smart+ doesnt give correct data

roflmao, 2018-06-05:
```
Version 2.05.28 / 5/3/2018
Firmware 260B0500 
```

Osram Smart+: https://www.osram-lamps.com/ecatalog/smart-home/smart-home-components/smart-plug/index.jsp

Output from REST:
```
$ curl -s 192.168.66.79/api/6D0XXXX545/sensors | jq .
{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "10f8a4da2598d615d5515b75a3e671ab",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": false,
      "daylight": true,
      "lastupdated": "2018-06-05T11:16:59",
      "status": 170
    },
    "swversion": "1.0",
    "type": "Daylight",
    "uniqueid": "00:21:2e:ff:ff:02:2c:f8-01"
  },
  "2": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 3,
    "etag": "d546a684a755dd5702bda05d9f71c655",
    "manufacturername": "OSRAM",
    "modelid": "Plug 01",
    "name": "Power 2",
    "state": {
      "lastupdated": "2018-06-05T16:14:28",
      "power": 40
    },
    "swversion": "V1.04.12",
    "type": "ZHAPower",
    "uniqueid": "84:18:26:00:00:0d:23:6a-03-0b04"
  }
}

```

The lastupdated is always updated accordingly to the current time (UTC).
The power however doesnt update when i turn my tv on/off by remote.
If i turn off the switch it updates it:

```
{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "10f8a4da2598d615d5515b75a3e671ab",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": false,
      "daylight": true,
      "lastupdated": "2018-06-05T11:16:59",
      "status": 170
    },
    "swversion": "1.0",
    "type": "Daylight",
    "uniqueid": "00:21:2e:ff:ff:02:2c:f8-01"
  },
  "2": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 3,
    "etag": "db61fa548a997f685db444eeef53c25b",
    "manufacturername": "OSRAM",
    "modelid": "Plug 01",
    "name": "Power 2",
    "state": {
      "lastupdated": "2018-06-05T16:16:14",
      "power": 0
    },
    "swversion": "V1.04.12",
    "type": "ZHAPower",
    "uniqueid": "84:18:26:00:00:0d:23:6a-03-0b04"
  }
}
```

If i connect a 2000watt heating fan it still shows as power: 40 (lastupdated is also updated and the switch is turn on/off)

Any debug info i can collect?

ebaauw, 2018-06-05:
This is a bug in the OSRAM firmware, see: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#supported-smart-plugs--switches.

roflmao, 2018-06-05:
Thats too bad :/

ebaauw, 2018-06-05:
Yeah, I'm still trying to find the ultimate EU smart plug, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/432.
- The measurement in the OSRAM plus doesn't work, and they're big;
- The Heiman plug supports measurement, but doesn't support groups and some folks have units making an irritable noise.  Also pretty big;
- Xiaomi seems perfect from a ZigBee perspective but needs an additional adapter, since it only comes with Asian pins;
- Just ordered the new innr plug, which finally has become available this week, to check whether it supports measurement (this question was too technical for innr support...). It does have the right size, though;
- If you're using HomeKit, the Elgato Eve Energy is almost perfect, except for the Bluetooth connection, which is not very responsive.

roflmao, 2018-06-05:
Are we sure it isnt fixed in later firmware as well?

sveip, 2019-03-01:
@ebaauw, do you know if the new Osram plug now supports power metering? 

It seems its there:
[image]

ebaauw, 2019-03-01:
That looks exactly like the Smart+ plug, which has been out for, I think, several years.  It’s got the same bug as the Lightify plug: no usable power readings.

kurteknikk, 2019-06-08:
Is this issue still not solved with some firmware update ?
If not can someone please tag this issue in the supported devices list of this project ?
The osram smart plug + is listed as supported, but this issue is not tagged.

closed 2019-10-13 by stale[bot]