dresden-elektronik/deconz-rest-plugin #625: undefined symbol: sqlite3_open

ebaauw, 2018-06-06:
Compiled and installed the latest commit, but deCONZ won't start:
```
Jun 06 20:47:30 pi1 systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Jun 06 20:47:30 pi1 deCONZ[12217]: libEGL warning: DRI2: failed to authenticate
Jun 06 20:47:31 pi1 deCONZ[12217]: libpng warning: iCCP: known incorrect sRGB profile
Jun 06 20:47:31 pi1 deCONZ[12217]: /usr/bin/deCONZ: symbol lookup error: /usr/share/deCONZ/plugins/libde_rest_plugin.so: undefined symbol: sqlite3_open
```
Do I need to install another dependency?

manup, 2018-06-06:
Yes I figure libsqlite3-dev is missing. A few commits ago I replaced the static compiled version of sqlite (was from 2013) with the dynamic system library on Raspbian and Ubuntu.

The libsqlite3-dev dependency is missing in the deconz-dev package, I'll add it in the next versions.

ebaauw, 2018-06-06:
Thanks.  It now works after `sudo apt install libsqlite3-dev` and recompiling the plugin.

closed 2018-06-06 by ebaauw