dresden-elektronik/deconz-rest-plugin #626: Xiaomi Sensors not working after reboot

paidmueller74, 2018-06-07:
Hello,

I have some of the "older" Xiaomi Door sensors(with rounded edges) and also some of the Aqara versions. Both versions are connected and react on closing and opening. But after a reboot the older ones do not change state anymore (the Aqara do). When I delete/relearn these sensors they work properly until next reboot. Futher on, I have some "old" temperature/humidity sensors of Xiaomi which do not have this problem. 
Is there any dedicated problem with the older door sensors?
System is an updated raspbian/conbee system

Thanks a lot

terafin, 2018-06-07:
I am seeing the same thing with the newer Aqara & Xiaomi sensors too.

Webserve, 2018-06-08:
did you update the firmware ?

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/610 

There some issues with the latest firmware.

paidmueller74, 2018-06-10:
I have 261F0500 which is not the most actual one that is described in the linked issue. But I have found
out that even the "old" sensors mentioned above show different behaviors according to different
manufacturing dates. Some of 04/2017 show the behavior mentioned aboved but one of 05/17 at least changes state after closing/opening, however still does not initialize correctly when doing nothing after reboot. To sum up: The new Aqara initialize correctly as after a certain time the first heartbeat of the sensor sets the state correctly what the old ones generally seem not to do.
So I have to assume the issue on Xiaomi side and therefore close it here.

terafin, 2018-06-10:
For me, I have downgraded, and until some period of time (not sure yet, at least a few hours) - no sensors update.

Console is filled with:

10:00:12:644 no button map for: lumi.sensor_magnet.aq2 cl: 0x0006 cmd: 0x0A pl[0]: 000
10:00:12:644 ZCL attribute report 0x00158D0002019BFC for cluster 0x0006, ep 0x01
10:00:12:646 discard sensor state push for state/open (already pushed)
10:00:12:648 discard sensor state push for state/lastupdated (already pushed)
10:00:13:260 no button map for: lumi.sensor_magnet.aq2 cl: 0x0006 cmd: 0x0A pl[0]: 000
10:00:13:260 ZCL attribute report 0x00158D0002019BFC for cluster 0x0006, ep 0x01
10:00:13:262 discard sensor state push for state/open (already pushed)
10:00:13:264 discard sensor state push for state/lastupdated (already pushed)
10:00:17:308 no button map for: lumi.sensor_magnet.aq2 cl: 0x0000 cmd: 0x0A pl[0]: 005
10:00:17:308 ZCL attribute report 0x00158D0002019BFC for cluster 0x0000, ep 0x01
10:00:17:308 0x00158D0002019BFC skip Xiaomi attribute 0x0005
10:00:17:308 0x00158D0002019BFC extract Xiaomi special

Which I looked at here:

            if (sensor->lastStatePush.isValid() &&
            item->lastSet() < sensor->lastStatePush)
            {
                DBG_Printf(DBG_INFO, "discard sensor state push for %s (already pushed)\n", e.what());
                return; // already pushed
            }

Is it possible lastSet or lastState push is off? 

I've checked date/time on the device, and the Raspberry Pi 3B+ seems to be right.

terafin, 2018-06-10:
It's also worth noting that the web UI does update, about 1-2 seconds slower than I'd expect with sensor state - but the websocket I have connected do not.

terafin, 2018-06-10:
To remove confusion I opened up a different pull request to cover my case:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/637

Apologies! 

#637

closed 2018-07-10 by paidmueller74