dresden-elektronik/deconz-rest-plugin #629: Ikea Tradfri: Color temperature not changing with scenes (Phoscon)

brandungskieker, 2018-06-08:
I created two scenes in phoscon app:

Brightness: 100% / Color temperature: minimum level (cold)
Brightness: 100% / Color temperature: maximum level (warm)
and saved both.
If the bulb is already in cold color switching to (2) works fine. But not otherwise, the temperature stays in warm mode.

Setup:

Raspberry Pi 3
RaspBee Version 2.05.28 / 3.5.2018, Firmware 26210500
TRADFRI bulb E27 WS clear 950lm (1.2.217)

manup, 2018-06-08:
For some reason the IKEA bulbs don't restore the color temperature in the scene, it seems to the a light firmware issue. If it get fixed in the light firmware we can OTA update the lights.

ebaauw, 2018-06-08:
Are you sure it's the Trådfri firmware?  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/53.

manup, 2018-06-08:
We may are able to build a workaround, but there sure is a bug in the firmware.
Just testet it with my IKEA GU10 tunable white in plain deCONZ scenes cluster

scene 1)
* setup light as warm white brightness ~70%
* store scene 1 in group 1 (unicast)

scene 2)
* setup light as cool white brightness ~50%
* store scene 2 in group 1 (unicast)

On store scene command the light should store the scene locally with current settings.
But when I recall the scenes only the brightness is restored and color temperature is always cool white.

So at least the `ZCL store scene` command in the firmware has a bug.
However a scene can also be created with `ZCL add scene` command which also carries the parameters (on, bri, x, y ...) this might work, I haven't testet it yet but there is already code in the REST plugin which we can use for that.

Not also due some weird reason in the specification scenes don't have a concept of color temperature, in a scene the color temperature is always stored as x,y color. Therefore the add scene command should contain the x,y color of the color temperature (point of blackbody curve).

manup, 2018-06-08:
A few tests later: add scene doesn't work, but when instead using move to color temperature command the move to color x,y is used, the `ZCL store scene` commands works as it should.

The problem is that the move to color temperature command doesn't update the local x,y values in the light (as it should).

So as a first fix, I'll try to convert move to color temperature commands into move to color x,y if the it is a unicast command to an IKEA light (so won't effect other lights).

manup, 2018-10-08:
Verified with IKEA GU10 WS 400lm tunable white and test firmware version 1.2.22.

Close the issue for now.

closed 2018-10-08 by manup

tobycheese, 2019-12-31:
This breaks color temp for people using Ikea Bulbs with homeassistant. Changing color temperature over the GUI works, but the color temperature attribute is not present in the bulb state and thus cannot be used from node-red for instance.

Could someone maybe revisit it? See the issue that was referenced above.

johnekarlsson, 2020-08-02:
This is still an issue!

kejXero, 2020-11-13:
This is still an issue!