dresden-elektronik/deconz-rest-plugin #631: Compile error with latest commit (ubisys C4 config)

ebaauw, 2018-06-09:
I get a compile error with the latest commit:
```
bindings.cpp: In member function ‘void DeRestPluginPrivate::processUbisysC4Configuration(Sensor*)’:
bindings.cpp:2084:27: error: ‘ZclWriteAttributesStructuredId’ is not a member of ‘deCONZ’
     zclFrame.setCommandId(deCONZ::ZclWriteAttributesStructuredId);
```
I assume it needs a newer version of deconz-dev.2.05.xx.deb ?

manup, 2018-06-09:
Sorry, forgot to upload. The -dev file is now updated and should work with latest commits.

holli73, 2018-06-11:
hello,

i have same issue un x64 linux but no new -dev file?

thanks
holli

manup, 2018-06-11:
Hey Holli,
version 2.05.29 is online now with updated -dev package.

closed 2018-06-12 by ebaauw