dresden-elektronik/deconz-rest-plugin #633: Use ConBee as a router to a RaspBee gateway

dghh, 2018-06-09:
I received a ConBee Stick and wanted to include it as a router in a existing zigbee network (RaspBee)
ConBee should run on Windows PC . How can I do that ? The stick is not joining the existing zigbee network. Imassume that i forgot something special....i am not really familiar with zigbee networks, although the RaspBee runs very fine with a lot of different devices.
I also tried to use BitCatcher software, but cannot find any download link. The original link to luxoft is dead....
Can anyone help me this both problems?

ebaauw, 2018-06-09:
> I received a ConBee Stick and wanted to include it as a router in a existing zigbee network (RaspBee)

See #21.  You would use the (default) ZHA link key instead of the ZLL link key.

Be sure to understand what you're doing, though.  While devices need to join the network only once, both gateways create REST API resources independently.  Both gateways will try and configure your devices, where most devices can only report to a single gateway.  Also both gateways will be polling the ZigBee network, increasing its load.

> I also tried to use BitCatcher software, but cannot find any download link. The original link to luxoft is dead....

Use ZShark and WireShark instead, see #405.  Tip: I'm using a 5m USB extension cable to position the ConBee close to the device I'm sniffing.

manup, 2018-06-09:
> I also tried to use BitCatcher software, but cannot find any download link. The original link to luxoft is dead....

We discovered that last week, Bitcatcher seems to be RIP now. As @ebauu mentioned please use ZShark instead. We'll update the related documentation and website in a few days.

dghh, 2018-06-09:
Thanks a lot for the help. I managed to setup the ConBee as a router to the zigbee ( raspbee) network. It works for a while and i could see all the devices and connections in the network. Some unknown groups where also somehow added, but Erik was right. After 3-4 hours the network gets slower and slower and i had to reboot my raspberry in the end. So i will use the ConBee as backup and for sniffing only. Awaiting the update and documentation of ZShark and especially the description how to config wireshark...

closed 2018-06-09 by dghh

ebaauw, 2018-06-09:
It's not too complex and there's some brief help in ZShark.  Basically it encapsulates each ZigBee packet in a UDP packet, which is forwarded to UDP port 17754 of the loopback interface (127.0.0.1).  So in WireShark, you'd simply sniff the loopback interface and filter for UDP port 17754.  No need to setup a capture device.

If you run WireShark on a different machine, you can configure ZShark to send the UDP encapsulated ZigBee packets to the IP address of the machine running WireShark.

ZShark takes care of flashing the ConBee firmware.

abmantis, 2018-06-23:
@ebaauw what do you mean by "Both gateways will try and configure your devices"? Are you talking about when adding a new device?

ebaauw, 2018-06-24:
Yes, but also after restarting deCONZ, re-discovery a device, and periodically.

Elliott32224, 2018-10-21:
Erik:
I am new to Zigbee and you seem most knowledgeable. I have deConz, Conbee dongle running on a Windows 10 box. I am using HomeSeer home automation software and hardware. I have integrated Zigbee via a plugin (I think authored by a countryman of yours). I am trying to get an answer to things that I see in the deConz app that are not well documented.
1) I have several devices (mainly sensors by Xiaomi and one light bulb by Sengled) that have a grey bar to the left of their icons that never have had any lines connecting to either the Conbee icon or any other icons (light bulbs by Ikea or by Osram Lightify). Is this to be expected and does this need to be fixed, because they all appear to work perfectly fine? 
2) There is a grey dot at the left of each icon (just before the bar at the end). It is usually grey, but occasionally it flashes blue and rarely flashes red. This activity cycles among the various devices, whether they are on or off. Is this just the various devices checking connectivity in the mesh?
Thanks in advance.
Elliott

ebaauw, 2018-10-21:
See the deCONZ user manual.  Under _Help_ in the deCONZ GUI.