dresden-elektronik/deconz-rest-plugin #635: Support for Innr Remote RC 110?

jo-me, 2018-06-09:
Hi
just saw that Innr offers a Zibee Remote, the "RC 110" which apparently can control the new innr plugs either directly or indirectly (using a bridge). 

See here: https://shop.innrlighting.com/en/shop/49/remote-control

Is the RC 110 supported by deconz?

Thank you

ebaauw, 2018-06-09:
> Is the RC 110 supported by deconz?

Not yet, but I'll be happy to give it a go, if some-one can lend me one.  I do have an innr gateway and some innr lights (and the new innr SP 120 smart plugs!), so I would be able to sniff standalone mode as well as bridge mode.

> This thing would be a good replacement for the old 443 mhz remotes and even more versatile that the available zigbee remotes - if the buttons/actions are customizable.

Looking at the [manual](https://www.innrlighting.com/clients/innr/themes/uploads/1484932356_Manual%20for%20Remote%20Control%20-%20RC%20110.pdf), it doesn't seem they are.  See also the [quick start](http://www.innrlighting.com/clients/innr/themes/uploads/1455036012_RC%2011x%20installation%20manual%20-%20IL160209.pdf).

I doubt we'll be able to use the numbered buttons in the "Lights" setting, but probably, they would be usable in the "Scenes" setting.  I expect that, effectively, this will be a 9-button switch (On, DImUp, DimDown, 1, 2, 3, 4, 5, 6) with each button only supporting release (so `buttonevent` values 1002, 2002, 3002, 4002, 5002, 6002, 7002, 8002, and 9002).  You would be able to specify custom actions using deCONZ rules.

l-mb, 2018-09-19:
Has anyone made progress on this one? It's a relatively cheap ZigBee remote with plenty buttons, so it'd be very desirable. (I'd be mostly interested in using the button events via Home Assistant.) It also seems to support continuous press, so maybe we can get button held and long release too?

(I'm not incapable of programming low-level bits, but I've never worked with ZigBee on the protocol level. If somebody has pointers to docs I'd be grateful. Or if somebody felt like writing a patch + a few notes on how that all works, I'd be delighted to send someone a remote in exchange ;-)

kasteleman, 2018-09-19:
Not yet. Did not yet include it. But hope to find Some time for it soon. Do
you own the device and tried to include it?

notifications@github.com>

l-mb, 2018-09-20:
> Not yet. Did not yet include it. But hope to find Some time for it soon. Do you own the device and tried to include it?

Yes, I own one of them. It added itself fine to the ZigBee network and I can see it deCONZ's UI, but I would need a bit of help on getting started on how to add support to the code and export the events through the REST API.

l-mb, 2018-09-24:
> Don’t know about the coding part. You can’t put it in a group like a HUE Dimmer for example?

Since the remote doesn't show up in the web app, no. I can see it in the network graph only. Phoscon doesn't realize this is a remote switch so it doesn't expose the device.

ebaauw, 2018-09-29:
The innr remote is on sale at one of my local webshops, so I got me one, for the price of a Xiaomi!  Paired it to my test network (hold the _Program_ and _-_ buttons for five seconds, until the light blinks five times):
[image]

When the _Scenes/Lights_ switch on the remote is in the _Scenes_ position, the remote sends broadcast (!) commands from endpoint 0x01.  When binding the _OnOff_ cluster on endpoint 0x01 to a group, the switch sends groupcast commands instead.

When pressing the _On/Off_ button, the remote alternates between sending _OnOff_ commands _On_ and 0x41 versus _Off_ and 0x40 (so two commands per press).  The remote seems to remember the _On/Off_ state!  There's no commands on hold, nor on release.

When pressing/releasing the _-_ or _+_ button, the remote sends _Level_ command _Step_, with _Step Mode_: _Down_ (for _-_) or _Up_ (for _+_), _Step Size_: 2, and _Transition Time_: 0xffff.  When holding one of these buttons, the remote sends _Move_, with _Step Mode_: _Down_ or _Up_, and _Rate_: 100.  On release, the remote sends _Stop_.

When pressing one of the numbered buttons _1_ to _6_, the remote sends _Move to Level (with OnOff)_, where _Level_ depends on the button: 2, 52, 102, 153, 194, or 254, for buttons _1_ to _6_ respectively.

When the remote's switch is in the _Lights_ position, it doesn't send any command on the number buttons.  Instead, it remembers the button last pressed, and uses the (group) binding of the _Level_ cluster of the corresponding endpoint as destination for the _OnOff_, _-_, and _+_ buttons.  So when binding 0x03 to group 0x1001, 0x04 to group 0x1002, after pressing _1_, the _OnOff_, _-_, and _+_ buttons send commands to group 0x1001 from endpoint 0x03; after pressing _2_, the commands are sent to group 0x1002 from endpoint 0x04.  The remote remembers the _OnOff_ state separately for each endpoint.

As far as I can tell so quickly, the _Program_ and associated _<_ and _>_ buttons don't send any ZigBee commands (at least, when the switch has joined a network).

Supporting the remote in the deCONZ REST API is going to be interesting.  For the _Scenes_ setting, it's indeed simply a 9-button switch.  Typically this would be exposed as a single ZHASwitch resource (for endpoint 0x01) with buttonevents 1002 (_On/Off_); 2001, 2002, 2003 (_-_); 3001, 3002, 3003 (_+_); 4002 (_1_); 5002 (_2_); 6002 (_3_); 7002 (_4_); 8002 (_5_); and _9002 (_6_);  We could change the order, so 1002, ..., 6002, would match the numbered buttons 1, ..., 6, but see below.

For the _Lights_ setting, it's probably easiest to expose an additional ZHASwitch resource per endpoint, with buttonevents 1002 (_On/Off_); 2001, 2002, 2003 (_-_); 3001, 3002, 3003 (_+_).  This is why we don't want to change the order for the ZHASwitch resource on endpoint 0x01.

In HomeKit, this would become a single accessory with 27 buttons!

For the REST API, it would be enough to simply bind all 7 endpoints to the same group - the source endpoint can be used to determine the appropriate resource.  For using the remote in dual mode (through the gateway and standalone, controlling the lights in a group directly), it would make sense to use a different group for each endpoint.  @manup, I need your help here: where to get the group IDs?  I suppose they need to be stored in `config.group` of the corresponding ZHASwitch resource?

ebaauw, 2018-09-29:
Some more insights.  The remotes sees the _On/Off_ state for endpoint 0x01 as the combined state of the other endpoints.  I.e. when setting endpoint 0x01 off, the remote assumes the groups for endpoints 0x03, ..., 0x08 are off as well.  Same for setting 0x01 on.  With the switch set to _Lights_, the remote forgets the number button pressed after some time - the next commands are sent from endpoint 0x01 and to the corresponding group.

I've been thinking about exposing seven different ZHASwitch resources.  While straightforward, it seems overkill.  Alternatively, we could just continue to number the buttonevents using a single ZHASwitch resource.  So 10002 (_On/Off_ for _1_); 11001, 11002, 11003 (_-_ for _1_); 12001, 12002, 12003 (_+_ for _1_); 13002 (_On/Off_ for _2_); ...; 27001, 27002, 27003 (_+_ for _6_).  Not only fewer resources; it would also be easier on homebridge-hue and, I suppose, on plugins for other home automation platforms.  I would like the buttonevents to have a continuous range in the 1000s, so `buttonevent div 1000` still indicates the button, and `buttonevent mod 1000` the action.

ebaauw, 2018-09-29:
OK, I got the REST API to create a single ZHASensor resource and to generate the buttonevent when pressing a button:
```
{
  "config": {
    "group": "4102",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "138fd8c4067c6132b1a7dab4f9c38967",
  "manufacturername": "innr",
  "mode": 1,
  "modelid": "RC 110",
  "name": "innr remote",
  "state": {
    "buttonevent": 27003,
    "lastupdated": "2018-09-29T20:48:39"
  },
  "swversion": "0x21000e46",
  "type": "ZHASwitch",
  "uniqueid": "00:15:8d:00:01:0c:44:37-01-0006"
}
```

I haven't yet figured out how to setup the bindings to the groups.  Also `config.group` is overwritten with the group that is (manually) bound to the endpoint corresponding to the last button pressed:
```
{
  "config": {
    "group": "4097",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "d867523eeb6bf9da8753f4acc06e8a57",
  "manufacturername": "innr",
  "mode": 1,
  "modelid": "RC 110",
  "name": "innr remote",
  "state": {
    "buttonevent": 10002,
    "lastupdated": "2018-09-29T20:49:39"
  },
  "swversion": "0x21000e46",
  "type": "ZHASwitch",
  "uniqueid": "00:15:8d:00:01:0c:44:37-01-0006"
}
```

ebaauw, 2018-09-29:
Some more playing around.  The remote returns a default response with status _Software Failure_ (0xc1) when trying to get the groups or endpoints from the _ZLL Commissioning_ cluster.  However, each controller endpoint has a server _Groups_ cluster, with the capacity of just one group.  When adding a group to that endpoint, the corresponding commands will be sent to that group.  No need to create a binding in that case.

So it might be enough to create seven groups (or rather `/groups` resources), add these to the remote's controller endpoints, and set `config.groups` to a list of these seven groups.  I'm not sure if Phoscon can handle that, @manup - maybe it would be better to use seven ZHASwitch resources (each with only one group)?

ebaauw, 2018-09-29:
Even some more playing around.  The trick with adding the group only seems to work for endpoint 0x01.  For the other endpoints, a binding is required.  The REST API plugin chokes on `config.group` containing a list of groups instead of a single one, so I changed to exposing the remote as seven ZHASwitch resources.  After manually binding each endpoint's _Level_ cluster to the corresponding group, the REST API plugin sets `config.group` _and_ creates the corresponding group when receiving the command!

Here's the `/sensors` resources:
```
  "9": {
    "config": {
      "group": "4096",
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "038a8aa3aa9a4a5ebba740e6e907d1da",
    "manufacturername": "innr",
    "mode": 1,
    "modelid": "RC 110",
    "name": "RC 110 9",
    "state": {
      "buttonevent": 1002,
      "lastupdated": "2018-09-29T23:35:52"
    },
    "swversion": "20161014",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:01:0c:44:37-01-0006"
  },
  "10": {
    "config": {
      "group": "4097",
      "on": true,
      "reachable": true
    },
    "ep": 3,
    "etag": "038a8aa3aa9a4a5ebba740e6e907d1da",
    "manufacturername": "innr",
    "mode": 1,
    "modelid": "RC 110",
    "name": "RC 110 10",
    "state": {
      "buttonevent": 3002,
      "lastupdated": "2018-09-29T23:33:37"
    },
    "swversion": "20161014",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:01:0c:44:37-03-0006"
  },
  "11": {
    "config": {
      "group": "4098",
      "on": true,
      "reachable": true
    },
    "ep": 4,
    "etag": "038a8aa3aa9a4a5ebba740e6e907d1da",
    "manufacturername": "innr",
    "mode": 1,
    "modelid": "RC 110",
    "name": "RC 110 11",
    "state": {
      "buttonevent": 3003,
      "lastupdated": "2018-09-29T23:33:50"
    },
    "swversion": "20161014",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:01:0c:44:37-04-0006"
  },
  "12": {
    "config": {
      "group": "4099",
      "on": true,
      "reachable": true
    },
[... 284 lines cut ...]
      "scene": null,
      "xy": [
        0,
        0
      ]
    },
    "class": "Other",
    "devicemembership": [
      "15"
    ],
    "etag": "038a8aa3aa9a4a5ebba740e6e907d1da",
    "hidden": false,
    "id": "4102",
    "lights": [],
    "lightsequence": [],
    "multideviceids": [],
    "name": "RC 110 15",
    "scenes": [],
    "state": {
      "all_on": false,
      "any_on": false
    },
    "type": "LightGroup"
  }
```

manup, 2018-09-30:
Reminds me of the Busch-Jaeger switches, these have up to 4 separated ZHASwitch resources where each can control a group. I have to admit the multiple ZHASwitch resources approach for a single physical switch is *very* cumbersome to handle in gui applications or any client in general, in the Phoscon App we've added a *switches* layer for that purpose just to combine the sensors of one device into a single switch resource or model, which it should be in first place.

For the ubisys C4 whch also can control 4 groups the implementation is different, I think config.group contains a string list (?) — which should better be a json array. Not perfect either but easier to handle for the REST-API clients.

ebaauw, 2018-09-30:
> Phoscon App we've added a switches layer for that purpose just to combine the sensors of one device into a single switch resource or model, which it should be in first place.

Sounds familiar - I did the same in homebridge-hue.

> For the ubisys C4 which also can control 4 groups the implementation is different, I think config.group contains a string list (?) — which should better be a json array.

Yeah, I saw `config.json` for my ubisys D1 contains two group IDs separated by a comma:
```
{
  "config": {
    "group": "230,240",
    "mode": "momentary",
    "on": true,
    "reachable": true
  },
  "ep": 2,
  "etag": "889ac780dcf77135f497dfeecb13702a",
  "manufacturername": "ubisys",
  "mode": 1,
  "modelid": "D1 (5503)",
  "name": "Dining Room Table",
  "state": {
    "buttonevent": 2002,
    "lastupdated": "2018-09-21T22:36:40"
  },
  "swversion": "20171108-DE-FB0",
  "type": "ZHASwitch",
  "uniqueid": "00:1f:ee:00:00:00:19:63-02-0006"
}
```
So I implemented the same for the RC 110 (with a list of seven group IDs), but then errors messages appeared in the deCONZ log about an invalid `config.group` value.  Also, the REST API only create/removes the groups automatically, when `config.group` contains a single group ID.  I commented out the code for the list of group IDs, so it will be easy to change that.

If memory serves, I manually created the bindings to the groups for _OnOff_ and _Level_ clusters in the _Dimmer Switch_ endpoints 0x02 and 0x03 of the ubisys D1.  I don't remember if I set `config.group` by hacking the database or if the REST API set it automatically.  There's still something missing, though, the `devicemembership` of these groups is empty.

I cannot seem find any code in the REST API plugin that maintains the bindings, though.  Does Phoscon maintain the bindings for the C4?  If so, using what REST calls?

I've been thinking about making `config.group` updatable through the REST API, for an easier way to maintain the bindings.  For the battery-powered RC 110, we probably need to use `config.pending`.  This logic will become very complicated if `config.group` is a list of group IDs (or a JSON array of group IDs).

ebaauw, 2018-09-30:
Fixed the `buttonevent` values to be consistent with other switches:
- Continue numbering button across endpoint, so _On/Off_ for Lights 1 (endpoint 0x03) yields 10002 instead of 1002;
- Place _DimUp_ before _DimDown_.

Added support in homebridge-hue.

ebaauw, 2019-02-10:
Still need to create and assign groups on pairing.

Also, still need to combine the resources into one with `config.groups` as an array.

manup, 2019-02-10:
Yes we have several of these remotes now which control multiple groups.

innr, Paulmann, Paul Neuhaus and Müller Licht.

It's a tricky problem, even simple remote with just one group are hard to maintain since the group ids might change for various reasons.

ebaauw, 2019-02-10:
ubisys

jo-me, 2019-04-06:
So, I've just paired my RC110 with deconz Version 2.05.60 w/firmware 26320500 by adding a new "Other" switch..

What happened is that the paring apparently worked, but Phoscon did not show any indication of that.
Also, the switch did not appear anywhere near the other switches or sensors.

The API had several (!?) RC 110 listed under sensors. one of them showed updated timestamps when pressing the 1-6 buttons.

The power button of the remote now turns on/off EVERY light in deconz..

@manup can you estimate when the remote will be properly supported? And properly configurable via Phoscon?

Thanks

ebaauw, 2019-04-06:
> The API had several (!?) RC 110 listed under sensors. 

Seven for now (until `state.group` can handle multiple groups), see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/635#issuecomment-425682814.

> one of them showed updated timestamps when pressing the 1-6 buttons.

That's with the switch set to _Scenes_.  When you set it to _Lights_, these buttons don't send a command themselves, but change the group for the _On/Off_, _+_ and _-_ buttons (causing the timestamp of another resource to be updated)

> The power button of the remote now turns on/off EVERY light in deconz..

Yes, you need to bind each endpoint to a group manually.

closed 2019-08-11 by stale[bot]

jo-me, 2019-08-25:
@ebaauw do you have a documentation on how to do this?
I've looked in the Windows deconz app but it is not apparent on how to configure this there..

Also, I do not see any groups for the RC110 (neither in phoscon nor in API).
Apart from that, buttons 4, 5 and 6 don't generate events anymore (1, 2, 3 still do).

How did you get this remote to work?

If I need to repair the remote, how can I remove the device? Do I need to delete each individual switch/sensor ?

ebaauw, 2019-08-25:
Open the _Bind Dropbox_ panel in the deCONZ GUI, drag the client _On/Off_ cluster of the RC110's first endpoint to the _Source_ field, select a group for the destination and press _Bind_.  Repeat for the _On/Off_ clusters in the other six end points.

> Apart from that, buttons 4, 5 and 6 don't generate events anymore (1, 2, 3 still do).

Probably because their endpoints aren't bound to a group.

> Do I need to delete each individual switch/sensor ?

You can just reset the remote an pair it again.  To start from scratch, you'd need to delete each ZHASwitch resource, and remove the node from the GUI.

jo-me, 2019-09-18:
So,
I paired the remote again without deleting anything.

Then in the GUI I dragged the on/off entry of the first table to the source field. 

When I enter a random group name it won't let me bind because the button remains disabled. 
Entering an existing group name has the same effect.

What groups is this thing looking for? 
I cannot select one as you said because it is a plain text field.  

Here is a screenshot:
[image]

There is a add/remove group at the top of the left panel. Do I have to create soemting there?
If yes, what do I need to enter for group ID?
I tried this the last time but it did not seem to work properly. I added a group with a certain name, requested it afterwards and then apparently the group was there but without the name..

ebaauw, 2019-09-18:
You need to enter the id (in hex) of the ZigBee group.

Note that a ZigBee group is more like a multicast address than a physical object.  “Adding” a group is telling a light to listen the corresponding address.  Binding a switch to a group is telling the switch to send commands to that address (irrespective whether any light is actually listening).  The REST API plugin polls each light for the groups it listens to, and creates the `/groups` resource out if that info.  It makes you think you add a light to the group, but in ZigBee, the group is added to the light instead.

jo-me, 2019-09-18:
Thank you for getting back to me.
The whole thing is not exactly an intuitive process..

I created a couple of empty groups with Phoscon, e.g. one that received the ID 9 (retrieved from REST API)

I entered 0x0009 as bind destination and then executed it. It hang for a bit but after I woke the remote by pressing (-) it said success. 
[image]

I assigned the first three on/off things to my new groups. 
The groups reported by the REST API seem to be unchanged.
When I press the number buttons on the remote, they still cause havoc turning everything on & off..

Shouldn't the first 3 buttons now turn the new groups on/off?

ebaauw, 2019-09-18:
You created a binding for endpoint 0x04, so that’s for group 2 when the switch is on the light setting.  If the binding succeeds, the corresponding ZHASwitch resource’s `config.group` is populated when the switch sends the next command.

It’s a while since I played with this, you might also need to bind the _Level Control_ cluster.  And _Scenes_ on the first endpoint.

jo-me, 2019-09-18:
Ok, so I added bindings to each on/off thing in the list to separate pre-created groups.
How can I check whether this has worked? I can't find a list of the existing bindings.
It said "success" every time but I'm left in the dark. 

When clicking through the buttons now I see that one `config.group` is filled, but all others are not. The one that's filled is the for the light no.1 (e.g. + - in "Lights" mode after clicking "1") and I can see button events like "buttonevent": 11002". For the other light numbers it does not work.

For the rest nothing group related happened unfortunately and except for one other zhaswitch there were not even buttonevents registered.

jo-me, 2019-09-18:
Shouldn't the number in the on/off cluster increase with each binding?

[image]

[image]

jo-me, 2019-09-18:
Here's what the REST api shows:

 "33": {
        "config": {
            "group": "8",
            "on": true,
            "reachable": true
        },
        "ep": 3,
        "etag": "c19f77f2f7ef267af60b66476dfd85d2",
        "manufacturername": "innr",
        "mode": 1,
        "modelid": "RC 110",
        "name": "RC 110 33",
        "state": {
            "buttonevent": 12002,
            "lastupdated": "2019-09-18T19:05:18"
        },
        "swversion": "0x22000e46",
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:01:7c:ed:a9-03-0006"
    },
    "34": {
        "config": {
            "group": null,
            "on": true,
            "reachable": false
        },
        "ep": 4,
        "etag": "a595bc16e69ddd910b754b6c999b0356",
        "manufacturername": "innr",
        "mode": 1,
        "modelid": "RC 110",
        "name": "RC 110 34",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "swversion": "0x22000e46",
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:01:7c:ed:a9-04-0006"
    },

ebaauw, 2019-09-18:
The number in the cluster list shows the number of attributes, typically 0 for client clusters.

Looks like the bindings didn’t succeed for the other endpoints.  The remote is a deep sleeper, so the sequence of waking it and pressing bind is somewhat delicate.  I don’t trust blindly the status reported by the GUI - always verify through `config.group` and button events that the binding was made.

If memory serves, the light on the remote should blink when it sends a command.  If it doesn’t, it might help to remove and re-insert the batteries.  And double-check the batteries.

jo-me, 2019-09-19:
I removed the remote today along with all zwaswitches and even purged the deleted items from the DB. After re-pairing everythin was back to square one.

So I started creating group bindings for the 6 light buttons again. I found that the config.group entry only appeared after binding the level control cluster of each endpoint. Binding the on/off cluster did not do anything visible. Is it still necessary behind the scenes?

The only problem child is the scenes mode now.
It shows the button clicks in the API, but no group. I tried to bind on/off , level control and scenes as you said but it does not have any effect. Is something else necessary here?

```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "f524a94ec3ccb1e3753071408bb0a168",
    "manufacturername": "innr",
    "mode": 1,
    "modelid": "RC 110",
    "name": "RC 110 26",
    "state": {
        "buttonevent": 2002,
        "lastupdated": "2019-09-19T09:14:15"
    },
    "swversion": "0x22000e46",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:01:7c:ed:a9-01-0006"
}
```

Thanks for your support!

ebaauw, 2019-09-19:
According to my notes above, https://github.com/dresden-elektronik/deconz-rest-plugin/issues/635#issuecomment-425658792, you need to bind the _OnOff_ cluster on the first endpoint and the _Level Control_ cluster on the other endpoints.

jo-me, 2019-09-19:
Yeah, I tried that dozens of times now, but binding the onoff cluster just does not work for the first endpoint. (it also did not work for the other endpoints). 

Binding to the level control cluster for the other endpoints worked fine though.. Once I understood that only that is necessary to produce the group assignment I went through the 6 buttons in no time.

Is there another trick involved?
Is onoff cluster binding maybe broken? or do I need to press a special button?

ebaauw, 2019-09-19:
You might try and add the group to the first endpoint using the _Groups_ cluster, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/635#issuecomment-425678121.

jo-me, 2019-09-20:
Aah, binding with the previous way did not work for the groups cluster (there are 2 with the same name) so I used the editor fields on the first "Groups" cluster. 

[image]

This finally worked.

One open issue that I also had before all of this: 
the buttons 4, 5 and 6 do not produce button events.  
The first three number buttons generate  4002, 5002, and 6002, but the other 3 are mute.

What can this be?

They do work in Lights mode, just not in Scenes mode..

ebaauw, 2019-09-20:
Odd.  Either a hardware failure (but they do select the right group in scenes mode?) or your remote is sending different commands than mine.  Could you check if the _Date Code_ and _SW Build ID_ match my screenshot above?  If the remote is sending something, there should be messages in the deCONZ log about no button map entry found, with the info needed to add them.

jo-me, 2019-09-24:
Yes, in the log I see these messages for buttons 4,5,6:

13:37:32:203 no button handler for: RC 110 ep: 0x01 cl: 0x0008 cmd: 0x04 pl[0]: 0x99
13:37:34:461 no button handler for: RC 110 ep: 0x01 cl: 0x0008 cmd: 0x04 pl[0]: 0xC2
13:37:37:085 no button handler for: RC 110 ep: 0x01 cl: 0x0008 cmd: 0x04 pl[0]: 0xFE

What can I do with that now?

Regarding Date code and SW Build id it looks like my remote has a newer software and a lot of empty fields in the basic cluster:

[image]

EDIT: after clicking "read" again, it filled up:

[image]

ebaauw, 2019-09-24:
These are the same values as in the button map.  I have no clue why they’re not matched (and moreover, why 1, 2, and 3 are matched).

ebaauw, 2019-10-05:
Above commit creates groups for all seven ZHASwitch resources and sets up the bindings for all but the first resource.  Now (after re-pairing) all buttonevents should work, but the scenes commands work on all lights, rather on the group associated with the first ZHASwitch resource.

The first 0x01 endpoint doesn't accept bindings.  Instead, it needs a group assignment, i.e. an _Add to group_ command to its _Groups_ server (!) cluster.  @manup, I'm not sure how to implement this.  There is a task to add a lightnode to a group, but this is a sensornode.

eddyvanroy, 2020-04-27:
Hello,
I've added the RC 110 to my HomeAssistant bridge. With the different buttons on the remote control, all my inner candles and switches are controlled.  

However is there a way to delete the default behavior via deconz, since I  don't have the innr bridge.
For example I want to assign a specific action on the on/off button via the button event, instead of switching off/on all innr endpoints.

I've tried to delete the different groups via the API, but didn't solve the issue. Any suggestions?

ebaauw, 2020-04-27:
The remote sends hard-coded commands that cannot be changed.  If you want to use the buttons differently, leave the groups empty and create deCONZ rules for the corresponding buttonevent values.

eddyvanroy, 2020-04-27:
Ok thanks for the answer on question on the way to delete the default behavior of the innr endpoints via deconz. 

Are all the remote controls using the same hardcoded values to trigger the on/off of all the endpoints? Meaning if I would buy another remote (non innr) would the on/off button trigger again the on/off for the innr endpoints?

ebaauw, 2020-04-27:
No, what commands each remote send is determined in their firmware.  Typically, you can only change the group (through a binding).  Professional grade remotes (like ubisys) can be configured, but those are a different price range.

ebaauw, 2020-05-27:
@jo-me I think we found the cause of the "no button map" messages.  See #2061.

Larx, 2020-07-01:
Is there a way to use the RC110 with the WebUI of deconz?
I can pair the device, however then it is not available within the WebUI to configure.
I can, however, receive the button presses (when the switch is turned to "scenes") within HomeAssistant. Nothing is received when in "lights" mode.
Am I doing something wrong here?

Larx, 2020-07-04:
To answer myself partially. Found some answers here: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1939#issuecomment-538682678

OSvalgaard, 2021-02-05:
I am very sorry to post a little off topic here, but I am unable to find any way to contact @ebaauw or one of you other guys directly. 

Do anyone of you know about SmartThings Device Handlers, and are able to provide a device handler for this remote (RC 110)?