dresden-elektronik/deconz-rest-plugin #636: New XIAOMI motion detector Aqara 2018 mit Illuminance sensor

BigWumpus, 2018-06-10:
Hello,
i got a new human motion-detector from XIAOMI, now labeled with Aqara and from 2018.
New ist the illuminance-sensor.
It works with Deconz, but the measurement ist never updated. On an original XIAOMI-Gateway it is running and updating.
[image]
[image]
[image]
[image]

Maybe you can fix this.

GatoPharaoh, 2018-06-16:
does the motion part work for you?
I got the exact same sensor - the illuminance value does update for me but  it doesn't detect motion - in the phoscon app it presented as motion sensor - but only shows LUX and temperature(don't know why?)

GatoPharaoh, 2018-06-16:
this is what it looks like on the web app - as the motion sensor doesn't have a temperature module, I assume deconz just doens't know the (new) sensor model?
[image]

GatoPharaoh, 2018-06-16:
had some similar trouble with the exact same sensor - what helped for me:

- delete the node
- delete it on the web app
- reset the motion sensor
- restart deconz 
- re-pair it with deconz

manup, 2018-06-16:
Looks like the ZHAPresence resource wasn't created for some reason. Easiest way to fix this is to start sensor search and reset the sensor with the small button, it should then rejoin and the missing resource should be created.

manup, 2019-01-28:
All known versions of this sensor should work with recent deCONZ versions.

closed 2019-01-28 by manup