dresden-elektronik/deconz-rest-plugin #637: Websockets for Aqara Sensors not working after reboot (alternate)

terafin, 2018-06-10:
Hi there. Every reboot of my Raspberry Pi 3B+ I lose all all web socket messages coming from them.

This is with SW version: 2.05.25
Firmware: 261F0500

Here are the startup logs, couple with me tapping the button on the devices once to check them in a 30 seconds after or so.

[marthoc/deconz] Current deCONZ version: 2.05.25
[marthoc/deconz] Web UI port: 80
[marthoc/deconz] Websockets port: 443
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
17:11:30:906 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
17:11:31:132 dev /dev/ttyAMA0
17:11:31:132 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
17:11:31:317 parent process /bin/sh
17:11:31:317 gw run mode: normal
17:11:31:317 sd-card cid: 1b534d454231515430ffffffff0114a1
17:11:31:339 don't close database yet, keep open for 900 seconds
10:11:31:340 Daylight now: goldenHour1, status: 160
10:11:31:342 started websocket server at port 443
10:11:31:344 discovery updated announce interval to 10 minutes
10:11:31:347 found node plugin: libde_rest_plugin.so - REST API Plugin
10:11:31:350 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
10:11:31:368 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
10:11:31:406 dev /dev/ttyAMA0
10:11:31:446 discard sensor state push for state/status (already pushed)
10:11:31:454 discard sensor state push for state/status (already pushed)
10:11:31:558 Device firmware version 0x261F0500
10:11:31:570 unlocked max nodes: 200
10:11:31:765 Device protocol version: 0x0106
10:11:31:778 CTRL restore cached node 0x00212effff0229cd
10:11:31:779 new node - ext: 0x00212effff0229cd, nwk: 0x0000
10:11:31:904 Current channel 15
10:11:31:947 CTRL ANT_CTRL 0x03
10:11:32:052 Device protocol version: 0x0106
10:11:32:191 Current channel 15
10:11:32:235 CTRL ANT_CTRL 0x03
10:11:32:459 neigbor 0x00158d00022c644b is unknown child
10:11:33:456 neigbor 0x00158d0002019bfc is unknown child
10:11:33:811 New websocket 10.0.90.243:45128 (state: 3) 
10:11:33:849 CTRL restore cached node 0x00158d00022c644b
10:11:33:852 new node - ext: 0x00158d00022c644b, nwk: 0xDA95
10:11:33:852 SensorNode 4 set node 0x00158d00022c644b
10:11:33:853 SensorNode 5 set node 0x00158d00022c644b
10:11:33:853 SensorNode 6 set node 0x00158d00022c644b
10:11:33:858 discard sensor config push for config/reachable (already pushed)
10:11:33:859 discard sensor config push for config/reachable (already pushed)
10:11:33:860 discard sensor config push for config/reachable (already pushed)
10:11:34:329 CTRL restore cached node 0x00158d0002019bfc
10:11:34:331 new node - ext: 0x00158d0002019bfc, nwk: 0xC89C
10:11:34:331 SensorNode 3 set node 0x00158d0002019bfc
10:11:34:336 discard sensor config push for config/reachable (already pushed)
10:11:36:582 dev /dev/ttyAMA0
10:11:36:584 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF
10:11:36:584 GW firmware version: 0x261f0500
10:11:36:584 GW firmware version is up to date: 0x261f0500
10:11:36:718 Announced to internet
10:11:36:718 discovery server date: Sun, 10 Jun 2018 17:11:36 GMT
10:11:36:718 	 local time seems to be ok
10:11:36:718 discovery found version 2.04.35 for update channel stable
10:11:39:939 New websocket 10.0.1.132:51517 (state: 3) 
10:11:41:501 Daylight now: goldenHour1, status: 160
10:11:42:790 New websocket 10.0.90.243:55278 (state: 3) 
10:11:47:232 no button map for: lumi.sensor_magnet.aq2 cl: 0x0000 cmd: 0x0A pl[0]: 005
10:11:47:232 ZCL attribute report 0x00158D0002019BFC for cluster 0x0000, ep 0x01
10:11:47:232 0x00158D0002019BFC skip Xiaomi attribute 0x0005
10:11:47:232 0x00158D0002019BFC extract Xiaomi special
10:11:47:232 	01 battery 3055 (0x0BEF)
10:11:47:232 	03 temperature 19 °C
10:11:47:232 	04 unknown 5032 (0x13A8)
10:11:47:232 	05 unknown 83 (0x0053)
10:11:47:232 	06 unknown 279172874753 (0x0000004100000201)
10:11:47:232 	64 on/off 0
10:11:47:235 discard sensor config push for config/battery (already pushed)
10:11:47:236 discard sensor config push for config/temperature (already pushed)
10:11:47:237 discard sensor state push for state/open (already pushed)
10:11:51:501 Daylight now: goldenHour1, status: 160
10:11:53:903 scan skip host .210
10:11:54:822 no button map for: lumi.weather cl: 0x0000 cmd: 0x0A pl[0]: 001
10:11:54:822 ZCL attribute report 0x00158D00022C644B for cluster 0x0000, ep 0x01
10:11:54:823 0x00158D00022C644B extract Xiaomi special
10:11:54:823 	01 battery 3015 (0x0BC7)
10:11:54:823 	04 unknown 5032 (0x13A8)
10:11:54:823 	05 unknown 6 (0x0006)

It's worth noting that the WebUI does update but about 1-2 seconds slower than usual - however I never get any web socket updates.

terafin, 2018-06-10:
I did just notice this oddity in the zll.db

sqlite> select * from sensors;
             sid = 2
            name = Window / Door Sensor
            type = ZHAOpenClose
         modelid = lumi.sensor_magnet.aq2
manufacturername = LUMI
        uniqueid = 00:15:8d:00:02:01:9b:fc-01-0006
       swversion = 
           state = {"lastupdated":"2018-06-03T22:06:44","open":false}
          config = {"battery":100,"on":true,"reachable":true,"temperature":2300}
     fingerprint = {"d":65535,"ep":1,"in":[6],"p":260}
    deletedState = deleted
            mode = 1

             sid = 3
            name = Window / Door Sensor
            type = ZHAOpenClose
         modelid = lumi.sensor_magnet.aq2
manufacturername = LUMI
        uniqueid = 00:15:8d:00:02:01:9b:fc-01-0006
       swversion = 
           state = {"lastupdated":"2018-06-10T06:39:03","open":true}
          config = {"battery":100,"on":true,"reachable":true,"temperature":2400}
     fingerprint = {"d":65535,"ep":1,"in":[6],"p":260}
    deletedState = normal
            mode = 1

Two entries for this sensor, same unique ID - but one is deleted?

terafin, 2018-06-10:
Tried on a clean install. Added one sensor. Worked until reboot, then same as above.

terafin, 2018-06-12:
Keen vents work fine (unresponsive for maybe 5 minutes post reboot - but do recover).

terafin, 2018-06-13:
Updated to the latest version, which I saw had some logging and flushing. New logs here which cover.

Pairing, verifying working, restarting docker (same result with the .deb fwiw), and logging

20:51:09:140 SensorNode 6: OpenClose 6 added
20:51:09:145 Websocket 10.0.1.132:49977 send message: {"e":"added","id":"6","r":"sensors","sensor":{"config":{"battery":null,"on":true,"reachable":true,"temperature":null},"ep":1,"etag":"dc5fea78cf284ff7d316e6080d0ff7a5","id":"6","manufacturername":"LUMI","modelid":"lumi.sensor_magnet.aq2","name":"OpenClose 6","state":{"lastupdated":"2018-06-13T03:51:09","open":true},"type":"ZHAOpenClose","uniqueid":"00:15:8d:00:02:01:9b:fc-01-0006"},"t":"event"} (ret = 396)
20:51:09:145 Websocket 10.0.1.132:49992 send message: {"e":"added","id":"6","r":"sensors","sensor":{"config":{"battery":null,"on":true,"reachable":true,"temperature":null},"ep":1,"etag":"dc5fea78cf284ff7d316e6080d0ff7a5","id":"6","manufacturername":"LUMI","modelid":"lumi.sensor_magnet.aq2","name":"OpenClose 6","state":{"lastupdated":"2018-06-13T03:51:09","open":true},"type":"ZHAOpenClose","uniqueid":"00:15:8d:00:02:01:9b:fc-01-0006"},"t":"event"} (ret = 396)
20:51:09:149 Websocket 10.0.1.132:49977 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:09","open":true},"t":"event"} (ret = 108)
20:51:09:149 Websocket 10.0.1.132:49992 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:09","open":true},"t":"event"} (ret = 108)
20:51:09:151 discard sensor state push for 6: state/lastupdated (already pushed)
20:51:09:312 no button map for: lumi.sensor_magnet.aq2 cl: 0x0006 cmd: 0x0A pl[0]: 000
20:51:09:312 ZCL attribute report 0x00158D0002019BFC for cluster 0x0006, ep 0x01
20:51:09:314 Websocket 10.0.1.132:49977 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:09","open":false},"t":"event"} (ret = 109)
20:51:09:314 Websocket 10.0.1.132:49992 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:09","open":false},"t":"event"} (ret = 109)
20:51:09:315 discard sensor state push for 6: state/lastupdated (already pushed)
20:51:09:336 Websocket 10.0.1.132:49977 send message: {"e":"changed","id":"3","name":"Multi Sensor (2)","r":"sensors","t":"event"} (ret = 76)
20:51:09:337 Websocket 10.0.1.132:49992 send message: {"e":"changed","id":"3","name":"Multi Sensor (2)","r":"sensors","t":"event"} (ret = 76)
20:51:09:338 Websocket 10.0.1.132:49977 send message: {"e":"changed","id":"4","name":"Multi Sensor (2)","r":"sensors","t":"event"} (ret = 76)
20:51:09:338 Websocket 10.0.1.132:49992 send message: {"e":"changed","id":"4","name":"Multi Sensor (2)","r":"sensors","t":"event"} (ret = 76)
20:51:09:339 no button map for: lumi.sensor_magnet.aq2 cl: 0x0000 cmd: 0x0A pl[0]: 001
20:51:09:339 ZCL attribute report 0x00158D0002019BFC for cluster 0x0000, ep 0x01
20:51:09:339 0x00158D0002019BFC extract Xiaomi special
20:51:09:339 	04 unknown 5032 (0x13A8)
20:51:09:341 Websocket 10.0.1.132:49977 send message: {"e":"changed","id":"5","name":"Multi Sensor (2)","r":"sensors","t":"event"} (ret = 76)
20:51:09:341 Websocket 10.0.1.132:49992 send message: {"e":"changed","id":"5","name":"Multi Sensor (2)","r":"sensors","t":"event"} (ret = 76)
20:51:09:360 no button map for: lumi.sensor_magnet.aq2 cl: 0x0006 cmd: 0x0A pl[0]: 000
20:51:09:360 ZCL attribute report 0x00158D0002019BFC for cluster 0x0006, ep 0x01
20:51:09:362 Websocket 10.0.1.132:49977 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:09","open":true},"t":"event"} (ret = 108)
20:51:09:363 Websocket 10.0.1.132:49992 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:09","open":true},"t":"event"} (ret = 108)
20:51:09:364 discard sensor state push for 6: state/lastupdated (already pushed)
20:51:09:460 no button map for: lumi.sensor_magnet.aq2 cl: 0x0006 cmd: 0x0A pl[0]: 000
20:51:09:460 ZCL attribute report 0x00158D0002019BFC for cluster 0x0006, ep 0x01
20:51:09:462 Websocket 10.0.1.132:49977 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:09","open":true},"t":"event"} (ret = 108)
20:51:09:462 Websocket 10.0.1.132:49992 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:09","open":true},"t":"event"} (ret = 108)
20:51:11:779 no button map for: lumi.sensor_magnet.aq2 cl: 0x0006 cmd: 0x0A pl[0]: 000
20:51:11:779 ZCL attribute report 0x00158D0002019BFC for cluster 0x0006, ep 0x01
20:51:11:782 Websocket 10.0.1.132:49977 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:11","open":false},"t":"event"} (ret = 109)
20:51:11:782 Websocket 10.0.1.132:49992 send message: {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-06-13T03:51:11","open":false},"t":"event"} (ret = 109)
20:51:11:784 discard sensor state push for 6: state/lastupdated (already pushed)
20:51:13:916 Content not completely loaded (got 0 of 0), wait 20ms
20:51:13:954 Bind response success
20:51:13:954 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x0022A3000013399B (wait reading or unsupported)
20:51:14:342 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 1
20:51:14:342 DB save zll database items 0x00000880
20:51:14:381 DB saved in 39 ms
20:51:14:381 don't close database yet, keep open for 900 seconds
[... cut ...]
20:54:13:843 poll node 00:22:a3:00:00:13:39:9b-01
20:54:17:843 poll node 00:22:a3:00:00:13:39:9b-01
20:54:21:341 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 1
20:54:21:843 poll node 00:22:a3:00:00:13:39:9b-01
20:54:25:843 poll node 00:22:a3:00:00:13:39:9b-01
20:54:30:362 Current channel 11
20:54:30:843 poll node 00:22:a3:00:00:13:39:9b-01
20:54:31:341 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 1
20:54:34:843 poll node 00:22:a3:00:00:13:39:9b-01
20:54:36:078 Remove websocket 10.0.1.132:50401 after error Unknown error

terafin, 2018-06-13:
Quick one for temp:
21:01:18:683 verified group capacity: 255 and group count: 1 of LightNode 0x0022a3000013399b
21:01:18:683 0x0022a3000013399b found group 0x0001
21:01:18:843 poll node 00:22:a3:00:00:13:39:9b-01
21:01:18:843 discard double entry in binding queue for for 0x0022A3000013399B, cluster 0x0006
21:01:18:843 discard double entry in binding queue for for 0x0022A3000013399B, cluster 0x0008
21:01:19:684 Bind response success
21:01:19:684 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x0022A3000013399B (wait reading or unsupported)
21:01:20:843 poll node 00:22:a3:00:00:13:39:9b-01
21:01:21:341 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 1
21:01:21:681 Bind response success
21:01:21:681 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x0022A3000013399B (wait reading or unsupported)
21:01:22:843 poll node 00:22:a3:00:00:13:39:9b-01
21:01:28:843 poll node 00:22:a3:00:00:13:39:9b-01
21:01:28:843 discard double entry in binding queue for for 0x0022A3000013399B, cluster 0x0006
21:01:28:843 discard double entry in binding queue for for 0x0022A3000013399B, cluster 0x0008
21:01:30:361 Current channel 11
21:01:30:421 no button map for: lumi.weather cl: 0x0402 cmd: 0x0A pl[0]: 000
21:01:30:421 ZCL attribute report 0x00158D00022C644B for cluster 0x0402, ep 0x01
21:01:30:423 discard sensor state push for 3: state/temperature (already pushed)
21:01:30:425 discard sensor state push for 3: state/lastupdated (already pushed)
21:01:30:440 no button map for: lumi.weather cl: 0x0405 cmd: 0x0A pl[0]: 000
21:01:30:440 ZCL attribute report 0x00158D00022C644B for cluster 0x0405, ep 0x01
21:01:30:442 discard sensor state push for 4: state/humidity (already pushed)
21:01:30:444 discard sensor state push for 4: state/lastupdated (already pushed)
21:01:30:463 no button map for: lumi.weather cl: 0x0403 cmd: 0x0A pl[0]: 000
21:01:30:463 ZCL attribute report 0x00158D00022C644B for cluster 0x0403, ep 0x01
21:01:30:465 discard sensor state push for 5: state/pressure (already pushed)
21:01:30:466 discard sensor state push for 5: state/lastupdated (already pushed)
21:01:31:341 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 1
21:01:31:758 Bind response success
21:01:31:758 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x0022A3000013399B (wait reading or unsupported)
21:01:32:204 no button map for: lumi.weather cl: 0x0000 cmd: 0x0A pl[0]: 005
21:01:32:204 ZCL attribute report 0x00158D00022C644B for cluster 0x0000, ep 0x01
21:01:32:204 0x00158D00022C644B skip Xiaomi attribute 0x0005
21:01:32:768 Bind response success

terafin, 2018-06-16:
Tried with a Conner too. Same result :-(  likely will have to return both soon.

manup, 2018-06-16:
Just a wild guess, but does the Pi has proper time, therefore does NTP work?

```
20:52:30:991 discard sensor state push for 1: state/daylight (already pushed)
20:52:30:000 discard sensor state push for 1: state/dark (already pushed)
20:52:31:002 discard sensor state push for 1: state/status (already pushed)
```

The second last log, looks strange since values weren't pushed before.
Currently the code uses absolute timestamps to decide if push is needed, might be problematic in some cases.

terafin, 2018-06-16:
Thanks you!

The time was right, but the timezone was wrong - and just fixed it - that said, sadly still seems to have the same issue :(

manup, 2018-06-16:
I'll see to fix it in the next version by not using timestamp for push handling but a simple flag, but still strange why it doesn't work in your setup.

terafin, 2018-06-17:
Thanks! I'll play around with time settings as I can in the mean time. Appreciate!

closed 2019-03-30 by stale[bot]