dresden-elektronik/deconz-rest-plugin #638: Support for ubisys S1?

felixstorm, 2018-06-12:
Would it be possible to add support for the ubisys S1 in addition to the existing support for the S2? The S1 is already detected as a light, but the power metering sensors are not yet available through the API.

The S1 seems to be pretty much identical to the S2 with the difference being that the S1 can only switch one load up to 3600W by using a relay instead of two triacs (as in the S2). Therefore the S1 does not have the second output endpoint and the first one is of type Mains Power Outlet (0x0009).

Thanks,
Felix

manup, 2018-06-15:
Yes it's on the list and should arrive in the next versions.

closed 2019-03-30 by stale[bot]

Thomas-Vos, 2020-07-01:
The Ubisys S1 (Power Switch S1) still does not show its sensors for power measurement. It is shown in deCONZ but not in the API. Can you have this looked into, please @manup? We love to see this passed on in Hue Essentials.

SwoopX, 2020-07-01:
Seriously? Got the S2 (which is essentially nothing different) and that happily talks through the API...

Thomas-Vos, 2020-07-03:
Hi @SwoopX, thanks for the suggestion. I have tried starting the sensor search again but unfortunately there are still no sensor resources created for the S1.

Here is the light resource:
```json
{
    "etag": "f0885dedcfc1e2fb0e4c7d4cae760055",
    "hascolor": false,
    "lastseen": "2020-07-03T18:06:00.865",
    "manufacturername": "ubisys",
    "modelid": "S1 (5501)",
    "name": "Ubisys On/Off plug-in unit",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": "20200525-DE-FB0",
    "type": "On/Off plug-in unit",
    "uniqueid": "00:1f:ee:00:00:00:5c:04-01"
}
```
And here are some screenshots of the device in deCONZ:
[image]
[image]
[image]
[image]

SwoopX, 2020-07-03:
Try the sensor search and read the basic cluster. While doing that, make sure a load is applied.

robertftvos, 2020-07-14:
@SwoopX Reading the clusters while doing a sensor search in Hue Essentials solved this issue. Also, the measurement value (mA) is now correct in your commit.

7wells, 2020-11-01:
Hello!

When I try to add the S1 to a new and empty group (latest stable phoscon gateway/GUI), I get this error message:
> Warning! Could not add OG_Bad_Heizung to group. Group capacity of the device is reached.
The S1 device (named `OG_Bad_Heizung`) would be the first device in the new and empty group `OG_Bad`.

Can you please give me a hint how I can drill down the reason for this problem (and eventually resolve it)? Thank you so much!

7wells, 2020-11-02:
Well, it worked with the latest stable phoscon gateway and current webGUI. Sorry that I first tried the old webGUI and reported about a problem with it.