dresden-elektronik/deconz-rest-plugin #640: cannot bind aqara switch WXKG11LM to OSRAM On/Off plug-in unit

tbergler, 2018-06-14:
Trying hard to switch my OSRAM plug with the Aquara switch, but no luck. What am I doing wrong? Can you please guide me into the right direction? I have tried using Bind Dropbox in deconz GUI. Also no luck using the REST API. (Version 2.05.29 / 11.6.2018) Is there anything left I can try? Thanks.
    "10": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true,
            "temperature": 2700
        },
        "ep": 1,
        "etag": "6cd742b5f29637c1965e89a8ce245a4a",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.sensor_switch.aq2",
        "name": "lumi.sensor_switch.aq2 10",
        "state": {
            "buttonevent": 1002,
            "lastupdated": "2018-06-14T13:39:45"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:01:e7:d6:04-01-0006"
    }

    "1": {
        "etag": "e354c1beeea089c2c0c72670615dd447",
        "hascolor": false,
        "manufacturername": "OSRAM",
        "modelid": "Plug 01",
        "name": "Thermomix",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": "V1.04.12",
        "type": "On/Off plug-in unit",
        "uniqueid": "84:18:26:00:00:0d:4d:6b-03"
    },

tbergler, 2018-06-14:
[image]

ebaauw, 2018-06-14:
Like most Xiaomi devices, the wireless switch doesn't support binding to lights, or even sending commands.  It only sends attribute reports, which need to be handled by a controller.  In deCONZ, you would create a rule to control the plug when the sensor's `state.buttonevent` changes.

tbergler, 2018-06-15:
Thank's Erik. This was very helpfull. Now I understand I need to put a rule or rules in place similar to take action: 
{
    "actions": [
        {
            "address": "/lights/2/state",
            "body": {
                "on": true
            },
            "method": "PUT"
        }
    ],

As a condition I need to use buttonevent. Only I need to figure out which value the button is reporting somehow. Also in order to toogle the state, how do I achieve this? A second condition for the current state of the light? (when button pressed and light is On, turn it Off // when button pressed and light is Off, turn it On)?
    "conditions": [
        {
            "address": "/sensors/9/state/buttonevent",
            "operator": "eq",
            "value": "1"
        }
    ],
    "name": "Switch On/Off"
}

Best Regards
Thomas

ebaauw, 2018-06-15:
> Only I need to figure out which value the button is reporting. 

Values for `state.buttonevent` are _x_00_y_, where _x_ indicates the button (which will be 1 for the Xiaomi smart swicth) and _y_ the action (_Press_, _Release_, _Hold_, ...).  Note that not all switches support all actions (for all buttons), see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#switches.

> Also in order to toogle the state, how do I achieve this?

I'm not sure if deCONZ yet supports conditions on `state.on` of a `/lights` resource.  Even if it does, I doubt the value will be reported reliably.  Best create a CLIPGenericFlag sensor to record the state explicitly.  You would need to create two rules, for the same `buttonevent` condition: one with an additional condition where the flag is false, and with two actions: setting the flag and the light to true; the other for when the flag is true, setting the flag and the light to false.

tbergler, 2018-06-15:
Awesome. Now I have working rules in place and I am able to use the button. However it is not working as desired, yet. Once I turned on the plug by a single button click (1002) I cannot turn it off using a single click again. The reason for this is, that the buttonevent does not change. As long as I just press the button once its always 1002 and the rule does not trigger. However I can turn off the plug by pressing the button twice (1004) or tree times (1005). In that case the buttonevent changes. I've tried to use "eq" 1002 as well, but no luck. These events at least work: {
    "9": {
        "actions": [
            {
                "address": "/lights/1/state",
                "body": {
                    "on": true
                },
                "method": "PUT"
            },
            {
                "address": "/sensors/3/state",
                "body": {
                    "flag": true
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/9/state/buttonevent",
                "operator": "dx"
            },
            {
                "address": "/sensors/3/state/flag",
                "operator": "eq",
                "value": "false"
            }
        ],
        "created": "2018-06-15T18:10:53",
        "etag": "662509dd57cd98d3f41ee5fe48df3a98",
        "lasttriggered": "2018-06-15T18:33:14",
        "name": "Switch On",
        "owner": "8B0439BD63",
        "periodic": 0,
        "status": "enabled",
        "timestriggered": 23
    },
    "10": {
        "actions": [
            {
                "address": "/lights/1/state",
                "body": {
                    "on": false
                },
                "method": "PUT"
            },
            {
                "address": "/sensors/3/state",
                "body": {
                    "flag": false
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/9/state/buttonevent",
                "operator": "dx"
            },
            {
                "address": "/sensors/3/state/flag",
                "operator": "eq",
                "value": "true"
            }
        ],
        "created": "2018-06-15T18:13:39",
        "etag": "a9d1c0a85d7a125853fb715f86151aca",
        "lasttriggered": "2018-06-15T18:33:16",
        "name": "Switch Off",
        "owner": "8B0439BD63",
        "periodic": 0,
        "status": "enabled",
        "timestriggered": 23
    }
} 

Sensor 3: 
{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "a9d1c0a85d7a125853fb715f86151aca",
    "manufacturername": "Me",
    "modelid": "T1000",
    "name": "My Switch",
    "state": {
        "flag": false,
        "lastupdated": "2018-06-15T18:33:16"
    },
    "swversion": "1.0",
    "type": "CLIPGenericFlag",
    "uniqueid": "0x001fee00000008bb"
}

Sensor 9:
{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true,
        "temperature": 3000
    },
    "ep": 1,
    "etag": "a9d1c0a85d7a125853fb715f86151aca",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.sensor_switch.aq2",
    "name": "Schalter Thermomix",
    "state": {
        "buttonevent": 1002,
        "lastupdated": "2018-06-15T18:33:16"
    },
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:01:e7:d5:4a-01-0006"
}

ebaauw, 2018-06-15:
You need to add a `dx` condition on `state.lastupdated` to capture the update.  Sorry, I should have mentioned that.

tbergler, 2018-06-16:
Thanks Erik. This is brilliant! It works perfect now.
Switch On Rule:
```
{
        "actions": [
            {
                "address": "/lights/1/state",
                "body": {
                    "on": true
                },
                "method": "PUT"
            },
            {
                "address": "/sensors/3/state",
                "body": {
                    "flag": true
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/9/state/buttonevent",
                "operator": "eq",
	    "value": 1002
            },
            {
                "address": "/sensors/9/state/lastupdated",
                "operator": "dx"
            },
            {
                "address": "/sensors/3/state/flag",
                "operator": "eq",
                "value": "false"
            }
        ],
        "name": "Switch On"
    }
```

SwitchOff Rule:
```
{
        "actions": [
            {
                "address": "/lights/1/state",
                "body": {
                    "on": false
                },
                "method": "PUT"
            },
            {
                "address": "/sensors/3/state",
                "body": {
                    "flag": false
                },
                "method": "PUT"
            }
        ],
        "conditions": [
            {
                "address": "/sensors/9/state/buttonevent",
                "operator": "eq",
	    "value": 1002
            },
            {
                "address": "/sensors/9/state/lastupdated",
                "operator": "dx"
            },
            {
                "address": "/sensors/3/state/flag",
                "operator": "eq",
                "value": "true"
            }
        ],
        "name": "Switch Off"
    }
}

```

closed 2018-06-16 by tbergler