dresden-elektronik/deconz-rest-plugin #643: XBee Router Settings

GadgetUK, 2018-06-15:
Despite setting XBEE2 settings as documented in

https://www.dresden-elektronik.de/funktechnik/service/support/faq/deconz/

I'm unable to get the XBEE2 to join the Deconz network.

manup, 2018-06-16:
After configuring the parameters you need to open the network (for example start light search in Phoscon App). Also if I recall correctly the XBee should be power-cycled to let it start joining.

ebaauw, 2018-08-11:
Got myself an Xbee ZigBee Mesh Kit, containing three XBee S2C modules (XB24C on firmware 4060) and three XBee Grove developer boards.  The XBees join the deCONZ network with above mentioned settings.  Pressing the Commission button on the developer board is enough to start the joining.

Note that you can only set the network key (NK parameter) for Coordinator nodes - the XBee returns ERROR when it's configured as a router.  Also note that the network (NK) and link (KY) keys are write-only.

While the XBees are joined to the deCONZ network, they don't (yet) expose anything useful to deCONZ:
[image]

I managed to get the XBees to talk to each other over the deCONZ network in Transparant mode, by setting the Destination Address (DH and DL parameters).  See the [_Basic communication_](https://www.digi.com/resources/documentation/Digidocs/90001942-13/#containers/cont_example_basic_communication.htm%3FTocPath%3DGetting%2520started%2520with%2520XBee%2520Zigbee%7CExample%253A%2520basic%2520communication%7C_____0) example in the Mesh Kit User Guide.

manup, 2018-08-11:
With the non ZCL E6/E8 endpoints it's possible to configure/read/write the XBee pins remotely and also transmit transparent serial data.

If you're interested I can provide the old XBee plugin for deCONZ, it's from 2015, not maintained, but has some hints on the protocol and endpoints.

manup, 2018-08-12:
See the attached zip, be aware it isn't very complete but might provide some value :)

[xbee.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2281284/xbee.zip)

>I tried to add the 0xc105 Digi protocol to general.xml, but no success. Maybe I need to put the modules in API mode, before pairing them.

The ZCLDB won't help much here since the modules don't use ZCL but rather proprietary APS commands. Although general.xml has some hacky support for this, I would just handle it in the plugin itself.

 
>I got the kit to try and prototype some ZigBee controller for my windows shades (wooden venetian blinds) and/or my ceiling fans (see #717). It should be easy enough to connect some actuators (servo motor for the blinds or 240V relay for the fan) to the XBee Grove developer board.

>I hope to be able to write an Arduino or Raspberry app for the XBee to expose a ZHA endpoint with appropriate clusters. I’ll start by exposing the user LED as an on/off light, followed by exposing the user button as a switch.

Sounds cool, might be another good candidate for the `/devices` endpoint. The modules are pretty universal due the free configurable pins.

ebaauw, 2018-08-13:
> See the attached zip, be aware it isn't very complete but might provide some value

Thanks.  Having a quick look, it will require some work (newer QT version, newer deconz C API) before it will even compile.  Do you want to create a public GitHub repository for this plugin?  If not, do you mind if I would?

manup, 2018-08-14:
Hopefully the required changes to support current deCONZ should be minimal and mostly related to Qt4 vs Qt5. The deCONZ library API tries to be backward compatible where possible :)

Feel free to create a repository out of it, maybe at some point it can be merged into REST-API plugin, or alternatively it could a own interface with a WebSocket server on a different port?

ebaauw, 2018-08-20:
https://github.com/ebaauw/deconz-xbee-plugin

ebaauw, 2018-08-20:
I got it to compile, copied it to `/usr/share/deCONZ/plugins`, restarted deCONZ, but nothing.  deCONZ GUI runs normally, but no new entry in the plugins menu, and no clue in the log.  I double-checked file permissions and type.

Do I need a QT5 replacement of `Q_EXPORT_PLUGIN2(xbee_plugin, XBeePlugin)`?  Or do I need to tell deCONZ which plugins to load (I thought it would anything in the directory)?

manup, 2018-08-20:
The Q_PLUGIN_METADATA define is missing, it is needed for Qt 5.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.h#L31-L33

deCONZ will load all plugins in the plugins directory  which implement the deCONZ::NodeInterface.

closed 2018-12-25 by stale[bot]