dresden-elektronik/deconz-rest-plugin #644: Can't find devices on Ubuntu

isabellaalstrom, 2018-06-16:
I downloaded deconz and it finds my Conbee but the Conbee doesn't find anything else. Downloaded deconz on windows too and the problem is not there, I can add everything. Is there some step I've missed on Ubuntu to get it running alright or is this a bug?

manup, 2018-06-16:
If the same ConBee already works on Windows it should show the there connected devices also on Ubuntu. Is the blue node shown on Ubuntu too &dash; meaning deCONZ is connected to ConBee?

>Is there some step I've missed on Ubuntu to get it running alright or is this a bug?

It might be a permission problem. The local user needs to be part of the `dialout` group so deCONZ can connect to the ConBee USB device. This can be fixed by:

`$ sudo gpasswd -a <username> dialout`

Might require a reboot afterwards, to work.

manup, 2018-06-16:
In the Phoscon App under *Menu > Settings > Gateway* you can see if it is connected properly. If it is connetced the firmware version should be visible like this:

[image]

isabellaalstrom, 2018-06-16:
Yes, that shows up just like that. I also checked and the user already is in dialout group!

manup, 2018-06-16:
Good so the device connection is working. If the on windows connected devices are sensors or switches they won't show up on Ubuntu. You can create a backup file on Windows with connected Conbee (Phoscon App > Settings) and restore that file on Ubuntu, it should bring back the devices.

isabellaalstrom, 2018-06-16:
I'm actually not seeing the firmware version I just realized. But that seems to be a known problem on the docker version. Will try to add some other things and do the backup thing tomorrow!

isabellaalstrom, 2018-06-17:
Okey, so I paired a few things on the windows pc, did the backup, changed to the nuc and restored backup there. I paired one xiaomi button/switch, one xiaomi motion sensor, one trådfri dimmer (the ones without buttons) and one trådfri bulb. In the api I can now see all entities, except the light. But only the xiaomi switch registers state changes, the others keep being what they were when on the windows pc. I cannot pair new things either.

isabellaalstrom, 2018-06-18:
I would be happy to help debug with logs etc if I get some guidance, since more ppl seem to be having problems.

manup, 2018-06-18:
> I'm actually not seeing the firmware version I just realized. But that seems to be a known problem on the docker version. Will try to add some other things and do the backup thing tomorrow!

Hmm, I have no experience with the docker image but the version should also be visible there.

Maybe @marthoc can help out here?

https://hub.docker.com/r/marthoc/deconz/
https://github.com/marthoc/docker-deconz

Nicxe, 2018-06-18:
I got the same issue on my rPi. I run deconz and it identifi the gateway and gives it a chanel and ID. But I can´t add any device.

isabellaalstrom, 2018-06-19:
@marthoc Yes, I hadn't gotten around to writing here yet, thank you.

closed 2019-03-30 by stale[bot]