dresden-elektronik/deconz-rest-plugin #645: Can't add IKEA lights to Rpi3

odhammar, 2018-06-17:
I can't add IKEA lights to my ConBee, if i do "Scan for Devices" in the webapp it finds the IKEA light but when I try to add it, it just can't find it. Tried both the "Old" and Phoscon App but no one could add the light. I put the ConBee stick in a windows machine and there it wasn't any problem to add the light.
So I'm stuck here, have uninstalled and reinstalled all needed software without any luck.
Running Raspbian Stretch an deConz 2.05.30

odhammar, 2018-06-18:
I had read your post as well during troubleshooting, but so far nothing. I have now made a brand new image with just Raspbian Stretch and updated it. And then just installed deConz, but still it can't connect to the IKEA light, if I search for devices it finds it, but can't connect it. I had it working before but I think it was when I did a update of Raspbian it stopped working, but not really sure it could have been updating deconz.

odhammar, 2018-06-18:
Are there any king of log files for deconz?

odhammar, 2018-06-21:
I have it working now, but don't ask why, I made a new image again and installed deConz, didn't work but after reset of the IKEA bulb, 5-6 times it suddenly worked.......

isabellaalstrom, 2018-06-21:
I got it working by adding a longer usb cable between my nuc and the conbee. Guess something was messing with the signal. I used the old ui to touchlink reset my lights and then adding them, since I'm having trouble with the power toggel reset.

pkielen, 2018-07-22:
I also have problem adding the IKEA lights, I am using the RaspBee with a pi 3. I can also see them fine when I do "Scan for Devices" in the old interface. Can you add new devices fine now @odhammar and @isabellaalstrom or do you still have problems? Since I'm using the pins on the pi I assume it's not a connection problem?

odhammar, 2018-07-25:
I just bought a IKEA Trådfri LED Driver and connected it without any problem, so it looks at least that  my problem are gone.

manup, 2018-07-27:
@pkielen can you please provide deCONZ and firmware version you are using (visible in the Phoscon App > Menu > Gateway)?

pkielen, 2018-07-27:
Sure, @manup 

Version: 2.05.32 / 6/12/2018
Firmware: 261F0500

closed 2018-12-02 by stale[bot]