dresden-elektronik/deconz-rest-plugin #647: ZHASwitch button presses not trigger websocket events

josephyancey, 2018-06-20:
I've got 4 remotes of two different brands and none of them trigger websocket events when I press any of the buttons:

/api/key/sensors
`{
    "1": {
        "config": {
            "configured": true,
            "on": true,
            "sunriseoffset": 30,
            "sunsetoffset": -30
        },
        "etag": "e5c4dfa8a83f09d5b6106cce0f6a04c6",
        "manufacturername": "Philips",
        "modelid": "PHDL00",
        "name": "Daylight",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2018-06-20T01:56:10",
            "status": 210
        },
        "swversion": "1.0",
        "type": "Daylight",
        "uniqueid": "00:21:2e:ff:ff:01:b9:d9-01"
    },
    "3": {
        "config": {
            "group": null,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "bca33369b7262efe0ce1e3185e388ff6",
        "manufacturername": "Lutron",
        "mode": 4,
        "modelid": "LZL4BWHL01 Remote",
        "name": "LZL4BWHL01 Remote 3",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "swversion": "1.1",
        "type": "ZHASwitch",
        "uniqueid": "ff:ff:00:0f:e7:ff:3c:28-01-1000"
    },
    "4": {
        "config": {
            "group": null,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "dab3ec6cb3acd7da98b77cbfbbfd829d",
        "manufacturername": "Lutron",
        "mode": 4,
        "modelid": "LZL4BWHL01 Remote",
        "name": "LZL4BWHL01 Remote 4",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "swversion": "1.1",
        "type": "ZHASwitch",
        "uniqueid": "ff:ff:00:0f:e7:ff:40:91-01-1000"
    },
    "5": {
        "config": {
            "group": null,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "98415d0e1116db1b87ed148210a5515c",
        "manufacturername": "Lutron",
        "mode": 4,
        "modelid": "LZL4BWHL01 Remote",
        "name": "LZL4BWHL01 Remote 5",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "swversion": "1.1",
        "type": "ZHASwitch",
        "uniqueid": "ff:ff:00:0f:e7:ff:3c:c4-01-1000"
    },
    "7": {
        "config": {
            "alert": "none",
            "battery": 60,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "343455346ca3f3c1c6b667425f37246b",
        "manufacturername": "IKEA of Sweden",
        "mode": 3,
        "modelid": "TRADFRI remote control",
        "name": "TRÅDFRI remote control",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:0b:57:ff:fe:47:86:61-01-1000"
    }
}`

When I connect to the web socket I see change events for the battery level for the IKEA remote as well as change events for the sunlight sensor. I don't see any events at all when pressing any of the buttons on any of the remotes.

I'm using a raspbee on with hassio and the deconz addon by @malloc. My lights are working perfectly but remotes just aren't sending websocket events.

manup, 2018-06-20:
Might be similar to the Philips dimmer switch, please try to Touchlink a remote with a light: hold the remote close to the light and press the `On` button for several seconds (5..10?). The remote should then create a group and put the light in.

*Note* Both devices must be in the same ZigBee network. Also due the lack of a Lutron remote I could never test this, so it might not work.

Reminder to myself, provide the RaspBee/Conbee as Touchlink target so that a remote can bind to the gateways directly.

josephyancey, 2018-06-20:
@manup if I touchlinked a light with my remote, wouldn't the remote then be paired with that light? I'm not wanting to control the lights directly with the remote but instead want to use it for shortcuts.

Regarding the lutron remote, I realize that it's a bit flaky so I've been trying my testing with the IKEA remote first. The lutron ones don't even show up in the web interface.

josephyancey, 2018-06-20:
@manup I tried this with both my IKEA remote as well as my lutron remote. It still doesn't report button presses via the websocket.

I am monitoring the websocket server like this:
```
var host = '10.1.1.4';
var port = 8443;

var ws = new WebSocket('ws://' + host + ':' + port);

ws.onmessage = function(msg) {
    console.log( JSON.parse(msg.data) );
}
```
When the touchlink pairs, I get a groups changed event websocket message. There is a new group visible in the web interface and the remote controls the light. The group in the web admin shows the status of the light changing when pressing buttons on the remote BUT there still aren't any buttonevents sent over websocket. Is there any way to see if deconz even detects the button press?

manup, 2018-06-27:
Strange at least the IKEA remote should trigger some button events since the setup process is more simple than the Philips and Lutron remotes.

Does the websocket work otherwise (light and group events..) or is there total silence?

josephyancey, 2018-06-29:
@manup I can see light and group events on the websocket but absolutely nothing remote related. It's almost like deconz isn't even seeing the button presses. It does see when I pair remotes and when I add lights to the remotes group. When pairing the ikea remote, one of the steps that it asks for is to press and hold the center button. That never seemed to do anything so after a while I wound up closing the modal and the remote was listed in the admin. What is the exact right process to pair that ikea remote? I feel like I'm missing something here.

manup, 2018-06-29:
The press of the center button should trigger the creation of two rules for the next/prev buttons. But even if this fails there should be some events popping up for the top, down and center buttons.

Can you please provide  which versions of deCONZ and the firmware you're running, it's visible in the *Phoscon App > Menu > Settings > Gateway*.

Further for debugging you may start deCONZ with logging:

```
$ deCONZ --http-port=80 --dbg-info=1 --dbg-aps=2 --dbg-zcl=1
```

It's quite noisy but it should show if there is at least received something when pressing some button at lower level in the stack.

josephyancey, 2018-07-01:
Version 2.05.30 / 6/16/2018
Firmware 26050500

It tells me that a new firmware version is available but the upgrade button doesn't seem to do anything except for set the firmware version to 00000000 until I restart. Since I'm using hassio, I think I'll have to SSH directly into the host to upgrade the firmware. With debugging turned on (setup via the hassio config) it doesn't show anything special when pressing buttons on remotes. I'll try and upgrade the firmware and report back.

manup, 2018-07-01:
The firmware is indeed a pretty old version and likely causes the problems here since the received groups commands from the remotes won't be forwarded.

Currently updating RaspBee from within Docker is not supported (when I recall correctly), this will be fixed in future versions but also requires a recent firmware versions.

So to update please do:

```
close deCONZ

$ sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF
```
For the update deCONZ must be installed outside Docker.

josephyancey, 2018-07-01:
@manup updating the firmware fixed my problem.  For anyone using the hassio version of deconz by @marthoc you can ssh directly into the hassio host and update the firmware by following this: https://github.com/marthoc/hassio-addons/issues/4

closed 2018-07-02 by josephyancey

maxlein, 2018-12-31:
I have the same problem currently ...
The tradfri remote is updating data over REST but not on websocket.
The remote has no lights assigned. Just want to use the switch state in openhab.
My sensors and lights work as expected, but switch events are not coming in.

Produkt RaspBee
Version 2.05.48 / 18.11.2018
Firmware 262F0500