dresden-elektronik/deconz-rest-plugin #651: How to trigger creation of default-ct-cw/ww rules

lbschenkel, 2018-06-21:
(This is more of a question than a real issue.)

I have a ConBee (firmware 261F0500) and I'm running DeCONZ 2.05.30 (with `--dbg-info=2`). I have recently purchased 3 IKEA TRÅDFRI remote controls (all with firmware 1.2.214) which I successfully set up in Phoscon. I noticed by exploring the API that two rules were automatically created, `default-ct-cw-N` and `default-ct-ww-N` where `N` was the id of one of the 3 remotes (both rules were for the same remote).

I was intrigued about the purpose of these 2 rules and why they were created for a single remote and not all three, so as an experiment I deleted them via the REST interface, removed the remotes from Phoscon, factoring reset them (by pressing the link button 4 times in quick succession) and re-added them. This time no rules were created. Repeated the process many times but I cannot trigger the creation of these rules any longer.

By inspecting the source code I was expecting the rules to be recreated when the remote was re-added and the UI asks for the remote to be pressed for 10s, but the `ikea remote setup button` message does not show up in the log. I do see it logged once in the past, when the 2 rules were created.

I can see that this behaviour was introduced on https://github.com/dresden-elektronik/deconz-rest-plugin/commit/d0e3ca6735a87c0a00f1c88d78a542fe4decd524, and I tried pressing the power button for 5s, 10s, both when re-adding the remote and afterwards, but I could never trigger this code path besides that single time.

So I have a bunch of questions from this experiment:

1. What's the intended purpose of `default-ct-cw` and `default-ct-ww` rules and commit https://github.com/dresden-elektronik/deconz-rest-plugin/commit/d0e3ca6735a87c0a00f1c88d78a542fe4decd524?
2. How can I trigger this code path to run: https://github.com/dresden-elektronik/deconz-rest-plugin/blame/ae1a917d2634c5d217e5055fcb58fd899675e179/rest_sensors.cpp#L2624?
3. Why does Phoscon ask for the power button on the remote to be pressed for 10s when it is being added? Does it trigger something in the remote itself and/or deCONZ/REST API?
4. When I long press the power button, I see this showing up in the log after approximately 3s:
`no button handler for: TRADFRI remote control cl: 0x0005 cmd: 0x09 pl[0]: 06E`
`no button handler for: TRADFRI remote control cl: 0x0008 cmd: 0x04 pl[0]: 0FE`
and then after approximately 1s (still while the button is being held):
`no button handler for: TRADFRI remote control cl: 0x0000 cmd: 0x00 pl[0]: 001`
Is this normal/expected behaviour?

Any help/clarification is greatly appreciated. Please let me know if you need any additional information.

manup, 2018-06-21:
>By inspecting the source code I was expecting the rules to be recreated when the remote was re-added and the UI asks for the remote to be pressed for 10s, but the ikea remote setup button message does not show up in the log. I do see it logged once in the past, when the 2 rules were created.

Yes normally the rules should be created on the setup button event. However the code was made and tested with older IKEA remote firmware. I'll need to retest it with current firmware to figure out why the process isn't finished.

lbschenkel, 2018-06-21:
> Yes normally the rules should be created on the setup button event. However the code was made and tested with older IKEA remote firmware. I'll need to retest it with current firmware to figure out why the process isn't finished.

Thanks for the quick response. Interestingly, all 3 remotes came loaded with the same firmware (1.2.214) — I did not update them. Since they are all identical, firmware-wise, I'm really struggling to understand how the code would trigger for a single remote, and only once. Let me know if I can assist with this process: I'm available to do any testing/experiments that you may need. 

Btw, just to deepen my understanding, I would appreciate if you could shed some light into my other questions (in case that does not take too much of your time). I'm somewhat new to ZigBee/ConBee/deCONZ and I would like to have better insight of what's going on (it comes with the trade, I'm a developer myself).

manup, 2018-06-22:
Here are some more insights to your questions:

>> 1. What's the intended purpose of default-ct-cw and default-ct-ww rules and commit d0e3ca6?
Normally the left/right buttons don't send commands which are understood by all ZigBee lights. The rules will provide color temperature control.

>> 2. How can I trigger this code path to run: https://github.com/dresden-elektronik/deconz-rest-plugin/blame/ae1a917d2634c5d217e5055fcb58fd899675e179/rest_sensors.cpp#L2624?
Why does Phoscon ask for the power button on the remote to be pressed for 10s when it is being added? 

The trigger should be pressing the large button for (up to) 10 seconds, the remote should then send a special command. Looks like there are differences in firmware version.

> Does it trigger something in the remote itself and/or deCONZ/REST API?

The remote will send the command.

>When I long press the power button, I see this showing up in the log after approximately 3s:
no button handler for: TRADFRI remote control cl: 0x0005 cmd: 0x09 pl[0]: 06E
no button handler for: TRADFRI remote control cl: 0x0008 cmd: 0x04 pl[0]: 0FE
and then after approximately 1s (still while the button is being held):
no button handler for: TRADFRI remote control cl: 0x0000 cmd: 0x00 pl[0]: 001
Is this normal/expected behaviour?

Yes and this seems to be the reason why it isn't triggered, the code to handle is already there but commented out. I think wee need to support both versions.

```
//else if (ind.srcEndpoint() == 0x01 && ind.clusterId() == SCENE_CLUSTER_ID && zclFrame.commandId() == 0x09 && zclFrame.manufacturerCode() == VENDOR_IKEA)
```

manup, 2018-06-27:
>This part I understood, a special "10s power" button (or something similar, I'm wring from memory) is logged in deCONZ. But what do the deCONZ/Phoscon auto-created rules do? I don't have the rules set up and I don't think I'm missing any obvious functionality from the remote (besides the prev/next buttons that don't work out of the box, but that's a known behaviour).

The rules are there to provide a default functions for the prev/next buttons without having to write custom rules. We will release a new switch editor in a few releases which does provide a way to customize button functions in a user friendly way.

>I was wondering why Phoscon asks the user to long press this button during set-up and why the button press is the trigger to create the rules (compared to just creating them). My guess was that the long button press has some side effect on the remote itself (to put it on a known state perhaps), and that is why it is important that this is done at this point. Since you had the trouble to code this and include this step in the set-up procedure, I presume there was a reason. Just curious about it.

To be honest it was the lazy developers wish of providing a way to create/test/recreate the rules with the switch itself :)

lbschenkel, 2018-06-27:
OK, understood it now. Thank you very much for the explanation. And being a developer, I fully understand the lazy developer scenario... :-)

Should I close this or do you want to leave it open to track the "button not recognized" issue due to different firmware? I can rephrase the title and issue description if you want.

closed 2018-11-23 by manup