dresden-elektronik/deconz-rest-plugin #652: Unknown command ID for serial interface

AlbaMustafaj, 2018-06-22:
I am trying to connect Conbee with a sensor (bitron video AV2010/21B) which is ZigBee and HA 1.2 compatible. During phoscon app search it does not become visible and it doesn't appear in deConz app as another node. The only node recognized by deConz is the ConBee despite that in the general.xml file  IAS zone type of sensors were defined, 
So I tried communicating with the sensor through serial port. I implemented the functions to communicate with conbee through the serial port (set/read parameter, create/join network, get state etc). However, when I press the button for 10seconds and the sensor's LED blinks, I sometimes receive an additional buffer which has a command ID of 1c (hexadecimal).  There is no information about this command ID In the deConz serial protocol document, thus I am not able to parse the information in this buffer. How can we access other types of commands rather than those specified in the deConz protocol document? Thank you in advance

AlbaMustafaj, 2018-06-22:
> Please double check the Node List panel, sometimes new nodes are hidden behind existing nodes or placed outside the visible area.

I've tried this, I constantly look at the Node list panel but nothing appears either. I tried and also by resetting it . The sensor and conbee are close to each other too and yes, it hasn't joined the network that's why it doesn't appear in the deConz GUI. There are no lights and this is a magnetic sensor so it is not discovered by the light sensors discovering app. This sensor and it's cluster are also included in general.xml file and I also open the network (set time=255s) in deConz GUI, so I don't understand why this Am I missing any step?

ebaauw, 2018-06-22:
> This sensor and it's cluster are also included in general.xml file

That's not even needed to join the network - the GUI will just show unknown clusters.

> There are no lights and this is a magnetic sensor so it is not discovered by the light sensors discovering app.

The Phoscon app just issues a REST API call to open the network, which works for lights as well as for sensors.

> Am I missing any step?

As I said before, make sure the sensor has been reset (typically holding the reset button for 10 seconds).  Some longshots:
- Open the network first, then reset the sensor;
- Move the magnet and/or briefly press the reset button, continuously while the network is open (to make sure the sensor is awake);
- You might try removing the battery and re-inserting it, while the network is open (the sensor should broadcast a device announcement);
- Can you see and reset the sensor using touchlink?
- You were able to join other devices?  How many end-devices are already connected to the ConBee?  There is a limit (somewhere around ten, if memory serves); add a light to connect more to the network;
- Double-check the network settings in the deCONZ GUI (Device Type: Coordinator, TC link key, Security Mode: High).

If that doesn't help: run deCONZ with `--dbg-info=2` and check the log to see what's going on.  Or sniff the ZigBee traffic using another ConBee, if you have one.

AlbaMustafaj, 2018-06-22:
> Open the network first, then reset the sensor;
Move the magnet and/or briefly press the reset button, continuously while the network is open (to make sure the sensor is awake);
You might try removing the battery and re-inserting it, while the network is open (the sensor should broadcast a device announcement);
Can you see and reset the sensor using touchlink?

I did this and I can't see it. This is the only sensor I have and I have only 1 conbee.

> If that doesn't help: run deCONZ with --dbg-info=2 and check the log to see what's going on. Or sniff the ZigBee traffic using another ConBee, if you have one.

I tried it, deConz runs but there is no output in the command line and no logs file is created (I'm using windows 10). 

>Double-check the network settings in the deCONZ GUI (Device Type: Coordinator, TC link key, Security Mode: High)
Did this as well. 
[image]
I'm not able to make the sensor visible after following all the steps . Can anything else be done?

manup, 2018-06-22:
Recently there were reports that when using ConBee it helped to use a extension cable between ConBee and the PC. It might be the same issue here.

AlbaMustafaj, 2018-06-22:
> 

Recently there were reports that when using ConBee it helped to use a extension cable between ConBee and the PC. It might be the same issue here.
--

 My sensor and conbee are very close to each other so there is no distance issue for me. How does the cable affect the connection, was it because of the distance?

manup, 2018-06-22:
Not the distance but it seems that the signal is getting *really* bad in some cases.

AlbaMustafaj, 2018-06-22:
I will try that, thank you both for trying to help me. What about the serial interface and the commands not specified in deConz serial protocol? I'm getting a buffer with command id 1c when I press LED blinks but its meaning is not specified anywhere.

kylegordon, 2018-06-22:
As I described in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/614#issuecomment-393999119 - my Conbee was adversely affected by the location on the computer it was occupying. Putting it on a 30cm USB extension improved my life dramatically.

manup, 2018-06-22:
> I'm getting a buffer with command id 1c when I press LED blinks but its meaning is not specified anywhere.

The command is quite new and not documented yet, it means that there was a mac data poll request received from a device (mostly battery powered devices do this).

AlbaMustafaj, 2018-06-22:
> The  command is quite new and not documented yet, it means that there was a mac data poll request received from a device (mostly battery powered devices do this).

Thank you! When will the serial protocol update be available to us?

manup, 2018-06-25:
We are currently working on a major update on the whole REST-API documentation. Serial protocol update will come after that, so it might take a while.

However if you're looking for the mac poll notification payload  `0x1C` it is basically:

```
U8   0x02    short address mode
U16  short address

or

U8   0x03  extended address mode
U64  extended address
```

There the address is the one of the polling node.

AlbaMustafaj, 2018-06-26:
yesterday and today I only plugged the conbee in and today I didn't even try to connect it to the sensor. However, I am receiving this data in the command line every hour. I know the last 2 are notifications but I can't understand the content of the long buffer. 
[image]

Is this a notification sent by the sensor since it wasn't able to join the network?

manup, 2018-06-27:
Hard to tell without the whole frame, but according to the values it looks like printable ASCII to me.
You may try to print it as a c-string?

AlbaMustafaj, 2018-06-27:
it means  raspbee bootloaderloader premium
vers. 1.02 
build 2013/starting app

This is what I get once every hour from conbee's serial interface from port.on(data) function.

manup, 2018-06-27:
Ah sorry my bad, I should have mentioned that, since in this case likely deCONZ isn't running and no devices are connected, the firmware will trigger a watchdog reset every hour — the message above is from the bootloader starting. It does no harm but is not in the current documentation. I put it on the todo list.

AlbaMustafaj, 2018-06-28:
I also receive 2 notifications immediately after this message:
c0 0e 01 00 07 00 81 00 69 ff c0
c0 0e 02 00 07 00 81 00 69 ff c0
The protocol says that notifications are received when a data frame is received. Do these notifications refer to the message as a data frame? I send the request to receive command when I get this notifications hoping there is some other data to be received but I don't get anything as a response

AlbaMustafaj, 2018-06-28:
There is 81 and b2 in the Device state field which is of type U8. But in the protocol it says"
Device state can be one of the following:
0000 0011 — Network state
0000 0100 — APSDE-DATA.confirm flag (0x04)
0000 1000 — APSDE-DATA.indication flag (0x08)
0001 0000 — Configuration changed flag (0x10)
0010 0000 — APSDE-DATA.request free slots flag (0x20)

I couldn't find any meaning for 81 and b2

manup, 2018-06-28:
Quick look in the implementation, when the highest bit is set there is also a second status byte included in the message.

```
        status[0] |= 0x80; // has byte1
```

The status byte1 is currently not documented. it contains information about the ZLL touchlink state. So in your above case it is 0x00 which means Touchlink is not active.

Similar to network state in byte0 the bits 0&ndash;2 do describe the Touchlink state:
```
    NOT_CONNECTED  = 0x00,
    TOUCHLINK      = 0x01,
    NET_CONNECTED      = 0x02
```

closed 2018-11-23 by manup

DoctorZIP, 2020-02-02:
> We are currently working on a major update on the whole REST-API documentation. Serial protocol update will come after that, so it might take a while.
>
> However if you're looking for the mac poll [...]

Hi,
Is any updates on serial protocol documentation? I using document version 1.14 that is the last one I hope :) but can't find any information about 0x1C command ID.
Thank you!