dresden-elektronik/deconz-rest-plugin #654: general.xml changes for innr SP120 missing from v2.05.30

ebaauw, 2018-06-23:
The _Electical Measurement_ domain needs to be added to the ZLL profile, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/432#issuecomment-395757777.

manup, 2018-06-23:
Ok, will be added to 2.05.31 scheduled later tomorrow :)

ebaauw, 2018-06-23:
@manup, can you please use this file: [general.xml.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2129960/general.xml.zip)?

It also contains some additional multipliers and divisors for the _Electrical Measurement_ and _Simple Metering_ clusters (only supported by the Heiman smart plug), and the _Window Covering_ cluster 0x0102, used by the Xiaomi curtain (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/593#issuecomment-390404203).

manup, 2018-06-24:
It's merged now and finally the *general.xml* is now part of the public repository :)

Zkronk, 2018-07-28:
I use a Innr SP 120-plug with Deconz v2.05.31 and Home Assistant 0.74.2 (hass.io w/docker), and are not getting any power measurments from the plug.
The only attributes I get from light.Innr_plug_1 in Home Assistant are 
`brightness: 254 
min_mireds: 153 
max_mireds: 500 
effect_list: colorloop 
friendly_name: Innr_plug_1 
supported_features: 41`
And no sensor-entity like sensor.Innr_plug_1 that I get from other power metering plugs like from Fibaro or Hank.

Kane610, 2018-08-02:
Same for me, no power measurement with Innr SP120

Gnaget2, 2018-08-04:
i dont get any power measurement with Innr SP120

deconx 2.05.31 and the innr SP120 reports version 2.0

using Home assistant 0.75.0_

ebaauw, 2018-08-08:
Is `uniqueid` really empty, or did you edit that?

Looks like the plug hasn't been fully configured by the REST API.  Could you open the network/search for devices in the web app/Phoscon app, and re-read the _Basic_ cluster in the deCONZ GUI?  That should trigger the creation of the `/sensors` resources.

If not, try resetting and re-pairing the plug.  If that doesn't help either, could you please post screenshots from the deCONZ GUI showing the node and its endpoints and clusters, the _Node info_ panel, and the _Custer info_ panel for the _Basic_ cluster (after reading the attributes)?

ssahlender, 2018-08-08:
@ebaauw 
Thanks for the quick reply ... I removed it ... here is the full output
`
    "1": {
        "etag": "b54ea3ef4d021de1803b00746e19149c",
        "hascolor": false,
        "manufacturername": "innr",
        "modelid": "SP 120",
        "name": "Innr SP 120",
        "state": {
            "alert": "none",
            "bri": 25,
            "on": true,
            "reachable": true
        },
        "swversion": "2.0",
        "type": "On/Off plug-in unit",
        "uniqueid": "00:15:8d:00:02:17:bd:2c-01"
    },
`
Sorry about the question ... but how do I 

>  re-read the Basic cluster in the deCONZ GUI

ssahlender, 2018-08-08:
@ebaauw 
Ok ... found it here:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/593
>In the web app, click the Open Network button on the Settings page (under Menu in the upper right corner. Then, in de deCONZ GUI, go to the Cluster Info panel of the Basic cluster and press the Read button. That should trigger the creation of the ZHAPresence /sensors resource.

Still quite tricky ... even with that description ... but it is working now

`"10": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "ca8f1d5e3e95945d4334538e5f427d09",
        "manufacturername": "innr",
        "modelid": "SP 120",
        "name": "Consumption 10",
        "state": {
            "consumption": 0,
            "lastupdated": "2018-08-08T21:44:28"
        },
        "swversion": "2.0",
        "type": "ZHAConsumption",
        "uniqueid": "00:15:8d:00:02:17:bd:2c-01-0702"
    },
    "11": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "02056a04136b6b6f101df783dcf7b829",
        "manufacturername": "innr",
        "modelid": "SP 120",
        "name": "Power 11",
        "state": {
            "current": 0,
            "lastupdated": "2018-08-08T21:44:39",
            "power": 0,
            "voltage": 241
        },
        "swversion": "2.0",
        "type": "ZHAPower",
        "uniqueid": "00:15:8d:00:02:17:bd:2c-01-0b04"
    }
}`

Thank you

ebaauw, 2018-08-08:
> This re-read basic cluster is mentioned a lot, (almost like a silver bullet) 

> Still quite tricky ... even with that description ... but it is working now

To create resources, the REST API plugin needs to receive a message from the node _while the network is open_ (for the REST API plugin).  If you previously paired a node with a previous version of deCONZ, it will only create the newly supported resources under these conditions.  The read of the _Basic_ cluster attributes causes such a message and makes sure the _Model Indentifier_ attribute has been read, which is used for whitelisting nodes for `/sensors` resources.

> could we get that available through Phoscon in some way @manup ?

Tricky.  The REST API (on which Phoscon and the web app are based) is asynchronous - it only queues ZigBee requests, but does not wait for the response from the ZigBee network.  In particular a GET only returns previously cached values.  We'd need a new endpoint (on both the `/lights` and `/sensors` resource?) to issue the request, and a web socket notification when the response has been received.  Then the app should wait for the notification, perhaps retrying on timeout, and issue a message when it's done (or fails).

Kane610, 2018-08-22:
@manup @ebaauw I think this is quite important to implement this feature, moving away from the desktop all together. Most people running raspberry pies are definitely locked out of using the desktop in an easy manner especially when using the rasbee.

LordVoodoo, 2018-08-22:
Hey folks,

I have just unpacked a new Innr SP120 and do not get any power measurement.
I have already updated to v2.0.5.35 and deleted the plug in Phoscon App, followed by pairing it again.

Wenn accessing the sensor section (/sensors) not entry is listed, in the /lights part, I get:

`"27":{
    "etag":"9e8cb3f37bd5706ad0d7a.....................",
    "hascolor":false,
    "manufacturername":"innr",
    "modelid":"SP 120",
    "name":"Wohnzimmer Ladestation",
    "state":{"alert":"none","bri":254,"on":false,"reachable":true},
    "swversion":"2.0",
    "type":"On/Off plug-in unit",
    "uniqueid":"00:15:8d:00:01:fe:55:98-01"
    }`

How can I force Deconz to integrate the power consumption?
Even further: After joining the Innr a second time, I did not had to enter a readable name again, it was still "Wohnzimmer Ladestation".

ebaauw, 2018-08-23:
> It is added as a ZLL device.

Because it _is_ a ZLL device (or, more precisely, the profile ID on the endpoints is set to ZLL).  That's why it also works with the Hue bridge (contrary to the SP 110).

> As the picture shows cluster 0702 ( Metering cluster (0×702) ) is unknown.

You don't have the latest general.xml.  Are you running the latest version of deCONZ?

> Further on I want to write to the following attribute in order to get the reporting working

The latest version of deCONZ sets up the attribute reporting alright.  As always, it might take a little while.

manup, 2018-08-23:
>@manup @ebaauw I think this is quite important to implement this feature, moving away from the desktop all together. Most people running raspberry pies are definitely locked out of using the desktop in an easy manner especially when using the rasbee.

The problem with the basic cluster is, that sadly often we just can't read the basic cluster since devices are may be sleeping or not supporting reading it at all (Xiaomi). Waking up a device is very device specific, some wake up periodically and send reports (here we can jump in) or might be triggered by pressing a button..

In my opinion a button won't fix the problems and is too geeky for the Phoscon App.

We need to improve:
* Collect which devices cause problems
* Querying of missing attributes after a device wakes up
* Improve or add proper setup wizards

kasteleman, 2018-08-23:
I’m on deconz 2.05.20 Ubuntu Linux, you want me to switch to beta?

> On 23 Aug 2018, at 19:52, Erik Baauw <notifications@github.com> wrote:
> 
> deCONZ

ebaauw, 2018-08-23:
Support for the SP120 is in the latest beta.  I'm not sure about availability of the Ubuntu releases, I'm on Raspberry.

kasteleman, 2018-08-23:
That did the trick. Now it shows the right info for the clusters, but the values are not visible in de Phoscon app.
What am I missing?

ebaauw, 2018-08-23:
Not sure if the measurements are supported by Phoscon yet.

manup, 2018-08-23:
The values are not visible in the Phoscon App yet.

kasteleman, 2018-08-23:
Ah yet! That pops up the question: estimate?

;-)

jo-me, 2018-09-02:
Hi, I'm on 

RPi2 + Conbee
Version 2.05.35
Firmware 261F0500

and just tried to add a Innr SP120 plug. The "light" part showed up but no sensors.

Tried to remove it from the DB and re-add it but did not help.

Any ideas?
Will it fix itself after a while or does deconz need to have the info right away?

Thanks

jo-me, 2018-09-03:
@ebaauw : it does not show up under sensors. 
Do you mean I just have to try paring until I get lucky? Or is something missing in deconz that currently prevents this alltogether?

ebaauw, 2018-09-07:
Changes have been made and general.xml is now included in deconz-rest-plugin package.

closed 2018-09-07 by ebaauw

Kane610, 2018-09-07:
What is the process to get the sensor? Remove and readd?

ebaauw, 2018-09-07:
> What is the process to get the sensor? Remove and readd?

I'd try powering down all routers, and re-pairing the plug close to the RaspBee/ConBee, but it might be tricky without the GUI to force-read the _Basic_ cluster attributes.

Make sure you're on a recent version of deCONZ, preferably with firmware 0x26230500.

> Most people running raspberry pies are definitely locked out of using the desktop in an easy manner especially when using the rasbee.

All my Raspberry Pies are headless, but I still run the graphical environment over VNC.  I set _Authentication_ to _VNC Password_ and created `/etc/avahi/services/rfb.service`, so they're discovered and working with macOS _Screen Sharing_ (incl. shared cut/paste buffer).  With [_ScreenSharingMenulet_](https://itunes.apple.com/us/app/screensharingmenulet/id578078659?mt=12), they're only a mouse click away.
```xml
<?xml version="1.0" standalone='no'?>
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">
<service-group>
	<name replace-wildcards="yes">%h</name>
	<service>
		<type>_rfb._tcp</type>
		<port>5900</port>
	</service>
</service-group>
```

Kane610, 2018-09-07:
Then I wait a couple of weeks before upgrading to deconz and firmware. Still sounds like a lot of luck to me. I'm running headless through docker so no option of GUI for me.

Gnaget2, 2018-09-27:
still no power sensor on my inner sp120 added a third one yesterday with 2.35.39 and latest FW (405).

running deconz web app with conbee