dresden-elektronik/deconz-rest-plugin #655: v2.05.30 is creating ZHAPresence sensor resources for my Hue dimmer switches

ebaauw, 2018-06-23:
```json
{
  "config": {
    "battery": 90,
    "duration": 60,
    "on": true,
    "reachable": true
  },
  "ep": 2,
  "etag": "47705c3ad60d4497d143f12ad3cf35e1",
  "manufacturername": "Philips",
  "modelid": "RWL021",
  "name": "Presence 2",
  "state": {
    "lastupdated": "none",
    "presence": false
  },
  "type": "ZHAPresence",
  "uniqueid": "00:17:88:01:10:49:55:03-02-fc00"
}
```

ebaauw, 2018-06-23:
It's probably because it has a _Binary Input_ cluster 0x000F on endpoint 0x02, which now triggers the creation of a ZHAPresence sensor for the Samsung SmartThings Arrival sensor.  The clusterId is overwritten in https://github.com/dresden-elektronik/deconz-rest-plugin/blob/a60076bd096269c8dd81aa78841c2aa1c2558720/de_web_plugin.cpp#L3667, so the `uniqueid` doesn't end in `-02-000f`.

I'll whitelist the SmartThings Arrival sensor under the _Binary Input_ cluster, so the ZHAPresence will be created only for this sensor.

manup, 2018-06-23:
Good find, if needed I can add a cleanup before database is loaded to get rid of eventually created RLW02x ZHAPresence sensors (modelid: RWL02* and type: ZHAPresence).

ebaauw, 2018-06-23:
I just deleted them manually.  The ZHAPresence sensors would only be created if you open the network long enough in v2.05.30, but it might be prudent to check anyway.