dresden-elektronik/deconz-rest-plugin #657: Xiaomi Cube appears twice

mtdcr, 2018-06-26:
After pairing my cube with deCONZ 2.05.30, it appeared twice:

```
  "4": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true,
            "temperature": 2300
        },
        "ep": 2,
        "etag": "<some etag id>",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.sensor_cube.aqgl01",
        "name": "Mi Magic Cube",
        "state": {
            "buttonevent": 7000,
            "lastupdated": "2018-06-26T12:33:50"
        },
        "type": "ZHASwitch",
        "uniqueid": "<mac>-02-0012"
    },
    "5": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true,
            "temperature": 2300
        },
        "ep": 3,
        "etag": "<different etag id>",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.sensor_cube.aqgl01",
        "name": "lumi.sensor_cube.aqgl01 5",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "type": "ZHASwitch",
        "uniqueid": "<same mac>-03-000c"
    },

```
I expected to see only one. I guess it wouldn't be a problem if the battery status wouldn't show up twice in Home Assistant.

Btw., it doesn't have a temperature sensor, AFAIK.

ebaauw, 2018-06-26:
> I expected to see only one.

Better adjust your expectation.  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/611#issuecomment-396733267.

>  guess it wouldn't be a problem if the battery status wouldn't show up twice in Home Assistant.

Best log that issue with the Home Assistant plugin for deCONZ.  It should combine the two resources into one device, based on the mac address, with only one battery status.  That’s no different from how the Hue bridge exposes the Hue motion sensor as three diffrent resources, each with the same config attributes for `battery`, `ledindication`, and `usertest`.

> Btw., it doesn't have a temperature sensor, AFAIK.

Welcome to the wonderful world of Xiaomi.  Most of their devices actually include a crude temperature sensor, that’s reported in a very Xiaomi specific way, together with the battery voltage.  Hence it’s exposed as `config.temperature` instead of a separate ZHATemperature resource.

mtdcr, 2018-06-28:
> Most of their devices actually include a crude temperature sensor, that’s reported in a very Xiaomi specific way, together with the battery voltage. Hence it’s exposed as config.temperature instead of a separate ZHATemperature resource.

Do you have proof for that? Because according to the values deCONZ lists for various types of Aqara devices, my room has 0.0, 26.0 and 30.0 °C at the same time, while it really has around 23°C. Aqara's LAN gateway doesn't show temperatures for these devices. Instead it shows the battery's voltage in units of mV, where 2600 and 3000 would make sense. See http://docs.opencloud.aqara.cn/en/development/gateway-LAN-communication/.

Btw., Phoscon reports 0°C for all of my magnet sensors in its user interface. This creates a bad impression for people getting in touch with it for the first time. The other "temperatures" of switches and the cube don't show up in Phoscon, however.

ebaauw, 2018-06-28:
> Do you have proof for that.

See #42 and https://github.com/dresden-elektronik/deconz-rest-plugin/blob/1cc9e0f80670a42d8498a1737c3feae2c45b8f0d/de_web_plugin.cpp#L7479.

> Because according to the values deCONZ lists for various types of Aqara devices, my room has 0.0, 26.0 and 30.0 °C at the same time, while it really has around 23°C.

I did say “crude”...

closed 2019-03-30 by stale[bot]