dresden-elektronik/deconz-rest-plugin #659: Phoscon APP behind nginx reverse proxy

markuspf, 2018-06-27:
I am running phoscon behind a nginx reverse proxy (amongst other things to have https. etc). Unfortunately the webapp seems to use a very hamstrung way to create URLs by concatenating `"http://"` + `ip` together.

I am not an expert in web-development, but there should be a way for the webapp to use a certain (configurable, detectable?) BASE-URL for use behind a reverse-proxy.

I have, for the time being, solved the problem by `sed`-ing the url patterns out of all `js` files in the webapp...

manup, 2018-06-27:
Indeed, I think if we use just `//` in the Javascript requests the browser should use the correct prefix — http or https — based on the initial site call. I put it on the todo list.

oscar-b, 2018-07-27:
Just ran into this as well, using Traefik as my Docker reverse proxy. 

```
Mixed Content: The page at 'blob:https://deconz.abc.xyz/8e93304b-7048-4e88-b47b-1578eef3f360' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://deconz.abc.xyz/api/config'. This request has been blocked; the content must be served over HTTPS.
```

For some reason, it seems like it retries on port 8080?

manup, 2018-08-07:
The latest version 2.05.34 has the mentioned fixes for the Phoscon app (not old webapp) so it should work with the reverse proxy, please let me know if it works.

oscar-b, 2018-08-12:
This forwards to an invalid address (This is not the page you are looking for):

Sensor control of group -> Add rule -> n×-sensor configuration (WebApp 2016) -> Create

manup, 2018-08-17:
Upps this link shouldn't be visible at all, will be fixed in 2.05.35.

closed 2018-12-22 by stale[bot]

flechaig, 2019-09-04:
This is how I managed to deal with this issue:
```
  location ~ (/.*) {
    set $p 80;
    if ($http_upgrade = "websocket") {
     set $p 443;
    }
    if ($http_upgrade = "") {
     set $p 80;
    }
    proxy_pass http://[censored IP address]:$p$1;
    proxy_set_header Host $host;
    proxy_set_header X-Forwarded-For $remote_addr;
    proxy_http_version 1.1;
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection "upgrade";
  }
```
With this setup, Phoscon app works fine and websockets also. Be careful: if statement in NGINX costs some performance, so it is not suitable for high load instances.