dresden-elektronik/deconz-rest-plugin #661: Raspbee as a router in raspbee network

Equidamoid, 2018-06-28:
Hi, 

I'm trying to add second raspbee as a router to existing network. I read numerous threads here on this topic (although most of them is about adding raspbee to hue net), but still can't make it to work.
What I did so far:
 - copypasted TC link key and network key from coordinator settings
 - also set channel mask to the same single channel
 - set static NWK address (as it was suggested in one of other threads)

As a result the router raspbee joins the network for ~5-10 sec, even shows some other nodes (with active links), also appears on coordinator's deconz graph with quite some links to other nodes, and then disconnects. 

May be related, after that the channel mask is reset to another single channel.

PS:
this is a bit offtopic here, but I haven't found any other places for it

manup, 2018-06-28:
>May be related, after that the channel mask is reset to another single channel.

Seems to be a case where deCONZ tries to be clever, the database contains the last used a channel and I guess it falls back revert to that. Please  try to use the channel change function in *Phoscon App > Menu > Settings > Gateway* and point it to the proper channel.

Equidamoid, 2018-06-28:
@manup 
Didn't help, unfortunately. However, deleting the local files used by deCONZ (`~/.local/share/dresden-elektronik`) and restarting it worked as charm.

Unfortunately, I can't give you a better reproduction scenario than "setup a network with one raspbee, try to connect another as a router doing some random (and probably not so smart) changes in the network configuration (both keys, TC address, etc) and then boom, the channel mask gets magically reverted shortly after connecting".

manup, 2018-06-29:
Hmm strange normally specifying the channel via app should pin it. The zll.db database contains then the chosen channel and should make sure the channel will be configured.

Do you use the latest versions?

closed 2019-03-30 by stale[bot]