dresden-elektronik/deconz-rest-plugin #662: Cannot discover devices previoisly paired elsewhere

jamesking420, 2018-06-28:
Struggling to get second with a conbee (attached to a raspberry pi running hassio), to connect to my hue bulbs or Aqara switches and buttons.

I've tried resetting them, but they don't get discovered in phoscon. 

I guess - with the hue bulbs - I'm not really sure on the correct way to reset them? They're currently paired to a hue bridge and I've tried factory resetting it (expecting that to reset the bulbs), but that didn't seem to have helped.

jamesking420, 2018-06-28:
It appears that using touchlink was an error and that will in fact simply pair the remote to the hue bridge.

Can we do touchlink with deconz conbee, for the same effect?

James

Otherwise - is there a recognized method to reset the hue bulbs, that doesn't involve buying the hue dimmer switch?

manup, 2018-06-28:
>Can we do touchlink with deconz conbee, for the same effect?

You can open the old WebApp *Phoscon > Menu > Help > Open Wireless Light Control*, in there go to *Menu > Settings* and scroll down. With the scan for devices you can initiate a Touchlink scan and then reset the hue bulbs.

jamesking420, 2018-06-28:
Thank you, that worked brilliantly.

Can you tell me if something similar will work for the Aqara wireless switches and buttons?

jamesking420, 2018-06-29:
Thanks - managed to get this working for one worless light switch this morning before work. Will try the rest later. :-)