dresden-elektronik/deconz-rest-plugin #665: Unknown serial interface commands

AlbaMustafaj, 2018-06-29:
I am having a difficult time with making the sensor and the conbee communicate. Both appear in deConz GUI now but my aim is to make them communicate through the serial interface. So I monitored the serial port while deConz app was running and some of the commands written and read were ; 

C0 1D F0 00 12 00 0B 00 01 BD 3F 2F EE 73 09 00  4B 12 00 E3 FB C0         
    
C0 05 ED 00 07 00 00 00 07 FF C0         

 C0 04 EA 00 07 00 00 00 0B FF C0 C0 05 EB 00 07  00 00 00 09 FF C0    
etc... There is no documentation of the commands with the following ids: 1D, 05, 04, 03. I think they are may be the reason that I am not able to establish the communication because maybe I need to use them in my serial interface functions. What do these 4 commands serve for?

manup, 2018-06-29:
Here is a dissect of the above `1D` command, note that the command is experimental and shouldn't be needed in practice since the firmware will do the mapping on it's own.

```
1D Update neighbor request

C0
1D  Update neighbor request
F0  Sequence number
00  Reserved (0)
12 00  Frame Length
0B 00  Payload Length
01  Action add
BD 3F   neighbor short address
2F EE 73 09 00 4B 12 00  neighbor mac address
E3 FB  CRC16
C0
```
The `05` command is the same `0x17` but in a older version which lacks the last byte (RSSI)

C0 05 ED 00 07 00 00 00 07 FF C0

The `03` command is the former version `0x12`.

Note in recent deCONZ versions the newer command versions 0x17 and 0x12 are automatically used as soon as deCONZ has queried the firmware version.

Which versions of deCONZ and firmware are you running (can be seen in deCONZ > Help > about deCONZ)?

The `04` command should be documented? It's the APS Data Confirm command.

AlbaMustafaj, 2018-06-29:
> `Which versions of deCONZ and firmware are you running (can be seen in deCONZ > Help > about deCONZ)?`

Version 2.5.20
Firmware 0x261F0500
This was in the official page, where can I find the latest?

> The 04 command should be documented? It's the APS Data Confirm command

Sorry about that

AlbaMustafaj, 2018-06-29:
Also when I make a  aps data request using nwk, short address destination 00 00 I start getting these messages consequently. Can it be that the request is transmitted to the sensor as well and it is replying?
[image]

The content of 1c command is still unclear to me

manup, 2018-07-03:
>Version 2.5.20
Firmware 0x261F0500
This was in the official page, where can I find the latest?

Firmware looks fine, this is the latest. The latest beta version is updated regularly and referred on the main page https://github.com/dresden-elektronik/deconz-rest-plugin#installation

>Also what is the meaning of this:
>    C0 1C 86 00 0C 00 05 00 02 BD 3F FF E6 6A FC C0

That's the MAC Poll command `1C`  this is received when for example the sensor asks the gateway if there is any new data. Receiving this often is common for some sensors in their wake or setup phase.

>The content of 1c command is still unclear to me

Here's a dissect of the above command:
```
1C  Mac Poll
86  Sequence number
00  Status OK
0C 00  Frame length
05 00  Payload length
02  Short address mode
BD 3F  Short address of the polling node
FF  LQI
E6  RSSI
6A FC  CRC16
```

AlbaMustafaj, 2018-07-03:
I thought 1c was sent when the sensor wants to pair with the dongle. I looked at the documentation and saw that the pairing happens automatically but I still don't understand how I can realise that the sensor has joined the network. I don't receive any other data from the serial port except the 1c command which happens once in a while. Does this mean that it is okay not to get any on.port(data) when the sensor joins the network?

There is one more issue with the commands. I cannot send or receive data from the sensor because either the sensor is not joining (and I don't know how to check if it has joined) or  I'm probably doing something wrong with this part of enqueue send request:
- U16             ASDU length is 0–127 —                   The APS frame payload length
- U8[*]           ASDU                                                    The APS frame payload

I couldn't find detailed explanation about these fields of the command so I don't know what to put there. For instance in the deConz GUI I see that the zone status of the sensor takes different values ( 0x0015, 0x0014,0x0010,0x0011) when I move the magnet. How can I use the serial protocol commands to get the sensor's zone status information? How can this be specified in the request, is it something to be put in the ASDU part?

manup, 2018-07-03:
> I thought 1c was sent when the sensor wants to pair with the dongle. I looked at the documentation and saw that the pairing happens automatically but I still don't understand how I can realise that [...]

Devices, which are in factory new state after a reset, can join a network when the permit join flag is set by *ZDP Permit join request*. Normally after join and after a power cycle the sensor should send at least a *ZDP Device announce* command as broadcast to the network.

>There is one more issue with the commands. I cannot send or receive data from the sensor because either the sensor is not joining (and I don't know how to check if it has joined) or I'm probably doing something wrong with this part of enqueue send request:

The ASDU holds the payload frames for layers on top of APS like ZDP and ZCL, it is best for the general frame format of those to look into ZigBee specification and also ZigBee cluster library specification.

Since the REST plugin is open source you can also look at how we assemble the payload, but it's quite scattered over the code base. For IAS zone status you may look how the ZCL frame is prepared to be put in the APS ASDU as payload:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/ias_zone.cpp#L69

AlbaMustafaj, 2018-07-03:
Thank you, I am investigating about them. Through monitoring I discovered that I should use  write command on parameter 0x0021 with the value 0xFF to permit join but this parameter was not in the protocol. Now the sensor sends 1c commands continuously and it is waiting for data from the dongle I guess. Is there any un/documented command that can serve as a dongle's response to these commands?

manup, 2018-07-05:
>Thank you, I am investigating about them. Through monitoring I discovered that I should use write command on parameter 0x0021 with the value 0xFF to permit join but this parameter was not in the protocol.

Thats also possible but will only allow nodes to join trough the coordinator, the Permit Join request on the other hand will allow nodes to join via other routers on the network too (mostly lights or other mains powered devices).

> Now the sensor sends 1c commands continuously and it is waiting for data from the dongle I guess. Is there any un/documented command that can serve as a dongle's response to these commands?

With the 1c mac poll request the sensor only asks if there is any data queued for it, since the sensor can only receive when it is awake. There is no specific response to that, each mac poll request might cause sending the next APS requests addressed to the sensor from the queue or just an mac ACK meaning no data queued.

AlbaMustafaj, 2018-07-05:
> Permit Join request on the other hand will allow nodes to join via other routers on the network too 

But where can I get the content of zdp permit join request?

>  cause sending the next APS requests addressed to the sensor 

Also I looked at Zcl specification but I couldn't find how to parse the APS  payload for request and receive commands, I am also still investigating your code but its complicated. Can you suggest some more specific material to look to? Not being able to send the correct ASDU is preventing me from communicating with the sensor because I still can't set the proper values to the command 0x012 to request the zone status or any other parameter

manup, 2018-07-05:
> But where can I get the content of zdp permit join request?

The ZDP commands are described in the ZigBee Pro specification, from the ZigBee website http://www.zigbee.org/zigbee-for-developers/zigbee-pro/

> Also I looked at Zcl specification but I couldn't find how to parse the APS payload for request and receive commands, I am also still investigating your code but its complicated. Can you suggest [...]

APS requests in general are also described in the ZigBee Pro spec. ZDP and ZCL do both have a general  frame format for ZCL it's described in the ZCL spec "General ZCL Frame Format".

For the IAS Zone Status you need to assemble the ASDU a frame as described in the ZCL spec: "Security and Safety / IAS Zone / Server".

closed 2018-12-02 by stale[bot]