dresden-elektronik/deconz-rest-plugin #666: Sending turn_off + transition via Home Assistant does not turn off light.

vandalon, 2018-06-29:
When sending a `"entity_id": "light.keuken_3","transition":"10"` in home assistant the light dims to 1% but does not turn off. This does work on hue bridge connected lights.

Kane610, 2018-06-29:
This is probably an issue with the component and not with deconz. I'll have a look in a day or two

vandalon, 2018-06-29:
What component exactly? Can I check something for you?

Kane610, 2018-06-29:
Well... The DeConz component :). You have already stated that transition doesn't work with this specific implementation

vandalon, 2018-06-29:
Ah within Home Assistant you mean :)

Kane610, 2018-06-29:
Yes

jamesking420, 2018-06-29:
I've just come to report this bug too! Coincidence!

Kane610, 2018-06-29:
A really weird coincidence that a third party submitted a fix for it just now. https://github.com/home-assistant/home-assistant/pull/15222

lbschenkel, 2018-06-29:
Since this was a bug in HASS that got fixed, I suppose this should be closed?

Kane610, 2018-06-30:
Yes it can. @vandalon can you close this issue?

vandalon, 2018-06-30:
Fixed in home assistant https://github.com/home-assistant/home-assistant/pull/15222

vandalon, 2018-06-30:
It was an evil issue anyway 😊

closed 2018-06-30 by vandalon

Kane610, 2018-06-30:
Rofl!

jamesking420, 2018-07-07:
This is still broken in 0.73

Kane610, 2018-07-08:
That is true @jamesking420 it will come with the 74 release in two weeks