dresden-elektronik/deconz-rest-plugin #668: Noob security questions

Blysterk, 2018-06-30:
Hello everyone, I'm new to this beautiful world.

I would like to buy the Raspbee to design a home alarm system, connecting it with xiaomi sensors for windows and movements, all managed by the home assistant.

The only doubt I have is if the sensors, once associated with my raspberry, can be accessed by third parties. For example, someone could connect to motion sensors and figure out if someone is at home.

Sorry for this noob question.
Congratulations for all your work.

manup, 2018-07-01:
When the sensors are paired with the RaspBee ZigBee network (think of it as a small LAN) the communication is encrypted within this network and only devices which are paired can communicate and listen in that network. You have control over which devices are allowed to join the network.

The RaspBee/Raspberry Pi is the bridge between the ZigBee network and your LAN. Therefore everything is as safe as your home network is.

Blysterk, 2018-07-01:
Thank you so much for the explanation. Now I'm relaxed.

manup, 2018-07-05:
Close this for now since, answered :)

closed 2018-07-05 by manup