dresden-elektronik/deconz-rest-plugin #669: Bosch Pro-Grade ZigBee Wireless Motion Detector ISW-ZPR1-WP13

mercenaryballs, 2018-07-02:
Tried to pair this PIR sensor and it seems to be having issues.

[image]

It claims to get Zigbee HA 1.2 certified.

http://www.zigbee.org/zigbee-products-2/#zigbeecertifiedproducts/productdetails3/5553fe56f7f0b6063927b0c5/
https://www.amazon.com/gp/product/B00PRBMDKW/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

mercenaryballs, 2018-07-03:
Also not sure if it helps, but here is the code for smart things to get it working.

https://raw.githubusercontent.com/tomasaxerot/SmartThings/master/devicetypes/tomasaxerot/bosch-motion-detector.src/bosch-motion-detector.groovy

mercenaryballs, 2018-07-27:
Any word on this? I still have this one sitting on my desk and would love to use it with deconz. Let me know if I can provide anymore info.

josephyancey, 2018-08-02:
Same here. +1

manup, 2018-08-02:
Since this is a IAS Zone based motion sensor it might work just by whitelist the model ids specified in the bosch-motion-detector.groovy file. I'll put it on the list for the next releases.

josephyancey, 2018-11-06:
Has there been any movement on this?

manup, 2018-11-07:
I've added the two models listed in the groovy file, not sure about the manufacturer code (0x1133).
With .46  (arriving soon) you may try to start the sensor search and then reset the device to let it rejoin.

Please let me know if the sensor appears in the REST-API I can then add it to the Phoscon App.

apmon, 2018-11-14:
I just tried pairing one of these devices with deCONZ .47. however, it doesn't look like it is working. the deCONZ gui doesn't seem to show the second dot to select a cluster and show the cluster info.

manup, 2018-11-14:
Hmm looks like some descriptors weren't queried.
How long is it running like this? Is the sensor connected directly to the gateway or a light in between?

apmon, 2018-11-14:
It looks like it is directly connected to the conbee, and it has now been connected roughly since an hour. I.e. shortly before I added the comment with the same result. I tried to reset and repair the device, but it looked like the same result

manup, 2018-11-14:
Does blink blue or green from time to time? Normally motion sensors would poll quite frequently (green blinking) and missing descriptors should be queried there.

apmon, 2018-11-14:
I have seen it blink yellow and red, but I don't think I have seen it blink blue or green

manup, 2018-11-14:
Hard to tell what's the cause maybe logging yields some more info, if you can please pair the sensor again while logging is enabled:

    deCONZ --dbg-aps=2 --http-port=80

apmon, 2018-11-14:
Not sure if this is of much use, but attached is a log during an attempted reset/repairing of the sensor. I did not delete any previous entries of that sensor in deCONZ.

The IEEE address of the sensor is 0x00155f006c247389.
[deCONZ.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2582769/deCONZ.log)

manup, 2018-11-14:
I see some traffic, right after join the sensor looks for a IAS Zone server:

```
15:36:53:200 APS-DATA.indication srcAddr: 0x00155f006c247389, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0006, lqi: 255, rssi: -42
15:36:53:200 	asdu: 02fdff040100010005

Match descriptor request

02 seq
fffd short addr
0104 HA profile id
00 server cluster count
01 client cluster count
0500 IAS Zone Cluster
```

This should be handled but requires a recent firmware, which firmware version are you running?

apmon, 2018-11-14:
I am running firmware 262F0500 so that should be the latest?

manup, 2018-11-14:
Yes that's the correct firmware, I wonder why the node and endpoint descriptors aren't queried by deCONZ.

Can you please make a screenshot as above where the sensor is selected with Node Info Panel as well as blue coordinator node endpoint list is shown.

manup, 2018-11-14:
I think I found an issue, it's the mac capabilities, they are 0x00.

```
15:36:53:097 APS-DATA.indication srcAddr: 0x00155f006c247389, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 255, rssi: -43
15:36:53:097 	asdu: 01a9db8973246c005f150000

01 seq
a9db nwk address
8973246c005f1500 mac address
00  mac capabilities
```

That's new, other end-devices had 0x80 (allocate address flag set).
The code doesn't like that.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/rest_sensors.cpp#L2301

I'll need to add an exception for the bosch devices, anyway the screenshot would be interesting to check if node descriptor was queried.

apmon, 2018-12-19:
I have just tried again with 2.05.52, but it still looks like the same issue. Is there any more information I can provide to help diagnose these issues?

manup, 2018-12-19:
Did you also reset the sensor and rejoin while sensor search is active?

manup, 2019-01-21:
Do you use the latest version 2.05.57? It has many fixes related to these old entries and always should create new ones.

apmon, 2019-01-21:
Yes I am.

But this seems to not even be about re-joining. I took out the battery of the device, so that it definitely doesn't send any updates. Then I deleted a device in the GUI. Restarted deCONZ and the device re-appeared.

So it is possible that when I thought I was re-joining a sensor, it wasn't actually doing anything and just an old record re-appeared.

manup, 2019-01-21:
>Then I deleted a device in the GUI. Restarted deCONZ and the device re-appeared.

Ah ok, currently deleting in the gui (deCONZ) won't delete anything in the REST-API or database. This will be supported in future versions.

You need to delete the sensor in the REST-API or if shown in the Phoscon App.

apmon, 2019-01-21:
Ah, thanks. Do you have a link to the instruction of how to delete a device through the API, as it doesn't show up in the Phoscon App.

Also, is there a way if it shows up neither in the API, nor in Phoscon App? I have a "IAS Ancillary Control Equipment" entry in the GUI, that doesn't seem to have initialized correctly. Possibly it was this Bosch motion detector, but I am not entirely sure when or where it came from.

apmon, 2019-01-21:
Great. After correctly deleting the entry, and then doing a re-join, it looks like this motion detector is now working! Thanks! It is correctly updating the presence through the API and home-assistant picks it up too.

manup, 2019-01-21:
Cool glad it works, I've added the sensor to the supported devices section in the wiki.

Closing the issue for now.

closed 2019-01-21 by manup

josephyancey, 2019-02-27:
@manup I'm not sure if I should reopen this issue or open a new one. I finally decided to connect my Bosch motion sensor and it isn't pulling temperature. I've attached the api results from my Bosch. The motion works perfect (as you can see in it's status) but the temperature just reports null. It's been paired for a few days and hasn't changed.

`    "11": {
        "config": {
            "battery": 100,
            "duration": 60,
            "on": true,
            "reachable": true
        },
        "ep": 5,
        "etag": "8e129f8a815f9b3f6fa9b630ded34a6f",
        "manufacturername": "Bosch",
        "modelid": "ISW-ZPR1-WP13",
        "name": "Upstairs Presence",
        "state": {
            "lastupdated": "2019-02-27T05:21:50",
            "lowbattery": false,
            "presence": true,
            "tampered": false
        },
        "swversion": "20140929",
        "type": "ZHAPresence",
        "uniqueid": "00:15:5f:00:b4:4c:fa:30-05-0500"
    },
    "12": {
        "config": {
            "battery": 100,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 5,
        "etag": "16661d0f89688b001578c0ee724eeb59",
        "manufacturername": "Bosch",
        "modelid": "ISW-ZPR1-WP13",
        "name": "Upstairs Temperature",
        "state": {
            "lastupdated": "none",
            "temperature": null
        },
        "swversion": "20140929",
        "type": "ZHATemperature",
        "uniqueid": "00:15:5f:00:b4:4c:fa:30-05-0402"
    }`