dresden-elektronik/deconz-rest-plugin #671: Firmware cannot update

raissim, 2018-07-03:
Hi!
I'm using deCONZ in a raspbee and when I run it (with or without sudo) it tells a firmware update is necessary. Even after I've clicked on update firmware and the update reaches 100%, after a few seconds the "A firmware update is needed" message appears again everytime.

I've tried updating from deCONZ 2.4.36 to 2.5.31 but it didn't solve the problem. Does anyone have a clue to its cause?

Thanks!

raissim, 2018-07-10:
Seems like the Raspbee board is broken, it's the same issue as https://github.com/dresden-elektronik/deconz-rest-plugin/issues/585

closed 2018-07-10 by raissim

manup, 2018-07-10:
Have you tried manually updating it on the command line while deCONZ is not running?

raissim, 2018-07-11:
That's what I did, with `sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF`  
What I get is  
```
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
no device specified use RaspBee (/dev/ttyS0)
using firmware file: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF
reset target

error: failed to read from device`