dresden-elektronik/deconz-rest-plugin #672: XIAOMI Aqara Switch 2018 - Ghost-tips ?

BigWumpus, 2018-07-03:
Hello,
I got 2 Aqara Switches ("modelid": "lumi.sensor_switch.aq2", yes 1x 2x 3x 4x press) and connect them to my home.
Now i go crazy. The system is switching more than one time an hour, because there are events from the switches, even when I'm not at home or sleeping.
Is there a problem with the software or what ?
I also have a cc2531-stick here, but it only knows 1x and 2x press...

manup, 2018-07-04:
Can you provide more details on your setup/versions? Do you mean Websocket events or events in  other systems like Home Assistant?

BigWumpus, 2018-07-05:
RaspBee-Gateway with the latest beta .31 running on an own RPi2B, connected via LAN to ioBroker on another RPi.
Now, even I am not at home, the switches gives their single-press-signal. 
Maybe the software gives updates to iobroker?

`CAPCOM
Flur-Switch2: 1002
Flur-Switch1: 1002
Flur-Switch1: 1002
CAPCOM
BaZi:958 / ArZi:0/0
CAPCOM
Flur-Switch2: 1002
Flur-Switch1: 1002
Flur-Switch2: 1002
Flur-Switch1: 1002
Flur-Switch2: 1002
Flur-Switch1: 1002`

manup, 2018-07-05:
The Xiaomi devices give hourly reports on battery/temperature (coarse measurement). These are reported over websocket events. The important detail here is that for the switches the `lastupdated` timestamp is not updated, because the timestamp refers to the button event.

So I guess ioBroker currently doesn't check if the received event is related to the button press?

BigWumpus, 2018-07-05:
Today i paired the switches to my cc2531-USB-stick and they are working fine now.
The light-sensor will stay at the same level for about 12 hours or so, in the night is is lighted as on afternoon, they say...
I also bought a OSRAM-Switch... They don't work with either interface...

manup, 2018-07-05:
The OSRAM switches aren't supported currently. For the Xiaomi issue please reach out to the ioBroker devs if they handle the mentioned events correctly.

BigWumpus, 2018-07-10:
So,
my Xiaomi Aqara switches work fine at my cc2531-zigbee-stick, but they only know 1xpress and 2xpress.
These switches on Deconz are recognized as 1x, 2x, 3x, 4x-press.
They should also have a "shake"-event at the switches with "gyroskop".
But at deconz there are several events per hours without pressing ("ghost"). And it is not every hour!
The measurement of the battery is maybe 12h or more.
The measurement of light (in motion-sensors) is about up to 20h.
My Aqara window-sensors and motion-sensors are working great the deconz without any false-event.
There is something strange, and there are new events to come !

tomfritz1, 2018-07-15:
My aqara switch is working great - adds two lights in Phoscon and I can turn on/off one light. But I failed to control it with sensor control because the group itself does not turn on/off (only the single light in the group can be switched). Is there any chance to integrate this in upcoming Phoscon versions @manup

BigWumpus, 2018-08-05:
OK,
I tried to install the homebridge-hue-utils, but stopped because of ...
For me it will be much easier, if I can send a device via parcel to someone (in germany) for testing...

Paalap, 2018-08-13:
The issue is actually caused by the iobroker driver/adapter. To be more detailed: on the battery/temperature report the driver is not using the websocket content, instead it pulls full property request which is actually updating also the buttonevent...

...thus I've changed the adapter's script to ignore buttonevent updates if these are too old...

closed 2018-12-19 by stale[bot]

chimpy, 2019-01-09:
Hi,

I am having the ghost issue with the Xiaomi motion sensors (older Mijia version). How have other users dealt with this? 

Here's a debug log courtesy of Node Red. We can see that the new state is on, the old state was off (the house was empty when this occurred):

    Jan 09 15:26:26 hassbox Node-RED[789]: 9 Jan 15:26:26 - [info] [debug:2da5fbea.9e830c]
    Jan 09 15:26:26 hassbox Node-RED[789]: { topic: 'binary_sensor.bathroom_motion',
    Jan 09 15:26:26 hassbox Node-RED[789]:   payload: 'on',
    Jan 09 15:26:26 hassbox Node-RED[789]:   data:
    Jan 09 15:26:26 hassbox Node-RED[789]:    { entity_id: 'binary_sensor.bathroom_motion',
    Jan 09 15:26:26 hassbox Node-RED[789]:      old_state:
    Jan 09 15:26:26 hassbox Node-RED[789]:       { entity_id: 'binary_sensor.bathroom_motion',
    Jan 09 15:26:26 hassbox Node-RED[789]:         state: 'off',
    Jan 09 15:26:26 hassbox Node-RED[789]:         attributes:
    Jan 09 15:26:26 hassbox Node-RED[789]:          { battery_level: 100,
    Jan 09 15:26:26 hassbox Node-RED[789]:            on: true,
    Jan 09 15:26:26 hassbox Node-RED[789]:            friendly_name: 'Bathroom Motion',
    Jan 09 15:26:26 hassbox Node-RED[789]:            device_class: 'motion' },
    Jan 09 15:26:26 hassbox Node-RED[789]:         last_changed: '2019-01-09T13:39:33.310509+00:00',
    Jan 09 15:26:26 hassbox Node-RED[789]:         last_updated: '2019-01-09T13:39:33.310509+00:00',
    Jan 09 15:26:26 hassbox Node-RED[789]:         context: { id: 'de0e8346b6ff441383e1c74a983f9349', user_id: null } },
    Jan 09 15:26:26 hassbox Node-RED[789]:      new_state:
    Jan 09 15:26:26 hassbox Node-RED[789]:       { entity_id: 'binary_sensor.bathroom_motion',
    Jan 09 15:26:26 hassbox Node-RED[789]:         state: 'on',
    Jan 09 15:26:26 hassbox Node-RED[789]:         attributes:
    Jan 09 15:26:26 hassbox Node-RED[789]:          { battery_level: 100,
    Jan 09 15:26:26 hassbox Node-RED[789]:            on: true,
    Jan 09 15:26:26 hassbox Node-RED[789]:            friendly_name: 'Bathroom Motion',
    Jan 09 15:26:26 hassbox Node-RED[789]:            device_class: 'motion' },
    Jan 09 15:26:26 hassbox Node-RED[789]:         last_changed: '2019-01-09T14:26:26.295955+00:00',
    Jan 09 15:26:26 hassbox Node-RED[789]:         last_updated: '2019-01-09T14:26:26.295955+00:00',
    Jan 09 15:26:26 hassbox Node-RED[789]:         context: { id: 'b25608effa0d4f6883d8e9d9120d5a7a', user_id: null } } },
    Jan 09 15:26:26 hassbox Node-RED[789]:   _msgid: '2479c92c.9b0006' }

Here is the the event logged by deconz:

    Jan 09 15:26:26 hassbox deCONZ[17299]: 15:26:26:291 no button map for: lumi.sensor_motion ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 002
    Jan 09 15:26:26 hassbox deCONZ[17299]: 15:26:26:291 ZCL attribute report 0x00158D00022777C3 for cluster 0x0000, ep 0x01
    Jan 09 15:26:26 hassbox deCONZ[17299]: 15:26:26:291 0x00158D00022777C3 extract Xiaomi special attribute 0xFF02
    Jan 09 15:26:26 hassbox deCONZ[17299]: 15:26:26:291     64 on/off 1
    Jan 09 15:26:26 hassbox deCONZ[17299]: 15:26:26:291     01 battery 3045 (0x0BE5)
    Jan 09 15:26:26 hassbox deCONZ[17299]: 15:26:26:291     03 unsupported index (data type 0x21)
    Jan 09 15:26:26 hassbox deCONZ[17299]: 15:26:26:291     04 unsupported index (data type 0x24)
    Jan 09 15:26:26 hassbox deCONZ[17299]: 15:26:26:291     05 unsupported index (data type 0x21)
    Jan 09 15:26:26 hassbox deCONZ[17299]: 15:26:26:291     06 unsupported index (data type 0x20)

manup, 2019-01-09:
Handling of the 0xFF02 report is new in 2.05.54. If it always says on/off 1, we can disable using this as event source for the Xiaomi Mi motion sensor.

chimpy, 2019-01-09:
Thanks for the reply :)
If I do a grep of the log for the 0xFF02 report and the corresponding on/off state would that be helpful?

manup, 2019-01-09:
>If I do a grep of the log for the 0xFF02 report and the corresponding on/off state would that be helpful?

Thanks, I think the info should be enough for now, will put an alpha version soon to check if it works and the next beta by the weekend.

chimpy, 2019-01-10:
OK thanks!

koen01, 2019-06-19:
Could it be this has been re introduced with latest release? I'm having ghosts events from xiaomi magnet sensors (started with latest deconz update)