dresden-elektronik/deconz-rest-plugin #673: RaspBee/deCONZ service does not start after boot

MrSponti, 2018-07-04:
I've installed the deCONZ modul on PI3 under Jessie. As long as there is no additional serial device connected everything is working as expected. For other communication purposes I connected an Arduino and an RF receiver via USB:
```
pi@pidev:~ $ lsusb
Bus 001 Device 005: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 004: ID 1915:1006 Nordic Semiconductor ASA
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
After rebooting the deCONZ modul will no loger initialize and communicate with the configured zigbee devices. 
Any idea to solve this issue?

```
pi@pidev:~ $ sudo /usr/bin/GCFFlasher_internal -l
rmmod: ERROR: Module usbserial is in use by: ch341
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
RaspBee: /dev/ttyS0 (default)

no FTDI devices found
```

MrSponti, 2018-07-05:
Sorry, my fault, an adjustment of **/boot/config.txt** solved my problem!

Using the lirc modul in parallel to the deCONZ modul require to change the default input/output pins used by lirc in **/boot/config.txt** by editing the  **dtoverlay=lirc-rpi** entry.

```
# Uncomment this to enable the lirc-rpi module
#
#dtoverlay=lirc-rpi                                             #  default pin GPIO18 is used
dtoverlay=lirc-rpi,gpio_in_pin=12,gpio_out_pin=16               # changes input pin to GPIO12
```

MrSponti, 2018-07-05:
Using lirc in parallel with the deCONZ modul requires an adjustment of /boot/config.txt.

closed 2018-07-05 by MrSponti

manup, 2018-07-05:
With the latest version of deCONZ you can also directly specify which device to use like ``--dev=/dev/ttyUSB0`.