dresden-elektronik/deconz-rest-plugin #674: IKEA DIMMER - "Turn the dimmer clockwise shortly." not working

FezVrasta, 2018-07-04:
Hi, I have successfully paired one of my IKEA dimmers and it's working fine, now I'm trying to pair a second one but after I press 4 times the pair button and it gets detected, the wizard asks me to "Turn the dimmer clockwise shortly", when I do that nothing happens.

If I close the wizard, I see the dimmer in the list, but no events are triggered from it.

I tried to replace the dimmer with a new one and I'm having the same problem. Do you have any idea to fix it?

renegade2k, 2018-07-09:
Got the same issue yesterday.
My solution after a few tries: 
I deleted the dimmer and paired it once more.
After the wizard asks me to turn it clockwise, i turned it just a bit (like 45°), but pretty fast clockwise, then it worked and i was finished with the wizard.
Still, my Dimmer does not work for connected lights, but this may be another problem ...

FezVrasta, 2018-07-10:
Do you have to re-assemble it before rotating or is it okay if I rotate just the "core"?

renegade2k, 2018-07-10:
I've not reassabled it, as it would cause "random movement" for the sensor, i thought.
Just turn the core carefully without pressing the button.

PS: i was using the dimmer for some time with the Tradfri gateway. Got kinda same problem there. Turning the Dimmer carefully and slow had no effect on light. Turning it once fast caused the lights to go to 100% and after this i was able to reduce the brightness by carefully turn in conter-clockwise.
So i think the dimmer got something like "deep-sleep-mode" or such, which you can "wake" when giving it a little more speed.

FezVrasta, 2018-07-10:
Thanks, I get the events just fine through Home Assistant, but I have to rotate it slowly or it register events on the opposite direction also

repvik, 2018-07-26:
I'm having issues connecting one of these dimmers. I managed to connect it once, but there was no reaction when trying to dim the light. So I deleted it and tried pairing it again, and it fails every single time on the rotating step. What does "Turn the dimmer clockwise shortly" even mean?

sveip, 2018-08-25:
Same problem here, clockwise can be either way, depending on how you hold it. Tried maybe 20 times now, all failed.

sveip, 2018-08-25:
I did some more testing, and it seems like Deconz is doing something wrong. Even when the inclusion fails, the dimmer is added, but doesn't work. Can it be that it's added too early, and that is blocking it to properly pair @manup?

FezVrasta, 2018-08-25:
Yes I noticed it as well, honestly, anyway, the dimmer isn't that great. I returned it and ordered some Xiaomi buttons 🤷‍♂️

kalkih, 2018-09-24:
Did anyone get the Trådfri dimmer working? For me It is recognized and shows up after adding it but doesn't register any events.

ACrazyConcept, 2018-09-25:
All of mine got paired some time back. Though I didn't check if they sent events to home assistant back then. I just used them as regular dimmers.
Then a few weeks ago I resat and paired a couple to test with home assistant. I couldn't get any of them to pair. Then I read a suggestion that the button should be held for at least 10 seconds. Well past the time the diodes start to blink. That seemed to do the trick for one of them. But only one, And it does not send events to home assistant. I then checked some of the initially paired ones, and they do send events to home assistant.
I had upgraded deconz firmware in the meantime, don't know if that has changed the success rate.

closed 2019-05-30 by stale[bot]

HeyJonny, 2019-08-20:
I have the same issue, so this issue should not be closed

paalolsen, 2019-09-22:
Same issue here. The dimmer works fine a few hours, then gets unresponsive. Changed battery and all.
The wired part, is tha the app still sees it as ok, and I can control the lights from the app

jplopezll, 2019-11-28:
I was able to make a dimmer work, but I had to use a combination of both the Phoscon web interface and the deCONZ app in the Raspberry I am using.

I firstly added the switch in Phoscon. Took me some time as the device looks to go to sleep if you stop moving it, so I kept moving it slowly while the first step of the recognition took place. Then I left it on a flat surface, pressed on the "Next" button and did a quick turn clockwise.

Once added, I was not able to make it respond to the lights I was associating with Phoscon. So, I decided to do a manual binding of the dimmer with the light to be controlled using the "Bind Dropbox" panel in deCONZ and selecting the ONOff cluster of the dimmer as "Source" and the light to be controlled as "Destination". After pressing the "Bind" button I was able to control the light.

To be tested if I am able to select groups or scenes as destination.

Crypticfish, 2020-02-03:
Same issue over here. 

Version 2.05.72 / 12/12/2019
Firmware 26490700 (Conbee II)

I managed to get one dimmer working after several attempts.
But for my second dimmer I can not get past "Turn the dimmer clockwise shortly." I spent close to 30 minutes resetting the dimmer, waiting for it to pair and then turning it again and again to no avail. I tried the quick 45 degree turn described above, both lying flat on a table and vertical against a wall, as well as continuous rotation for 60 seconds in both orientations.

The one time I was successful with the first dimmer was two consecutive quick 45 degree turns against a vertical surface.

If anyone has a reliable technique for pairing these dimmers, maybe they could post a video of the correct motion?