dresden-elektronik/deconz-rest-plugin #676: ATSMART Z603 3 gang light switch

aronteh, 2018-07-04:
Possible to get Deconz to identify this as light switch. Currently can turn on and off all 3 switches via On/Off cluster. But it not recognize as a light switch in Phoscon and REST API.
[image]
Basic Cluster Info
[image]
[image]

ebaauw, 2018-07-04:
Is this a wired switch?  Mains powered?  In that case, it should be exposed already as three light resources, or three lights in Phoscon.  It doesn’t look like it can be used to control other devices.

If you don’t see any light resources, could you please post a screenshot of the _Node info_ panel as well?  Are you running a recent version of deCONZ?  It’s an end device, we added support for those as lights a while ago.

They seem to support it in SmartThings, https://community.smartthings.com/t/dh-for-atsmart-z6-3-gang-switch/78849.

aronteh, 2018-07-05:
It is a wired switch and powered from main 240V. It is not exposed in Phoscon at all.
[image]
Running on deCONZ 2.05.31-qt5 and it is an end device.
It work in SmartThings, but I am migrating all my Zigbee devices to deCONZ and HASS

aronteh, 2018-07-05:
Yes, this switch does not require or have any neutral wire connection. It will stop communication if disconnected from S1/S2/S3 or the light bulb(s) is remove.

@manup 
Below are the DH groovy script I used previously in SmartThings.

/*
 *  Custom Device type for ATSMART Z6 Triple Switch Controller http://www.atsmart.net/index.php?s=/Home/index/atsmart_switch
 *
 *  Copyright 2017 Andy Chan
 *
 *  Licensed under the Apache License, Version 2.0 (the "License"); you may not use this file except
 *  in compliance with the License. You may obtain a copy of the License at:
 *
 *      http://www.apache.org/licenses/LICENSE-2.0
 *
 *  Unless required by applicable law or agreed to in writing, software distributed under the License is distributed
 *  on an "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the License
 *  for the specific language governing permissions and limitations under the License.
 *
 */
metadata {
	definition (name: "ATSMART Z6 Triple Switch Controller", namespace: "simic", author: "simics@gmail.com") {
        capability "Refresh"
        capability "Polling"
        capability "Sensor"
        capability "Configuration"
        capability "Switch"
		capability "Actuator"
		
        command "on2"
        command "off2"
        command "on3"
        command "off3"

        attribute "switch2","ENUM", ["on","off"]
        attribute "switch3","ENUM", ["on","off"]
 		attribute "switchstate","ENUM",["on","off"]        
 

    	//fingerprint profileId: "0104", inClusters: "0000", outClusters: "000D,0006"
        //fingerprint inClusters: "0000 0001 0003 0004 0005 0006", endpointId: "01", deviceId: "0100", profileId: "0104"

	}

	// simulator metadata
	simulator {
    }

	// UI tile definitions
	tiles {

		standardTile("switch1", "device.switch", width: 2, height: 2, canChangeIcon: true) {
			state "off", label: '${name}', action: "Switch.on", icon: "st.switches.switch.off", backgroundColor: "#ffffff"
			state "on", label: '${name}', action: "Switch.off", icon: "st.switches.switch.on", backgroundColor: "#79b821"
		}

		standardTile("switch2", "device.switch2", width: 2, height: 2, canChangeIcon: true) {
			state "off", label: '${name}', action: "on2", icon: "st.switches.switch.off", backgroundColor: "#ffffff"
			state "on", label: '${name}', action: "off2", icon: "st.switches.switch.on", backgroundColor: "#79b821"
		}

               standardTile("switch3", "device.switch3", width: 2, height: 2, canChangeIcon: true) {
			state "off", label: '${name}', action: "on3", icon: "st.switches.switch.off", backgroundColor: "#ffffff"
			state "on", label: '${name}', action: "off3", icon: "st.switches.switch.on", backgroundColor: "#79b821"
		}
        
        standardTile("switchstate", "device.switchstate", width: 2, height: 2, canChangeIcon: true) {
			state "off", label: '${name}' , icon: "st.switches.switch.off", backgroundColor: "#ffffff"
			state "on", label: '${name}', icon: "st.switches.switch.on", backgroundColor: "#79b821"
		}
[... 107 lines cut ...]
	
    def cmd = []

    cmd << "st rattr 0x${device.deviceNetworkId} 0x01 0x0006 0x0000"	// Read on / off value at End point 0x01 
    cmd << "delay 150"

    cmd << "st rattr 0x${device.deviceNetworkId} 0x02 0x0006 0x0000"	// Read on / off value at End point 0x02 
	cmd << "delay 250"
	cmd << "st rattr 0x${device.deviceNetworkId} 0x03 0x0006 0x0000"	// Read on / off value at End point 0x03
	cmd << "delay 350"
	
    cmd
}

def configure() { 
    log.debug "Binding SEP 0x01 DEP 0x01 Cluster 0x0006 On/Off cluster to hub"         
        def configCmds = [

        "zdo bind 0x${device.deviceNetworkId} 0x01 0x01 0x0006 {${device.zigbeeId}} {}", "delay 500",
        "zdo bind 0x${device.deviceNetworkId} 0x02 0x01 0x0006 {${device.zigbeeId}} {}", "delay 500",        
        "zdo bind 0x${device.deviceNetworkId} 0x03 0x01 0x0006 {${device.zigbeeId}} {}", "delay 1500",        
        ]
    log.info "Sending ZigBee Bind commands to 3 Button Switch"
    return configCmds
    }

ebaauw, 2018-07-06:
> We’ll have to whitelist the switch. Note to self: also for setting up attribute reporting.

See my latest PR.  It uses manufacturer code 0x1002, VENDOR_EMBER, which was already whitelisted for attribute reporting for the Heiman plug, so only one line of code.

aronteh, 2018-07-06:
Thanks you for the PR. I just notice that the VENDOR_EMBER  does not match the IEEE address manufacturer prefix from the ATSMART Node Info.

ebaauw, 2018-07-06:
It’s the manufacturer code from the node descriptor.  That’s why we needed the screenshot of the _Node info_ panel.

aronteh, 2018-07-07:
Got it. Looking forward to the new release once the PR is merge.
Thanks you.

closed 2018-07-07 by aronteh